espressif/esp-idf #9797: CoreDump: assert failed in vTaskStepTick

ztefanjo, 2022-09-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

No coredump

### What is the actual behavior?

We collect coredumps from our test-fleet. The attached coredump has been reported once so far and indicates that the vApplicationSleep is not working fully.
The following log can be seen in the dump:
assert failed: vTaskStepTick tasks.c:2951 (( xTickCount + xTicksToJump ) <= xNextTaskUnblockTime)\")

[core-dump_51.1_Wifi-BLE-App_42.3_220915-140032.elf.txt](https://github.com/espressif/esp-idf/files/9584637/core-dump_51.1_Wifi-BLE-App_42.3_220915-140032.elf.txt)

### Steps to reproduce.

Unfortunately, this is not known, as the coredump was fetched from the device a while after it crashed.

ztefanjo, 2022-11-03:
@zikalino : This error still happens occasionally while using LightSleep and we are very close to our final customer release. Please let us know what we can do to avoid the error (any patch/configuration change or similar?).

aircable, 2023-04-26:
I can also confirm this as a bug. It happens during Wifi initialization and is likely a race condition when light sleep power management is enabled. This is on the ESP32-S3R8V and the master branch of the IDF.
Sometimes I can get a build that does not run into this problem, but then I just have to disable light sleep. 
Also note, FreeRTOS is configured with TICKLESS_IDLE at 1000Hz.
So far I have not seen the problem on the ESP32 we use in a different product, only on the S3.

Stack trace:
```
assert failed: vTaskStepTick tasks.c:3061 (( xTickCount + xTicksToJump ) <= xNextTaskUnblockTime)

Backtrace: 0x40375cfa:0x3fcb18b0 0x40382389:0x3fcb18d0 0x40388bb2:0x3fcb18f0 0x4038348a:0x3fcb1a10 0x4037b68d:0x3fcb1a30 0x403831e3:0x3fcb1a50
0x40375cfa: panic_abort at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/esp_system/panic.c:451

0x40382389: esp_system_abort at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x40388bb2: __assert_func at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/newlib/assert.c:81

0x4038348a: vTaskStepTick at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3061 (discriminator 1)

0x4037b68d: vApplicationSleep at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/esp_pm/pm_impl.c:622

0x403831e3: prvIdleTask at /home/juergen/AIRcableLLC/ESP_SW/NEWEST/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4363

```

expresspotato, 2023-12-05:
This is also present on an ESP32, running v4.4.6 and a 1000 Hz tick rate and BLE.

`assert failed: vTaskStepTick tasks.c:2999 (( xTickCount + xTicksToJump ) <= xNextTaskUnblockTime)
Backtrace: 0x4008277e:0x3ffbc840 0x40096c29:0x3ffbc860 0x4009fcb1:0x3ffbc880 0x4009aac2:0x3ffbc9a0 0x4008ebe4:0x3ffbc9c0 0x4009a696:0x3ffbc9e0 0x4009d235:0x3ffbca00
0x4008277e: panic_abort at /Users/kevin/esp/esp-idf-v4.4.6/components/esp_system/panic.c:408
0x40096c29: esp_system_abort at /Users/kevin/esp/esp-idf-v4.4.6/components/esp_system/esp_system.c:137
0x4009fcb1: __assert_func at /Users/kevin/esp/esp-idf-v4.4.6/components/newlib/assert.c:85
0x4009aac2: vTaskStepTick at /Users/kevin/esp/esp-idf-v4.4.6/components/freertos/tasks.c:2999 (discriminator 1)
0x4008ebe4: vApplicationSleep at /Users/kevin/esp/esp-idf-v4.4.6/components/esp_pm/pm_impl.c:666
0x4009a696: prvIdleTask at /Users/kevin/esp/esp-idf-v4.4.6/components/freertos/tasks.c:4082
0x4009d235: vPortTaskWrapper at /Users/kevin/esp/esp-idf-v4.4.6/components/freertos/port/xtensa/port.c:142
`

tshcherban, 2024-01-08:
the same for me, ESP32-S3. Power management, automatic light sleep, DFS enabled, 1000 Hz tick rate. Using BLE listening, WiFi connectivity and some Serial printing runtime stats (arduino as a component). Asserting at different places/times, sometimes can run for a few minutes, sometimes crashes even before successful connecting to WiFi

hbf-sthlm, 2024-02-26:
Any news on this bug @sudeep-mohanty ? Do you know when it could be solved, approximately? Thank you.

/Henrik

sudeep-mohanty, 2024-02-27:
Hi @hbf-sthlm, Unfortunately we do not have a timeline yet for this issue. What would help us in debugging this is if you could provide a minimal example/application which can reproduce this issue.

hbf-sthlm, 2024-02-27:
Thank you for a quick response @sudeep-mohanty.

Actually, in my case, I found that the problem didn't appear when I powered the ESP32-S3 independently (not from the USB port of my computer). When measuring current consumption with the Power Profiler Kit II from Nordice Semiconductor, I could see that the ESP32-S3 did indeed enter light sleep mode without crashing repeatedly. The ESP32-S3 only crashed when exiting light sleep _if I powered it with the USB port_ on my computer. That seemed strange, and after a lot of running in circles I found the following information:

https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32s3/api-reference/kconfig.html#config-usj-no-auto-ls-on-connection

I interpret above linked section of the ESP-IDF docs to mean that it is _expected_ that the ESP32-S3 should crash when exiting light sleep while being connected via USB to a computer.

I would suggest you add this information a little more prominently, for example in the Power Management, Sleep Modes and other relevant parts of the ESP-IDF online documentation.

Also, is this a bug that will be solved? It would be good if the ESP32-S3 didn't crash while entering/exiting light sleep while connected to a computer.

Thx

hbf-sthlm, 2024-02-28:
@TD-er Exactly, I am using the native USB of the ESP32-S3 and its connected from my computer to the USB port of the ESP32-S3 board I am using for my current project. In addition to the unofficial board I am using in my project, I have this morning also tested with an official ESP32-S3-DevKitC-1 from Espressif, and the "problem" is apparent there as well. But as mentioned in the docs (here: https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32s3/api-reference/kconfig.html#config-usj-no-auto-ls-on-connection) it seems like this is expected behaviour.

tshcherban, 2024-02-28:
I've faced the problem regardless Native USB or Serial monitor (and pwoering externally)

HiFiPhile, 2024-02-28:
Me too, I'm using ESP32-C3 without USB function, I'll try if I can make a minimal example.

ztefanjo, 2024-02-28:
Also in our case, it happened without USB. But it was a while since I saw the error now.

michaelboeding, 2024-11-09:
Also seeing this error

sudeep-mohanty, 2024-11-11:
Hi, We are investigating potential solutions to avoid this assert failure and would update the thread once we have a viable one. Thanks.

ztefanjo, 2024-11-18:
@michaelboeding , if I remember correctly, the workaround that "solved" the problem for us was to increase CONFIG_FREERTOS_IDLE_TIME_BEFORE_SLEEP from 2 to 3 (which is the default value).

esp-wzh, 2024-12-01:
Have you enabled `CONFIG_ESP_PHY_MAC_BB_PD` in sdkconifg? @michaelboeding

esp-wzh, 2024-12-01:
Could you please try this fix? @michaelboeding 

[IDFGH-8319.patch.zip](https://github.com/user-attachments/files/17967966/IDFGH-8319.patch.zip)

esp-wzh, 2024-12-01:
Sorry, I accidentally added extra changes to the patch. The patch has been updated. Please download it again and try again.

esp-wzh, 2024-12-01:
Okay... I need to ask my BLE colleagues tomorrow if they have any opinions. I will update here if there is any progress. You can use tshcherban's workaround for now.

tshcherban, 2024-12-01:
I have not debugged that too deep because had to complete a project. I have just made a dumb change to the comparison logic include greater+equal condition
https://github.com/espressif/esp-idf/issues/9797#issuecomment-1880779912

tshcherban, 2024-12-01:
my changes are the following:

add
```TickType_t toRemove = xTicksToJump - (xNextTaskUnblockTime - xTickCount);```
change increment to account new value (whithin _taskENTER_CRITICAL_ protected section):
```xPendedTicks += toRemove;```
change decrement of xTicksToJump to account new value:
```xTicksToJump -= toRemove;```

esp-wzh, 2024-12-02:
> Not sure if this means anything also but it seems the --- 0x42000020: _stext at ??:? may be related somehow to https://github.com/espressif/esp-idf/issues/11558 ?

This is not a problem, this log is from idf_monitor trying to perform `addr2line` on the hexadecimal number printed by the chip.

>  the device would completely power off

Does this mean that the rtc_pin in the hold state of esp32s3 has lost its output? Can we know the status of esp32s3 when the problem occurs, from log or current wave?

> Any suggestions or other steps for testing a fix?

From the config you attached. Could you try the following configuration to see if it affects this issue?
1. Enable `CONFIG_PM_SLP_IRAM_OPT` and `CONFIG_ESP_WIFI_SLP_IRAM_OPT`  (It is strongly recommended to enable!!)
2. Try Disable  `CONFIG_ESP_PHY_MAC_BB_PD`  ? 
3. esp32s3 is a dualcore chip, why use unicore mode? (`CONFIG_FREERTOS_UNICORE=y`)

The BLE team said that they have not encountered such problems under cpu_max=160 / cpu_max=40 configuration. 

If you can provide a minimal reproducible routine, it will help us find the root cause quickly. @michaelboeding

esp-wzh, 2024-12-02:
> There is no output since it's in light sleep - unless there is a way to make sure all logs come through while in light sleep? The connection will disconnect/reconnect as its running and switching between.

Maybe `esp_rom_output_tx_wait_idle` in esp_rom_uart.h is what your want.

> Is this incorrect? Also in this guide it also says to put the [CONFIG_PM_SLP_IRAM_OPT](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/kconfig-reference.html#config-pm-slp-iram-opt) as "OFF".

Oh, sorry for this, here is just the default configuration, the default configuration is for memory optimization, but if there is enough memory, all `xxx_IRAM_OPT` options should be turned on as much as possible, especially for scenarios such as WiFi that require high accuracy of wake-up time point during sleep.

> Would it be worth trying different wifi power modes to see if that has something to do with it or if it is just related to DFS?

all ps modes will wake up the chip, but the wake-up frequency is different. It should not be the root cause, but it will affect the probability of recurrence.

esp-wzh, 2024-12-02:
> Is there anything I can do or enable to make sure my external crystal is also performing correctly? And make sure that is not causing any issues?

Hi, you can use the `esp_clock_output feature` to monitor the XTAL32K clock, just add the following code:
```
#include "esp_clock_output.h"
esp_clock_output_mapping_handle_t clkout_mapping_hdl;
ESP_ERROR_CHECK(esp_clock_output_start(CLKOUT_SIG_RC_SLOW, IO_NUM, &clkout_mapping_hdl));
```
Note: Only GPIO_NUM_18 or GPIO_NUM_19 or GPIO_NUM_20 is supported for esp32s3's clock output feature.

esp-wzh, 2024-12-24:
`ESP_PD_DOMAIN_XTAL`  controls the power domain of the main 40MHz XTAl and does not affect the 32kHz XTAL, which is reasonable.

Does the image show the waveform at the time (or after) the crash occurred?

BTW, can you identify the scenario in which the crash occurs? WiFi only, BLE only, WIFi BLE coexist, or is it possible to crash in all three scenarios?

esp-wzh, 2024-12-24:
It is difficult to debug without a reproducible environment... Can you report the issue [here](https://www.espressif.com/en/contact-us/technical-inquiries/software-bug-report) and provide a reproducible environment?

esp-wzh, 2024-12-25:
I think this fix will not fix this problem because the assert: `configASSERT( ( xTickCount + xTicksToJump ) <= xNextTaskUnblockTime );` that caused the problem still exists

esp-wzh, 2024-12-26:
Found  bug, but I'm not sure if it's related with the assert failed in vTaskStepTick.
Note the abort in `Panic reason: assert failed: esp_sleep_pd_config sleep_modes.c:2118 (refs >= 0)`, which is called from vApplicationSleep, if `CONFIG_PM_TRACE` is enabled, it will call `esp_sleep_pd_config` to force `RTC_PERIPH` keep power on in sleep, in the implementation of `esp_sleep_pd_config`, it will increase the ref_cnt of `RTC_PERIPH`, which is a  variable of `int16_t`, as the auto lightsleep continues, this value will eventually overflow and trigger this positive assertion.

So can you try disabling option `CONFIG_PM_TRACE`? @michaelboeding

esp-wzh, 2025-01-03:
Thanks for feedback @michaelboeding , the fix for `CONFIG_PM_TRACE` is merged in internal repo and will sync to github release soon.

ztefanjo, 2025-01-07:
@esp-wzh / @michaelboeding , good that you find the problem with CONFIG_PM_TRACE, but that was not the original issue that I reported as we do not have CONFIG_PM_TRACE enabled. As written, for us, the problem happens much less frequently after we increased CONFIG_FREERTOS_IDLE_TIME_BEFORE_SLEEP from two to three, but it is still happening occasionally. Attached is a CoreDump reported on v5.1.4, very similar to the original reported dump.
[core-dump_51.1_Wifi-BLE-App_67.18_241226-085342.elf.txt](https://github.com/user-attachments/files/18330487/core-dump_51.1_Wifi-BLE-App_67.18_241226-085342.elf.txt)

esp-wzh, 2025-01-07:
@ztefanjo  Can you share your sdkconfig?

ztefanjo, 2025-01-07:
@esp-wzh : 
[sdkconfig.defaults.txt](https://github.com/user-attachments/files/18331036/sdkconfig.defaults.txt)

esp-wzh, 2025-01-08:
> @esp-wzh : [sdkconfig.defaults.txt](https://github.com/user-attachments/files/18331036/sdkconfig.defaults.txt)

Looks like this isn't the final sdkconfig file, if you enabled the `CONFIG_ESP_PHY_MAC_BB_PD`, you can try this [patch](https://github.com/user-attachments/files/17967968/IDFGH-8319.patch.zip).

If not, you can try the method mentioned in this [thread](https://github.com/espressif/esp-idf/issues/9797#issuecomment-2482660664), especially the 3rd and 6th, increase `min_freq_mhz` in `esp_pm_configure` and don't let the chip sleep for too long at a time.

And you can check the startup log at the time when the problem occurred, make sure that no `32 kHz XTAL not found` warning appears.

ztefanjo, 2025-01-13:
@esp-wzh : `CONFIG_ESP_PHY_MAC_BB_PD` is not enabled. We are not sleeping for very long, a timer is waking up the system continuously. We could increase the `min_freq_mhz`, but considering how seldom this error happens, I think we prefer holding the current consumption to a minimum.

sudeep-mohanty, 2025-01-24:
Fix merged at 35cdecb1214c775ef2274280a6dbf05c28e811da. Closing this issue. Please feel free to reopen this issue or open a new one for further support.

closed 2025-01-24 by sudeep-mohanty