espressif/esp-idf #9798: NimBLE extended advertising example does not work

Hoiss, 2022-09-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### IDF version.
v4.4
### Operating System used.
Linux
### Power Supply used.
USB
### SoC
ESP32-C3
### Used example
examples/bluetooth/nimble/bleprph
### What is the expected behavior?
I expect when using the official example that setting the define CONFIG_EXAMPLE_EXTENDED_ADV is enough to test extended advertising. It isn't, you will run into several kernel exceptions without code modifications (see Steps To Reproduce).
Fixing the problems doesn't really help, still no BLE packets are transmitted, contrary to the same example without the define CONFIG_EXAMPLE_EXTENDED_ADV. 
This means the standard advertising works, but not the extended one.
### Steps to reproduce.
Take the official example and add the following in the beginning:
`#define CONFIG_EXAMPLE_EXTENDED_ADV 1`
Also ensure the extended advertising is activated in the sdkconfig!

But compiling now won't work. 
With some debugging it seems you need to change part of ext_bleprph_advertise function:
Remove `params.connectable = 1;
    params.scannable = 1;
   params.legacy_pdu = 1;` or set them to 0 (done by the memset before anyway, so a remove is fine)

But here I get stuck. It compiles and runs without complains, but I can't see the see bluetooth advertisement (with a BLE5 device). Using this example without extended advertisement works.

Please help to get extended advertising to work.

rahult-github, 2022-09-17:
Hi @Hoiss  ,

 Extended ADV support in nimble stack is not enabled by default at compilation time for ESP32-C3. Even though the example configuration has option to enable extended adv , but it is under a flag. Guess we can update the help / documentation to make it more clear. 

For your issue, i suggest to also enable the support for "Ext Advertising" in stack available at  "Component config → Bluetooth → NimBLE Options → Enable BLE 5 feature -> Enable Extended advertising "

This above setting, coupled with the example flag being enabled, should then help your case. Please give it a try.

Hoiss, 2022-09-17:
Hi @rahult-github ,

> i suggest to also enable the support for "Ext Advertising" in stack available at "Component config → Bluetooth → NimBLE Options → Enable BLE 5 feature -> Enable Extended advertising "

this is what I mean with 
> Also ensure the extended advertising is activated in the sdkconfig!

Without activating this option you won't be able to compile, so that seems not to be the root of this issue.

rahult-github, 2022-09-19:
Hi @Hoiss ,

I tried a few things at my end and so now have some observations and queries.

Your initial post mentions IDF version is 4.4 .  Can you please confirm the version ? The flag  CONFIG_EXAMPLE_EXTENDED_ADV doesn't exist on v4.4   https://github.com/espressif/esp-idf/blob/v4.4.2/examples/bluetooth/nimble/bleprph/main/Kconfig.projbuild 

If you are enabling this flag, are you modifying the application too? If yes, please do share such information also depicting what changes you did .

The change is present on release/v5.0 . So, on release/v5.0   I tested using bleprph and blecent and confirmed the functionality works ok. Attaching debug logs for same . I also tested bleprph against nrf connect application on mobile ( a third party phone application and able to see the device and connect ) .

On bleprph side, i enabled debug logs and able to see below commands: 

208:D (1261) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0036 len=25
240:D (1341) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0037 len=33
280:D (1441) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x0039 len=6

These are commands for ext adv Parameters, ext adv data and enabling ext adv 

On blecent side, i am able to see the device :  

 D (4014) NimBLE: props=19 data_status=0 legacy_event_type=0
 D (4014) NimBLE: address = 7c:df:a1:a2:9b:f2
 D (4014) NimBLE: rssi=-35 tx_power=127
 D (4014) NimBLE: sid=255 prim_phy=1 sec_phy=0
 D (4014) NimBLE: periodic_adv_itvl=0 length_data=29
 

So , from above, i think the functionality does work ok. But let's try to find what is failing at your end.

1. What is the IDF version you are using ?
2. Can you share with me the output of "git describe" and the first commit you observe from output of "git log".
3. Can you please share the code for the modification you are attempting to do ? I am able to cleanly compile on release/v5.0 branch as of today with the flags enabled from menuconfig. 
4. I assume "git submodule update --recursive --init -f " has been done successfully.
5. If issue is still seen , can you please enable debug logs on bleprph side and share console log. Steps to enabled debug : 
    a. Component Config  --> Log Output --> Default Log Verbosity --> Debug 
    b. Component Config --> Bluetooth --> NimBLE Options --> NimBLE Host log verbosity --> Debug Log

6. Do you have any OTA logs to confirm that over the air ESP32C3 is NOT sending ext adv reports despite being set from host correctly ? 

[blecent_console.txt](https://github.com/espressif/esp-idf/files/9596748/blecent_console.txt) 
[bleprph_console.txt](https://github.com/espressif/esp-idf/files/9596749/bleprph_console.txt)

Hoiss, 2022-09-21:
Hi @rahult-github,

you're absolutely right, sorry. I accidentally checked an older installation idf. And one of the eclipse paths was additionally still attached to the old one. Cost me some time to identify the problem - thank you for your hints.
Used is ESP-IDF v5.0-beta1 in eclipse.
All the modifications in the main.c are described in the starting post, but see below because of changes.

I started a new project again with above mentioned version, compiled and flashed it: OK, I'm receiving the normal advertising packets.
Then I activated the "Enable extended advertising" option in the sdkconfig (according to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html this is equivalent to the menuconfig command that you mentioned) compiled and flashed again: Running into a `NimBLE: error setting advertisement data; rc=8` error. 
-> The preprocessor somehow doesn't know about the `CONFIG_EXAMPLE_EXTENDED_ADV` and therefore I additionally have to add `#define CONFIG_EXAMPLE_EXTENDED_ADV 1` to the beginning of main.c.
That's a bit confusing, maybe I miss a step to refresh the resources or similar?

Anyway, the example works now. Thank you for the support.

closed 2022-09-22 by rahult-github

Hoiss, 2022-09-22:
@rahult-github, maybe you could add a comment why it's necessary to manually set CONFIG_EXAMPLE_EXTENDED_ADV in the example if using sdkconfig. It seems it isn't necessary using menuconfig, is it?

rahult-github, 2022-09-28:
Hi @Hoiss ,

 I missed replying to your comment. I think, it may be some setup issue at your end. I tried a fresh eclipse setup at my end and did below steps.
1. Import project after starting eclipse.
2. IDF v5.0 was used
3. Opened eclipse terminal. 
4. Enabled the example flag via menuconfig
5. Compiled and executed. 
6. Able to see the feature working. 

In order to double confirm, i placed one print in the ext_bleprph_advertise function, and during my run i observed the print coming. Thereby confirming, that flag is getting enabled. 

[bleprph.txt](https://github.com/espressif/esp-idf/files/9662349/bleprph.txt)

Thanks,
 Rahul

Hoiss, 2022-12-14:
@rahult-github, I need to ask another question, because I'm running into problems sending more than ~30 bytes within the **EXTENDED advertising** ext_adv_pattern_1 array.
Just try to use

> static uint8_t ext_adv_pattern_1[] = {
    0x02, 0x01, 0x06,
    0x03, 0x03, 0xab, 0xcd,
    0x03, 0x03, 0x18, 0x11,
    0x14, 0X09, 'n', 'i', 'm', 'b', 'l', 'e', '-', 'b', 'l', 'e', 'p', 'r', 'p', 'h', '-', 'e','1','2','3',
};
which means to add 3 characters to the devices name and increase the length to 0x14 because of that.

It results in a return code of 3 for the function ble_gap_ext_adv_set_data, which means BLE_HS_EINVAL (One or more arguments are invalid)
The original value without the added `'1','2','3',` works.

What am I doing wrong, how can I send more data (e.g. also the added full 255 byte with the Manufacturer Specific type 0xFF) ?

rahult-github, 2022-12-15:
Hi @Hoiss 

 `What am I doing wrong, how can I send more data (e.g. also the added full 255 byte with the Manufacturer Specific type 0xFF) ?`

Example demo code has legacy_pdu set to 1 . Hence the max size in this case becomes 31 ( and not 251). 

The demo example code can be modified to include more information as per your requirement. You can refer to the BT specification and look for detailed information for  LE Set Extended Advertising Data command  ( and also LE Set Extended Advertising Parameters command) that explains the different combinations that are possible.  As part of IDF headers, basically the structure is defined to capture all parameters and one sample usage in example is provided, but user should be able to set the desired fields in the application to get the expected behaviour.

Thanks,
 Rahul

Hoiss, 2022-12-15:
Thank you for your help again @rahult-github , this helped me to get it to work. I really appreciate the fast feedback.
The BT spec is a lot to read, therefore it was good to know where to look at in the beginning.

As a proposal: The flags could be set different according to the CONFIG_EXAMPLE_EXTENDED_ADV setting in the example. It's easier for people starting with your examples then, because I think most people trying an extended advertising example also want to send more then the 31 bytes of a standard advertising packet.

Thanks

rahult-github, 2022-12-16:
Hi @Hoiss 

 Thank you for the valuable feedback. We will check internally and would work to include this suggestion .

Thanks.

ujur007, 2025-03-20:
I am facing the same/similar issue.
I am using Nimble and for me advertising with extended advertisement failed. The parameters that I set are the following. The return code is 3 (invalid parameters). What am I doing wrong? 

   ```
 struct ble_gap_ext_adv_params params;
    
    params.legacy_pdu = 1;
    //params.sid = 3;
    params.connectable = 1;
    params.scannable = 1;
    params.directed = 0;
    params.high_duty_directed = 0;
    params.own_addr_type = BLE_OWN_ADDR_PUBLIC;
    params.primary_phy = BLE_HCI_LE_PHY_1M;
    params.secondary_phy = BLE_HCI_LE_PHY_1M;
    //params.tx_power = 9;
    params.include_tx_power=1;
    memset(&params, 0, sizeof(params));
    rc = ble_gap_ext_adv_start(instance, duration, 0);
```

rahult-github, 2025-03-20:
Hi @ujur007  ,  the provided information is not sufficient to identify what the issue is . I suggest : 

1. Refer to the example code [here](https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/nimble/bleprph/main/main.c#L89)  which shows the steps needed to perform extended adv. 

2. If even after doing same steps, you can't get it working, request you to share your application code and sdkconfig.