espressif/esp-idf #9799: ESP IDF timer example code

ChethiyaW, 2022-09-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Symbol 'TIMER_BASE_CLK' could not be resolve error occur when this code build

ESP-Marius, 2022-09-18:
Hi,

Can you describe in a bit more details what you are trying to do? 

E.g. which IDF version are you using? Have you made any changes to the example code?

I see `TIMER_BASE_CLK` was removed in IDF 5.0, are by any change trying to build example code from an earlier version of IDF using v5.0 or later?

suda-morris, 2022-09-19:
Hi @ChethiyaW  `TIMER_BASE_CLK` has been removed since esp-idf 5.0 because that macro is useless. Timer group clock sources can have multiple choices: APB, XTAL, etc. So if you want to fetch the frequency of the clock source, please call APIs e.g. `esp_clk_apb_freq`.

ChethiyaW, 2022-09-20:
Thanks for the tip.