espressif/esp-idf #9800: BLE client disconnection after 30 seconds

fabiolagatta, 2022-09-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

No disconnection event 

### What is the actual behavior?

I receive an unexpected disconnection event at 30 seconds from connection. Why?

I (1448) GATTC_DEMO: searched device Galaxy A22 5G

I (1458) GATTC_DEMO: connect to the remote device.
I (1458) GATTC_DEMO: 5f 98 49 85 cc fd 
I (1468) GATTC_DEMO: searched Adv Data Len 31, Scan Response Len 0
I (1468) GATTC_DEMO: searched Device Name Len 0
I (1478) GATTC_DEMO: 

I (1488) GATTC_DEMO: stop scan successfully
I (2578) GATTC_DEMO: ESP_GATTC_CONNECT_EVT conn_id 0, if 1
I (2578) GATTC_DEMO: REMOTE BDA:
I (2578) GATTC_DEMO: 76 9b 2c c7 7a 6b 
I (2578) GATTC_DEMO: open success
I (3448) GATTC_DEMO: discover service complete conn_id 0
I (3448) GATTC_DEMO: SEARCH RES: conn_id = 0 is primary service 1
I (3448) GATTC_DEMO: start handle 40 end handle 44 current handle value 40
I (3458) GATTC_DEMO: service found
I (3458) GATTC_DEMO: UUID16: 180f
I (3458) GATTC_DEMO: SEARCH RES: conn_id = 0 is primary service 1
I (3468) GATTC_DEMO: start handle 53 end handle 65535 current handle value 53
I (3478) GATTC_DEMO: service found
I (3478) GATTC_DEMO: UUID16: 180f
I (3488) GATTC_DEMO: Get service information from remote device
I (3488) GATTC_DEMO: ESP_GATTC_SEARCH_CMPL_EVT
I (3498) GATTC_DEMO: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 500, conn_id 0
I (3508) GATTC_DEMO: ESP_GATTC_REG_FOR_NOTIFY_EVT
I (3548) GATTC_DEMO: write descr success 
**W (32618) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13
W (32618) BT_APPL: gattc_conn_cb: if=1 st=0 id=1 rsn=0x13**
I (32618) GATTC_DEMO: ESP_GATTC_DISCONNECT_EVT, reason = 19

### Steps to reproduce.

[gattc_demo.zip](https://github.com/espressif/esp-idf/files/9586458/gattc_demo.zip)

### Debug Logs.

```plain
W (32618) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x13
W (32618) BT_APPL: gattc_conn_cb: if=1 st=0 id=1 rsn=0x13
I (32618) GATTC_DEMO: ESP_GATTC_DISCONNECT_EVT, reason = 19
```

chegewara, 2022-09-16:
https://github.com/chegewara/esp32-ble-wiki/issues/5
> 0x13 Remote User Terminated Connection

It looks like peer device is requiring some sort of authentication (pin, Yes/No), and is disconnecting after 30 seconds when its not authenticated.

fabiolagatta, 2022-09-17:
Hi,
I think not. I'm using an android APP peripheral simulator as BLE server (https://play.google.com/store/apps/details?id=io.github.webbluetoothcg.bletestperipheral&hl=it&gl=US) and this APP  require no authentication.

Please...give me some suggestions

esp-cjh, 2022-09-19:
Hi @fabiolagatta,

I use ESP32C3 as BLE slave and ESP32C3 as BLE master (use your source code) and the problem did not occur. So there may be some compatibility issues in your phone or app.

Could you please capture the air interface packets and provide the capture file? You can use ellisys , nrf sniffer or other capture tools?

fabiolagatta, 2022-09-19:
Hi my friend,
I have attached the log captured with wireshark where you can see everything that happens.
At no. 12869 you can see the disconnection message which shouldn't be there
Thanks for your important help

[connessione espressif.zip](https://github.com/espressif/esp-idf/files/9597789/connessione.espressif.zip)

onurdof, 2022-11-17:
I have same problem when I am using classic bluetooth spp_acceptor example . I cant make connection between pc and esp32. 
:(

NaveenKoyyada, 2023-01-05:
Hi, I am new to this community. 
Past few weeks I am working on ESP32 board. I have taken BLE mesh example code for generic server on off module.

When I trying to operate from App, the device is working fine but some times the following error is seen in the log.
[image]

So I looking for root cause for this error.
Thankyou in advance!

forx157, 2024-10-24:
Hi, @NaveenKoyyada,

`"Proxy SAR timeout"` in the log indicates that the node did not receive a complete Mesh packet from the proxy client within the specified time. 

Do you have the complete log? Could you please introduce your test environment and testing method?