espressif/esp-idf #9802: Wanting to understand variance in ping response time of ESP32

mbratch, 2022-09-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have two separate applications that use the ESP32 Pico DevKit v4.1 and I'm using, currently, ESP-IDF 4.4.2.

I noticed sometimes ping response was a bit slow, over 100ms, but wanted to see if there was some pattern in the performance of ping since my applications have a couple of FreeRTOS software timers running. Running a once-per-second ping, here's what I observed with the two applications:

[image]

[image]

Application 1 has two software timers running. One just flashes an LED at a 1Hz rate, so it is a 500ms timer. The other shows some simple state information via other LEDs, and updates that state every 2 seconds. So it's a 2s timer.

Application 2 has the same two software timers as application 1, but also has eight independent 25ms timers each independently scanning a single, separate I2C ADC channel. So application 2 is much more intensive.

Regardless, I was curious about the pattern that I see in the ping performance. It has a cadence of increased ping latency over a period of 4-5s then back to normal latency. I don't know that I can explain this "sawtooth" performance pattern as being a result of the software timers. Has anyone else observed this kind of behavior? I'm not calling this a "defect", but would just like to understand it, if anyone has some ideas.

readmodifywrite, 2022-09-17:
That looks like modem sleep with a DTIM interval of 1.

mbratch, 2022-09-17:
@readmodifywrite interesting. Sounds logical. I have the wifi power save turned off, though.

readmodifywrite, 2022-09-17:
If modem sleep is off IE esp_wifi_set_ps(WIFI_PS_NONE), then yeah that's a bit odd.  Usually I'm getting just a few milliseconds in that configuration.

Are you sure it's actually off?  I'm not sure what else it could be.  Those ping timings are exactly what mine look like with modem sleep enabled (when it is working correctly - I have an issue open examining interesting ways to break it).

mbratch, 2022-09-17:
@readmodifywrite I know I'm calling `esp_wifi_set_ps(WIFI_PS_NONE)` and that doing so did resolve some issues I had some time ago with http request response latency. Maybe I'll tinker with it and comment out that line to see if I get a different behavior.

mbratch, 2022-09-17:
@readmodifywrite ok.... this whole thing was me being daft. Yes, I was calling `esp_wifi_set_ps(WIFI_PS_NONE)` but I was calling it too soon in the initi process so it had no effect. Duh. I still think the sawtooth behavior is interesting, though, even assuming ps is on. :)

closed 2022-09-21 by mbratch

readmodifywrite, 2022-09-21:
That init sequencing has gotten me before too.  Glad you got it worked out, cheers!

cy18, 2023-04-09:
> @readmodifywrite ok.... this whole thing was me being daft. Yes, I was calling `esp_wifi_set_ps(WIFI_PS_NONE)` but I was calling it too soon in the init process so it had no effect. Duh.

Thank you for your sharing. Same problem just occors to me. Espressif should put it in the document.