espressif/esp-idf #9804: Esp32s2 Build fails with: ninja: build stopped: subcommand failed

skickar, 2022-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf4.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

I am exactly following the setup documentation for MacOS and trying the default HelloWorld example.

I have an ESP32s2 chip plugged in and in flashing mode. When I run:
` idf.py -p /dev/cu.usbmodem01 -b 460800 flash`

I expect it to build and flash the helloworld example. 

### What is the actual behavior?

When I run the command, this is the end of the output:

`WARNING: ESP32-S2FNR2 chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
CMake Error at run_serial_tool.cmake:56 (message):
  /Users/skicka/.espressif/python_env/idf4.4_py3.10_env/bin/python
  /Users/skicka/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip
  esp32s2 failed

FAILED: CMakeFiles/flash /Users/skicka/esp/hello_world/build/CMakeFiles/flash 
cd /Users/skicka/esp/esp-idf/components/esptool_py && /usr/local/Cellar/cmake/3.24.2/bin/cmake -D IDF_PATH="/Users/skicka/esp/esp-idf" -D SERIAL_TOOL="/Users/skicka/.espressif/python_env/idf4.4_py3.10_env/bin/python /Users/skicka/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32s2" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/Users/skicka/esp/hello_world/build" -P /Users/skicka/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

To troubleshoot, I have also tried selecting "only most common certificates" in the menu, and disabling certificates entirely. There is no effect.

After the failed flashing, the esp32s2 will not show up as a serial device unless it's put into flash mode.

### Steps to reproduce.

1. Install Espressif IDF on MacOS
2. Select the default HelloWorld example
3. Run `idf.py -p /dev/cu.usbmodem01 -b 460800 flash`
4. Build fails with:
`ninja: build stopped: subcommand failed.
ninja failed with exit code 1`

### Build or installation Logs.

```plain
skicka@Tammy-Napkins hello_world % idf.py -p /dev/cu.usbmodem01 -b 460800 flash
Executing action: flash
Running ninja in directory /Users/skicka/esp/hello_world/build
Executing "ninja flash"...
[0/1] Re-running CMake...
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s2
-- Project sdkconfig file /Users/skicka/esp/hello_world/sdkconfig
-- App "hello_world" version: 1
-- Adding linker script /Users/skicka/esp/hello_world/build/esp-idf/esp_system/ld/memory.ld
[... 114 lines cut ...]
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 501.4 kbit/s)...
Hash of data verified.

Leaving...
WARNING: ESP32-S2FNR2 chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.
CMake Error at run_serial_tool.cmake:56 (message):
  /Users/skicka/.espressif/python_env/idf4.4_py3.10_env/bin/python
  /Users/skicka/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip
  esp32s2 failed

FAILED: CMakeFiles/flash /Users/skicka/esp/hello_world/build/CMakeFiles/flash 
cd /Users/skicka/esp/esp-idf/components/esptool_py && /usr/local/Cellar/cmake/3.24.2/bin/cmake -D IDF_PATH="/Users/skicka/esp/esp-idf" -D SERIAL_TOOL="/Users/skicka/.espressif/python_env/idf4.4_py3.10_env/bin/python /Users/skicka/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32s2" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/Users/skicka/esp/hello_world/build" -P /Users/skicka/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

### More Information.

I've tried the fixes mentioned here: https://github.com/espressif/esp-idf/issues/5322#issuecomment-962219803

It was not effective.

dobairoland, 2022-09-19:
> To troubleshoot, I have also tried selecting "only most common certificates" in the menu, and disabling certificates entirely. There is no effect.

> I've tried the fixes mentioned here: https://github.com/espressif/esp-idf/issues/5322#issuecomment-962219803

Hi @skickar. This has nothing to do with certificates.

The issue is explained in the output message:

> WARNING: ESP32-S2FNR2 chip was placed into download mode using GPIO0.
esptool.py can not exit the download mode over USB. To run the app, reset the chip manually.
To suppress this note, set --after option to 'no_reset'.

If you put the chip into download mode manually then `idf.py flash` cannot reset it. You need to reset it manually after flashing and then you can connect to it by `idf.py monitor`. There is really no issue here.

dobairoland, 2022-09-19:
BTW, I forgot to mention that putting the chip into download mode manually is not necessary. `idf.py flash` is able to do that automatically and then you will not receive the above error.

Note that `idf.py monitor` using the same port will work only if you've built and USB console application. Otherwise, the same port won't show up after reset.

closed 2022-09-19 by dobairoland

francudina, 2023-11-02:
@skickar I found out for this link, it helped me with the flashing: [link](https://github.com/espressif/esp-idf/issues/11035#issuecomment-1479471954)