espressif/esp-idf #9806: Vendor Server Model does not receive a publish message from a Vendor Client node

djbs291, 2022-09-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello guys, I am trying to publish a message using the API esp_ble_mesh_model_publish(model, opcode, sizeof(data), data, ROLE_NODE) into a group, but the node subscribing it is not receiving the message. The publication is done with a client vendor model I created. The role defined is a Node.

To do that publication in the client node, I set a esp_ble_mesh_model_t model using the esp_ble_mesh_find_vendor_model api, and then manually set the model->pub->publish_addr with the group address I defined (0xC001).

[Codebox]    esp_ble_mesh_model_t *model = NULL;
    uint32_t opcode = ESP_BLE_MESH_VND_MODEL_OP_TEMPERATURE_ACTUATOR_SET;
    uint8_t data[3] = {};
    data[DATA_TEMPERATURE_POSITION] = (uint8_t)&dht_state[0].temperature;
    
    ESP_LOGI(TAG,"Inside gen_onoff_set_handler()");

    model = esp_ble_mesh_find_vendor_model(&elements[ELEMENT_INDEX],CID_ESP, ESP_BLE_MESH_VND_MODEL_ID_TEMPERATURE_CLIENT);
    if (!model) {
        return ESP_ERR_INVALID_ARG;
    }
    
    // Add the Group address into the model publish_addr in which the message
    // will be published 
    model->pub->publish_addr = TEMPERATURE_GROUP_ADDRESS;

....

err = esp_ble_mesh_model_publish(model, opcode, sizeof(data), data, ROLE_NODE);[/Codebox]

In the server node, the subscribing is done inside ESP_BLE_MESH_MODEL_OP_MODEL_APP_BIND case, in the callback function for configuration server:

[Codebox]
        case ESP_BLE_MESH_MODEL_OP_MODEL_APP_BIND:
            ESP_LOGI(TAG, "ESP_BLE_MESH_MODEL_OP_MODEL_APP_BIND");
            ESP_LOGI(TAG, "elem_addr 0x%04x, app_idx 0x%04x, cid 0x%04x, mod_id 0x%04x",
                param->value.state_change.mod_app_bind.element_addr,
                param->value.state_change.mod_app_bind.app_idx,
                param->value.state_change.mod_app_bind.company_id,
                param->value.state_change.mod_app_bind.model_id);

                esp_err_t err = esp_ble_mesh_model_subscribe_group_addr(param->value.state_change.mod_app_bind.element_addr,
                                                                      param->value.state_change.mod_app_bind.company_id,
                                                                      param->value.state_change.mod_app_bind.model_id,
                                                                      TEMPERATURE_GROUP_ADDRESS);
                if(err != ESP_OK){
                    ESP_LOGE(TAG, "Failed to subscribe the Group Address");
                }
                
            break;
[/Codebox]

When I try to publish the messages in the client node, the  ESP_BLE_MESH_MODEL_PUBLISH_COMP_EVT is not generated, according to serial logs:
[Codebox]
I (171980) DHT: model->pub->publish_addr = 49153
I (171980) ESP: Inside ble_mesh_model_send_msg
I (171990) ESP: Inside bt_mesh_model_msg_init
I (171990) ESP: Inside net_buf_simple_add_mem
I (172000) ESP: ble_mesh_model_send_msg: device_role = 0
I (172000) ESP: Inside btc_transfer_context
I (172010) ESP: Inside if (arg)
I (172010) ESP: Inside btc_task_post
I (172020) ESP: btc_task_post: return BT_STATUS_SUCCESS
I (172020) ESP: ble_mesh_model_send_msg status: 0
I (172030) DHT: Publish success. 0
[/Codebox]

Am I missing any step to proceed with the publication.

JiaLWang, 2022-09-20:
Hi @djbs291 
Can you provide the complete log？ `I (172030) DHT: Publish success. 0` only means the message sent to BTC task successfully. 
In your client model side, do you add the cb handler of `ESP_BLE_MESH_MODEL_PUBLISH_COMP_EVT` event?

djbs291, 2022-09-21:
The ESP_BLE_MESH_MODEL_PUBLISH_COMP_EVT  event handling was defined in the following way:

```   
static void esp_ble_mesh_model_cb(esp_ble_mesh_model_cb_event_t event,
                                  esp_ble_mesh_model_cb_param_t *param)
{
  case ESP_BLE_MESH_MODEL_PUBLISH_COMP_EVT:
          ESP_LOGI(TAG, "ESP_BLE_MESH_MODEL_PUBLISH_COMP_EVT, err_code %d",
                   param->model_publish_comp.err_code);
          if (param->model_publish_comp.err_code) {
              ESP_LOGE(TAG, "Failed to publish message 0x%06x", param->model_publish_comp.opcode);
              break;
          }
          break;

   ...
}

...

static esp_err_t ble_mesh_init(){
  ...
  esp_ble_mesh_register_custom_model_callback(esp_ble_mesh_model_cb);
  ...
}

```

JiaLWang, 2022-09-22:
Hi @djbs291 
After the log that you already give above, is there any error or exception information?

djbs291, 2022-09-24:
Hi @JiaLWang,
The problem was solved.  This was caused by a while(1) loop, that I was using to request data to a temperature sensor, and the publish requests were being done inside that while(1). The while(1) instructions proceeded without giving time to trigger ESP_BLE_MESH_MODEL_PUBLISH_COMP_EVT.