espressif/esp-idf #9808: Coredump when BLE disconnects while writing - gatt_cb.p_tcb_list not protected

ztefanjo, 2022-09-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

To be able to read/write BLE data without crashes while the peer disconnects the connection.

### What is the actual behavior?

The attached coredump shows a scenario where an application thread is crashing in gatt_get_tcb_by_idx called by esp_ble_gattc_write_char at the same time that the BTC_TASK is finalizing a close.
[core-dump_51.1_close_and_write.elf.txt](https://github.com/espressif/esp-idf/files/9596709/core-dump_51.1_close_and_write.elf.txt)

When reviewing the code, it seems that the gatt_cb.p_tcb_list is not protected by any mutex or similar construction. It can also be seen that the gatt_tcb_free function is called before any DISCONNECT_EVT or CLOSE_EVT is sent to the application, so it is very hard for the application to protect itself from this problem.

### Steps to reproduce.

Theoretically:
1. Read/Write BLE data
2. Let peer disconnect
3. Repeat 1-2 until crash

Note: It is not clear in the core dump if it was actually the peer that disconnected or if it was the application itself that requested the disconnect while writing. But from what I can see by code review, this error can occur if the peer disconnects.

zztiger123, 2022-10-10:
Hi [ztefanjo](https://github.com/ztefanjo),

May we know more information? For example :
1. If it can be reproduced with an simple code snippet?
2. Is there a corresponding log?

ztefanjo, 2022-10-10:
Hi @zztiger123,
1. Sorry, this exact core dump has only occurred once so far, but reviewing the code, it seems plausible that the crash was due to the lack of mutex protection of the p_tcb_list. We do however get frequent core dumps indicating heap corruption. This may be related. 
2. No, unfortunately not. We do collect also logs from our test fleet, but when the MCU crashes, there is no time for the controlling node to read them out.

esp-cjh, 2024-02-27:
Hi, @ztefanjo This bug has been fixed (see e748546365468d0f0efdeadd31b84bf9e06b0d98). Thanks for your report.

closed 2024-02-27 by Alvin1Zhang