espressif/esp-idf #9809: how to remove a node was provisioner by esp ble mesh

xuanchibkd14, 2022-09-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

hi, im trying to know remove a node from my provisioner, but i cant find how to do it,
like that:
~~~
I (576) BLE_MESH: Restored Primary Address 0x0001, next address alloc 0x0006
I (584) BLE_MESH: Restored next NetKeyIndex alloc 0x001
I (589) BLE_MESH: Restored next AppKeyIndex alloc 0x000
I (596) BLE_MESH: Restored NetKeyIndex 0x000
I (600) BLE_MESH: Restored NetKey d4b89072ca3945b78555c498ceab9784
I (609) BLE_MESH: Restored AppKeyIndex 0x000, NetKeyIndex 0x000
I (614) BLE_MESH: Restored AppKey d7e55176d0c14de9a6c3648ce84de1bc
I (622) BLE_MESH: Restored Node 0x0005, UUID 246f2875efe2414e3232303832333032
I (629) BLE_MESH: Restored Node 0x0005, Name NODE-00
I (634) BLE_MESH: Restored Node 0x0005, Composition Data e502000000000a000200000001010000e5020115
~~~
how to remove node `0x0005, Name NODE00`?
thank for helps

JiaLWang, 2022-09-20:
Hi @xuanchibkd14 
You can use `esp_ble_mesh_config_client_set_state`function to send `config node reset` message to reset a node(other than a provisioner) and remove it from the network.
Hope it can help you.

xuanchibkd14, 2022-10-07:
> 

Can you explain in more detail?

forx157, 2022-10-08:
the `esp_ble_mesh_config_client_set_state` has been used  in many examples such as  " ble_mesh_sensor_model" and "ble_mesh_vendor_model"  , you can refer to these examples to get more information, and the opcode named `ESP_BLE_MESH_MODEL_OP_NODE_RESET` was defined in the `esp_ble_mesh_defs.h` is used to reset a node.

xuanchibkd14, 2022-10-08:
i founded ```esp_ble_mesh_provisioner_delete_node_with_uuid(node->dev_uuid);``` , it is exactly i need, tks for help

closed 2022-10-11 by xuanchibkd14

xd3011, 2024-06-20:
Hello. I also have the same problem. Using esp_ble_mesh_provisioner_delete_node_with_uuid(node->dev_uuid) from provisioner to reset the node with uuid as node->dev_uuid. Does it work?