espressif/esp-idf #9810: [esp32s3] Backport internal temperature sensor from master to 4.4.2

KonssnoK, 2022-09-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

As per title.

In 4.4.2 you have the internal temperature sensor example but the driver code doesn't exist.
It does exist in master (and 5.0) thou.

mythbuster5, 2022-09-19:
Thanks, I think temperature sensor is already available on v4.4.2 ? (https://github.com/espressif/esp-idf/blob/v4.4.2/components/driver/esp32s3/rtc_tempsensor.c)

KonssnoK, 2022-09-19:
aha, so the example needs update? 🤔

mythbuster5, 2022-09-19:
Yeah, I think readme should be updated, code can be compiled successfully right?

KonssnoK, 2022-09-19:
yes i just compiled it.
Thanks!
Should i close or we rename to update the example?

mythbuster5, 2022-09-19:
you can close it, I will find time to update the backport

closed 2022-09-19 by KonssnoK