espressif/esp-idf #9812: idf repo is always dirty

albertyue, 2022-09-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.1 v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### What is the expected behavior?

after i unzipped the zip file, `git status` command should tell this worktree is clean.

### What is the actual behavior?

after execute `git status`, i got this message:
``` 
当前不在任何分支上。
尚未暂存以备提交的变更：
  （使用 "git add <文件>..." 更新要提交的内容）
  （使用 "git checkout -- <文件>..." 丢弃工作区的改动）
  （提交或丢弃子模组中未跟踪或修改的内容）

        修改：     components/openthread/openthread (修改的内容)

修改尚未加入提交（使用 "git add" 和/或 "git commit -a"）
```

then i execute `git status` in `components/openthread/openthread`, output message is shown as follow:
``` 
头指针分离于 c36c0e7
尚未暂存以备提交的变更：
  （使用 "git add <文件>..." 更新要提交的内容）
  （使用 "git checkout -- <文件>..." 丢弃工作区的改动）

        修改：     third_party/mbedtls/repo/scripts/data_files/vs6-app-template.dsp
        修改：     third_party/mbedtls/repo/scripts/data_files/vs6-main-template.dsp
        修改：     third_party/mbedtls/repo/scripts/data_files/vs6-workspace-template.dsw
``` 

### Steps to reproduce.

1. unzip the zip file
2. enter ief directory
3. excute `git status` or `git describe --dirty`, you will find out the worktree is dirty.

Dazza0, 2022-09-19:
@albertyue could you point to what zip file you are referring to?

albertyue, 2022-09-20:
@Dazza0 zip file means "esp-idf-v4.4.2.zip", i download it form [https://dl.espressif.com/github_assets/espressif/esp-idf/releases/download/v4.4.2/esp-idf-v4.4.2.zip](url)

> @albertyue could you point to what zip file you are referring to?

Dazza0, 2022-09-21:
@albertyue  This is unusual. When generating the zips for each release, the esp-idf repo is [cloned recursively](https://github.com/espressif/github-actions/blob/master/release_zips/release_zips.py#L39) before it is zipped, thus all of the submodule files should be up to date before it is zip.

Could you run `git diff` to see what is different?

albertyue, 2022-09-21:
@Dazza0 i execute `git diff` in `components/openthread/openthread`, here is the command output:
``` 
$ git diff
warning: CRLF will be replaced by LF in third_party/mbedtls/repo/scripts/data_files/vs6-app-template.dsp.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in third_party/mbedtls/repo/scripts/data_files/vs6-main-template.dsp.
The file will have its original line endings in your working directory.
warning: CRLF will be replaced by LF in third_party/mbedtls/repo/scripts/data_files/vs6-workspace-template.dsw.
The file will have its original line endings in your working directory.
diff --git a/third_party/mbedtls/repo/scripts/data_files/vs6-app-template.dsp b/third_party/mbedtls/repo/scripts/data_files/vs6-app-template.dsp
index 87dbea2..ff7c00e 100644
--- a/third_party/mbedtls/repo/scripts/data_files/vs6-app-template.dsp
+++ b/third_party/mbedtls/repo/scripts/data_files/vs6-app-template.dsp
@@ -1,101 +1,101 @@
-# Microsoft Developer Studio Project File - Name="<APPNAME>" - Package Owner=<4>
-# Microsoft Developer Studio Generated Build File, Format Version 6.00
-# ** DO NOT EDIT **
-
-# TARGTYPE "Win32 (x86) Console Application" 0x0103
-
-CFG=<APPNAME> - Win32 Debug
-!MESSAGE This is not a valid makefile. To build this project using NMAKE,
-!MESSAGE use the Export Makefile command and run
-!MESSAGE 
-!MESSAGE NMAKE /f "<APPNAME>.mak".
-!MESSAGE 
-!MESSAGE You can specify a configuration when running NMAKE
-!MESSAGE by defining the macro CFG on the command line. For example:
-!MESSAGE 
-!MESSAGE NMAKE /f "<APPNAME>.mak" CFG="<APPNAME> - Win32 Debug"
-!MESSAGE 
-!MESSAGE Possible choices for configuration are:
-!MESSAGE 
-!MESSAGE "<APPNAME> - Win32 Release" (based on "Win32 (x86) Console Application")
-!MESSAGE "<APPNAME> - Win32 Debug" (based on "Win32 (x86) Console Application")
-!MESSAGE 
-
-# Begin Project
-# PROP AllowPerConfigDependencies 0
-# PROP Scc_ProjName ""
-# PROP Scc_LocalPath ""
-CPP=cl.exe
-RSC=rc.exe
-
-!IF  "$(CFG)" == "<APPNAME> - Win32 Release"
-
-# PROP BASE Use_MFC 0
-# PROP BASE Use_Debug_Libraries 0
-# PROP BASE Output_Dir ""
-# PROP BASE Intermediate_Dir "temp"
-# PROP BASE Target_Dir ""
-# PROP Use_MFC 0
-# PROP Use_Debug_Libraries 0
-# PROP Output_Dir ""
-# PROP Intermediate_Dir "temp"
-# PROP Target_Dir ""
-# ADD BASE CPP /nologo /W3 /GX /O2 /D "WIN32" /D "NDEBUG" /D "_CONSOLE" /D "_MBCS" /YX /FD /c
-# ADD CPP /nologo /W3 /GX /O2 /D "WIN32" /D "NDEBUG" /D "_CONSOLE" /D "_MBCS" /YX /FD /c
-# ADD BASE RSC /l 0x40c /d "NDEBUG"
-# ADD RSC /l 0x40c /d "NDEBUG"
-BSC32=bscmake.exe
-# ADD BASE BSC32 /nologo
-# ADD BSC32 /nologo
-LINK32=link.exe
-# ADD BASE LINK32 kernel32.lib user32.lib gdi32.lib winspool.lib comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib uuid.lib odbc32.lib odbccp32.lib  kernel32.lib user32.lib gdi32.lib winspool.lib comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib uuid.lib odbc32.lib odbccp32.lib /nologo /subsystem:console /machine:I386
-# ADD LINK32 kernel32.lib user32.lib gdi32.lib winspool.lib comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib uuid.lib odbc32.lib odbccp32.lib  kernel32.lib user32.lib gdi32.lib winspool.lib comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib uuid.lib odbc32.lib odbccp32.lib /nologo /subsystem:console /machine:I386
-
-!ELSEIF  "$(CFG)" == "<APPNAME> - Win32 Debug"
-
-# PROP BASE Use_MFC 0
[... 358 lines cut ...]
-}}}
-
-###############################################################################
-
+Microsoft Developer Studio Workspace File, Format Version 6.00
+# WARNING: DO NOT EDIT OR DELETE THIS WORKSPACE FILE!
+
+APP_ENTRIES
+###############################################################################
+
+Global:
+
+Package=<5>
+{{{
+}}}
+
+Package=<3>
+{{{
+}}}
+
+###############################################################################
+
(END)
```
it seems like something was wrong with line endings.

albertyue, 2022-09-23:
Everything finally back to normal after i upgrade git to latest version (fom 2.7.5 to 2.37.3). I still can't figure out what happened 😂. Maybe i should just let it go .
BTW, thanks to answering @Dazza0

closed 2022-09-23 by albertyue