espressif/esp-idf #9814: ESP hangs when receive GPIO interrupt during sending MQTT message if can't send it

mohammadjavadpirhadi, 2022-09-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-D0WD-V3

### Power Supply used.

USB

### What is the expected behavior?

Doesn't hang!

### What is the actual behavior?

If eps lose internet connection during sending an MQTT message and simultaneously gets GPIO interrupt, it hangs.

### Steps to reproduce.

1. Connect MQTT client
2. Send something by MQTT client on GPIO interrupt
3. Touch that GPIO continuously to send MQTT messages 
4. During touching, disconnect the router from the internet
5. Continue touching and you will see that ESP hangs

### More Information.

My project is too big to share the codes.

mohammadjavadpirhadi, 2022-09-20:
> One question:
> 
> * how are sending mqtt message, with publish or with enqueue?

I have tested both of them, no difference! Even after network timeout (client config parameter), ESP continues to hang.

chegewara, 2022-09-20:
I asked because publish is blocking and you cant call it from ISR, especially when you disconnect wifi.
Im not sure about enqueue.
Maybe higher debug log level could help.

mohammadjavadpirhadi, 2022-09-21:
> I asked because publish is blocking and you cant call it from ISR, especially when you disconnect wifi. Im not sure about enqueue. Maybe higher debug log level could help.

I don't call anything from the ISR function except xQueueSendFromISR.

mohammadjavadpirhadi, 2022-09-22:
> Im not saying its not a bug, but without code i am only trying to understand what may cause hangs. Sometimes it may be queue which is waiting forever when it is full. If you could strip down you code to minimal which could help to reproduce it it would be helpful.

I was able to reproduce it while a higher log level is enabled. The last debug logs are:
`
D (71171) HTTP_CLIENT: Begin connect to: http://google.com:80
D (71171) esp-tls: host:google.com: strlen 10
.
.
.
D (74361) MQTT_CLIENT: Reconnecting...
D (74361) esp-tls: host:***: strlen ***
`

Both tasks get stuck on line 146 of esp-tls: 
`
int res = getaddrinfo(use_host, NULL, &hints, &address_info);
`

mohammadjavadpirhadi, 2022-09-22:
> Someone with better knowledge should answer this, but i am guessing it is blocking function, and for sure its not related to mqtt. @suda-morris

I don't have control over not calling `getaddrinfo`, It is called by HTTP_CLIENT and MQTT_CLIENT.
Probably yes, With a lower debug level what I saw last was attempting to publish an MQTT message so I was thinking it was because of it.

mohammadjavadpirhadi, 2022-09-23:
@chegewara 
I have found something, in this part of my code, we don't reach the second `ESP_LOGD` when the ESP hangs.

```
ESP_LOGD(TAG, "--------PUBLISHING %s", payload_str);
int msg_id = esp_mqtt_client_enqueue(mqtts_client, topic.topic, payload_str, strlen(payload_str), topic.qos, retain, false);
ESP_LOGD(TAG, "--------ENQUEUED %s", payload_str);
```

mohammadjavadpirhadi, 2022-09-27:
> I was able to reproduce it while a higher log level is enabled. The last debug logs are: `D (71171) HTTP_CLIENT: Begin connect to: http://google.com:80 D (71171) esp-tls: host:google.com: strlen 10 [...]

I found something important. These two requests are generated in two different tasks, when I remove the google request task, ESP doesn't hang anymore, but I need to request google to know whether I have internet or not!

chegewara, 2022-10-10:
If you want to check internet connectivity and you found the app stuck in `getaddrinfo` is not easier to test connectivity to static IP instead?
Also i dont see the point of testing connectivity since mqtt component will detect it and emit disconnect event.

mohammadjavadpirhadi, 2022-10-10:
> If you want to check internet connectivity and you found the app stuck in `getaddrinfo` is not easier to test connectivity to static IP instead? Also i dont see the point of testing connectivity since mqtt component will detect it and emit disconnect event.
> 
> This does not looks like a esp-idf bug, more like the problem in your app.

It's important for us to know that we have internet and are disconnected from MQTT or we do not.

> @chegewara I have found something, in this part of my code, we don't reach the second `ESP_LOGD` when the ESP hangs. I'm enqueuing and that is nonblocking so even if it is related to `esp-tls`, we should reach this line!
> 
> ```
> ESP_LOGD(TAG, "--------PUBLISHING %s", payload_str);
> int msg_id = esp_mqtt_client_enqueue(mqtts_client, topic.topic, payload_str, strlen(payload_str), topic.qos, retain, false);
> ESP_LOGD(TAG, "--------ENQUEUED %s", payload_str);
> ```

Why do I get stuck in this line? It is non-blocking! @chegewara

ESP-YJM, 2022-10-11:
@mohammadjavadpirhadi `esp_mqtt_client_enqueue` got stuck may due to `MQTT_API_LOCK(client)` in this API. You can check whether the mqtt is re-connecting the MQTT broker in mqtt main task function(`esp_mqtt_task()`), this task also will call `MQTT_API_LOCK(client)`.

mohammadjavadpirhadi, 2022-10-17:
> @mohammadjavadpirhadi `esp_mqtt_client_enqueue` got stuck may due to `MQTT_API_LOCK(client)` in this API. You can check whether the mqtt is re-connecting the MQTT broker in mqtt main task function(`esp_mqtt_task()`), this task also will call `MQTT_API_LOCK(client)`.

The `esp_mqtt_task()` locks the API but does not unlock it. It gets stuck every time in a different line but always in `case MQTT_STATE_CONNECTED`. It happens when I shut down the router's internet and then connect it again. Why it should behave like that? I'm trying to strip down my code to minimal but it's so hard to keep the issue.

In our other products, we don't face such an issue, can using PSRAM have an effect?

robocircuits, 2023-04-05:
Hi @mohammadjavadpirhadi and @ESP-YJM 
I am also facing the same issue. 
I am using touch interrupts to turn on and off my lights. but whenever esp tries to connect with MQTT my touch pins hang. 
this issue is preventing my product from launching. i need to solve this as soon as possible.

euripedesrocha, 2023-04-11:
@mohammadjavadpirhadi @robocircuits could you provide code to reproduce the issue?

mohammadjavadpirhadi, 2023-04-11:
@euripedesrocha @robocircuits 
Sorry for my late reply.
In our case, the problem was that we accept GPIO change command via MQTT and send the feedback again using MQTT. Also, we publish the GPIO changes by MQTT. The concurrency of receiving a command, sending feedback, and publishing changes made by physical touch was the reason. Although I think this is the IDF issue, I solved the problem by sending the feedback and publishing changes with a common separate task which is different from the task that receives the command. In fact, I send the feedback that I have received the command and then publish the change in the mentioned separate task. Otherwise, the semaphore in the mqtt_task causes a deadlock.

euripedesrocha, 2023-04-11:
@mohammadjavadpirhadi If I got you correctly, you had 2 tasks handling the MQTT API and caused a deadlock. 
You tested with publish and enqueue and still got a deadlock, right? 
Did you stop the client in the process?  We had an issue related to restarting the client that was fixed a while ago, I wonder if it's related. 

I still need some reproducer to be able to check if the issue is in esp-mqtt.

rmarkus, 2023-04-20:
I am having the same issue as euriopesrocha. An interrupt that should be triggered via an GPIO change is NOT executed/captured during MQTT communication.

euripedesrocha, 2024-04-24:
Closing since I wasn't able to reproduce.

closed 2024-04-24 by euripedesrocha