espressif/esp-idf #9816: esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)

rtheil-growlink, 2022-09-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP32-WROOM-32UE with IDF 4.3.1

I have many controllers in production and on controllers that have iffy internet connectivity, I seem to eventually get this error set, and it never stops until I reset the controller.

```
E (53626022) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (53626032) esp-tls: Failed to open new connection
E (53626032) TRANS_SSL: Failed to open a new connection
E (53626042) tls_freertos: Failed establishing TLS connection (esp_transport_connect failed)
```

I was finally able to reproduce this issue on one of my controllers, though it can take upwards of a day to do so. When I rebooted the wifi access point, the connectivity eventually came back.

This is happening when using Azure middleware client, but also seems to affect other connectivity, as it seems that standard http calls do not work once this error occurs. I have no idea how to fix this issue without resetting the controller.

This is similar to https://github.com/espressif/esp-idf/issues/6940, and other people who have come in and said they have a similar problem.

I've created an ESP forum post on this issue as well: https://esp32.com/viewtopic.php?f=13&t=29821

ESP-YJM, 2022-09-22:
@rtheil-growlink It seems the socket has run out and allocate new socket failed. You can check your socket fd number whether increasing when you create new socket.

rtheil-growlink, 2022-09-23:
This seems to be the "socket exhaustion" bug in idf 4.3.x that doesn't exist in 4.4.x. However, after switching to 4.4.2, I've found that it makes things even worse and I get frequent crashes that I have never seen on 4.3.x.

For example:
```
Core  1 register dump:
PC      : 0x401696d3  PS      : 0x00060330  A0      : 0x8016a160  A1      : 0x3ffb2050  
0x401696d3: capture_tcp_transport_error at C:/Users/Ricky/esp/esp-idf/components/tcp_transport/transport.c:332
A2      : 0x3ffc2934  A3      : 0x00008008  A4      : 0x00000001  A5      : 0x3ffb7b40  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x801696a4  A9      : 0x3ffb2030  
A10     : 0x00000000  A11     : 0xffff8d80  A12     : 0x00000001  A13     : 0x3ffb8684  
A14     : 0x00000001  A15     : 0x00060023  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace:0x401696d0:0x3ffb20500x4016a15d:0x3ffb2070 0x401d7d95:0x3ffb20a0 0x40136d77:0x3ffb20c0 0x4013a8f9:0x3ffb20f0 0x401393c5:0x3ffb2110 0x401394e4:0x3ffb2140 0x4013816d:0x3ffb2160 0x40137a6a:0x3ffb2180 0x400dd96b:0x3ffb21b0 0x400d8636:0x3ffb21e0 0x40095929:0x3ffb2200 
0x401696d0: capture_tcp_transport_error at C:/Users/Ricky/esp/esp-idf/components/tcp_transport/transport.c:332
0x4016a15d: ssl_read at C:/Users/Ricky/esp/esp-idf/components/tcp_transport/transport_ssl.c:241
0x401d7d95: esp_transport_read at C:/Users/Ricky/esp/esp-idf/components/tcp_transport/transport.c:190
0x40136d77: TLS_Socket_Recv at C:/repos/ESPfw2/controller-firmware/components/azure-iot/transport_tls_esp32.c:165
0x4013a8f9: MQTT_GetIncomingPacketTypeAndLength at C:/repos/ESPfw2/libs/azure-iot-middleware-freertos/libraries/coreMQTT/source/core_mqtt_serializer.c:2359
0x401393c5: receiveSingleIteration at C:/repos/ESPfw2/libs/azure-iot-middleware-freertos/libraries/coreMQTT/source/core_mqtt.c:1297
0x401394e4: MQTT_ProcessLoop at C:/repos/ESPfw2/libs/azure-iot-middleware-freertos/libraries/coreMQTT/source/core_mqtt.c:2185
0x4013816d: AzureIoTMQTT_ProcessLoop at C:/repos/ESPfw2/libs/azure-iot-middleware-freertos/ports/coreMQTT/azure_iot_core_mqtt.c:182
0x40137a6a: AzureIoTHubClient_ProcessLoop at C:/repos/ESPfw2/libs/azure-iot-middleware-freertos/source/azure_iot_hub_client.c:917
0x400dd96b: AzureService::doWork() at C:/repos/ESPfw2/controller-firmware/main/Services/AzureService.cpp:735
0x400d8636: azureTask(void*) at C:/repos/ESPfw2/controller-firmware/main/controller-firmware.cpp:408
0x40095929: vPortTaskWrapper at C:/Users/Ricky/esp/esp-idf/components/freertos/port/xtensa/port.c:131
```

klaudiusz223, 2022-09-24:
Hi @rtheil-growlink,

Looks like I've just created ticket for the same bug.#9847
Please check if my proposal fixes Your issue.
By the way, I couldn't register on esp forum to replay to Your post. I didn't get a confirmation email:-/
Klaudiusz

ESP-YJM, 2022-09-26:
@rtheil-growlink It seems `err_handle` is NULL, can you add a check for this?

rtheil-growlink, 2022-09-26:
@klaudiusz223, I had the same issue with esp forums and my personal email address. Had to use work email account despite both being hosting in Exchange Online. Looks like your fix is specific to 4.4.2, but I've yet to see the issue occur there. This could be because 4.4.2 has other issues causing frequent crashes, but we're researching to see if it may be related to our code somewhere. Once I get that part sorted out, I will check to see if we get this same problem. Glad to see you found a solution for your issue.

klaudiusz223, 2022-09-26:
@rtheil-growlink After quick research, it appears that the problem has occurred since 4.2.1, so the proposed fix can be applied from 4.2.1 onwards.

rtheil-growlink, 2022-09-26:
@klaudiusz223 I'm guessing one issue was fixed in the socket exhaustion issue, while another was created, as my issue is specific to 4.3.x. Thanks for the extra info.

ESP-YJM, 2022-10-10:
Hi @rtheil-growlink , could you please try this [commit](https://github.com/espressif/esp-idf/commit/3a92e82e9749110ae03ba95e0c5bd16332393fbc) to check the socket exhaustion issue. And another [issue ](https://github.com/espressif/esp-idf/issues/9816#issuecomment-1256208407), do you have any new progress.

rtheil-growlink, 2022-10-10:
@ESP-YJM Thanks. I have that applied and will attempt to flash to the customer's device today to see how well it works. I will also test on one of my devices but it can take a day or two before the problem occurs.

Alvin1Zhang, 2022-11-07:
@rtheil-growlink Thanks for reporting, would you please help share if any updates for the issue? Thanks.

rtheil-growlink, 2022-12-07:
After a ton of testing, I have confirmed that this problem is fixed. I no longer have the socket exhaustion issue. My previous report that the issue was not fixed has something to do with other code causing a memory leak and heap fragmentation, but I have concluded it has nothing to do with espressif code. This can be closed. Thanks so much for the fix.

closed 2022-12-07 by rtheil-growlink