espressif/esp-idf #9817: Mesh Internal Comm Example Question

superteece, 2022-09-20:
I'm just trying to clear this up. In the example config, are the `Router SSID` and `Router Password` values for an internet connected wifi AP so that the mesh has an internet gateway? Or are they for the mesh?

Also, if I flash three dev boards with this example, do any of the example config settings need to be unique to each dev board?

shenjun7, 2022-09-20:
`Router SSID` and `Router Password` are configured for root node. Internal communication between child nodes and root node does not require unique configuration．

superteece, 2022-09-20:
I have the example running now. For future answer seekers I'll leave this summation:

1. `Router SSID` and `Router Password` are the values required to connect a root node to a local WiFi AP.
2.  Leave the `channel` set to 0 unless you know the specific channel for the WiFi AP and know that it does not hop channels for optimization.

closed 2022-09-20 by superteece