espressif/esp-idf #9818: hardware timer isr miss when transmitting data through mqtt?

weizzh, 2022-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi.
I set my hardware time at 4096 hz to get data from a sensor, then put sensor data to a buffer. When the buffer is full, a notify is sent to other task, where these data are published through mqtt.

Then I found the timer become unstable: I miss some data. I visulized the data, it seemed that every some time, data become more ……tight. On a oscilloscope, I watched the timer interrupt might miss sometimes.
Is it possible the timer isr can be affected by wifi?
I have tried to init the timer on another core and changed timer group and also the timer index, but things didn't get better.

The esp32 forum has a similar post:https://www.esp32.com/viewtopic.php?f=2&t=28061

Desperate for any advice, this ghost bothered me monthes.

suda-morris, 2022-09-21:
Yes, peripheral interrupt affected or preempted by wifi is possible on ESP32.

which IDF version? I asked because since 5.0, there's a new driver for timer group, called `driver/gptimer.h`.

Assume you're using the `driver/timer.h`.

Are you installing the timer interrupt on Core1? Typically by calling `timer_isr_callback_add` in a dedicated task, which is **pined** to Core1.

Have you tried to set the `intr_alloc_flags` with `ESP_INTR_FLAG_IRAM` when calling `timer_isr_callback_add`?  Maybe the wifi library is doing some flash operation, which turns off the cache and increases the interrupt latency.

BTW, could you share your timer group initialization code, including the alarm callback function?

Alvin1Zhang, 2022-10-24:
Thanks for reporting, would you please help share if any updates? Thanks.

weizzh, 2022-10-24:
Thanks, I missed github notifications.
I will offer a copy of code which can be run on dev boards to reproduce this problem.

weizzh, 2022-10-26:
> Thanks for reporting, would you please help share if any updates? Thanks.

Hi:
These days I have been preparing a minial reproducible example and have had some interesting progress, which may help.
- HARDWARE: ESP32. You can build and run the code below in a esp32-wroom dev board or esp32-wrover chip(our project's select); IDF: V4.3
- At first, I wanted to prepare the code in a clean environment, so I used `idf.py set-target esp32` to get a `sdkconfig`. After all coding work complete, I build, flash and run on `ESP32-WROOM-32D` board. When I watched pin 12(SPI CS line) and pin 33 (SPI CLK line) using a logic analyzer. I got very regular signals: I just couldn't find the bug! 
[image]

- Then I realized that In the former project I had done some modifications to sdkconfig. So I copy the sdkconfig to this new code directory. And this time, these bugs appeared. My sample frequent is 4096, so the period should be 244.1 us. 
[image]

- So I will offer two copy of `sdkconfig` to reproduce my problem. 
[ad_test_for_git.zip](https://github.com/espressif/esp-idf/files/9869738/ad_test_for_git.zip)

- PS: to use a different `sdkconfig` file, you should replace the original file, and run `idf.py menuconifg` and save to exit.
- Any results please let me know. I'd like to offer any information needed.

weizzh, 2022-11-01:
Hi, can the code reproduce the problem?

suda-morris, 2022-11-01:
Hi @weizzh I think the ghost is here:

```
CONFIG_PM_ENABLE=y
CONFIG_PM_DFS_INIT_AUTO=y
```

On esp32, the clock source of timer group is fixed to APB (no other choice), so if you enabled the above Kconfigs, the system will try to reduce the power consumption by lowering the APB frequency. 
You can find more details [here](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/system/power_management.html#dynamic-frequency-scaling-and-peripheral-drivers) and yes the solution.

Sorry for the late reply.

weizzh, 2022-11-02:
@suda-morris
I think your diagnosis is correct. After changing power management config, the code wokes well.
I have some other questions from former objects about this issue.
case 1:sample a period of time, for example, 10 seconds, then stop sampling and transmit these data through wifi(mqtt). After transmitting, continue to sample work......I can get corrent data.
case 2:sample and transmit work simultaneously using ping-pong buffers. Just as the first post in this thread discribes, I get bad data priodically.
case 3:i2s module added to case 2, but use apll, i2s data aren't transmitted through wifi. Still priodical bad data.
case 4: i2s module added using apll, but the i2s data are transmitted through wifi, I can get corrent data.
the wifi config: WIFI_PS_MIN_MODEM
I think in case 4, the wifi is busy, so it cannot release the ESP_PM_APB_FREQ_MAX lock, thus the hardware timer get stable APB, right?
But I cannot figure out case 1 and case 2.

esp-wzh, 2023-08-11:
if `CONFIG_PM_DFS_INIT_AUTO` is enabled, esp32's DFS will switch the CPU frequency between 40mhz and  the CPU freq you configured in menuconfig. 

The APB clock of esp32 will be affected by the change of CPU frequency, if the CPU uses pll clock source (>=80MHz), the APB frequency is fixed at 80MHz, while if the CPU downclocked to use crystal as clock source (<=40MHz), the CPU frequency is the same as the APB frequency. On each frequency update during DFS, IDF will adjust the clock divider inside the timergroup module to compensate it, but this inevitably introduces some drift each time the frequency is switched, as shown in the table below.

[image]

So there are two ways to solve this problem, if you care about power consumption and accept a slight timer inaccuracy, you can turn on lightsleep at esp_pm_configure, which will reduce the frequency of DFS and thus reduce the cumulative error.

Or if you are more concerned about timer accuracy you can hold the APB_MAX lock while using the timer.

esp-wzh, 2024-02-26:
If no more issues, please close this thread.

Indastri, 2024-04-22:
As there was no response from the user since Feb closing this issue. If you will encounter any other problems or issues feel free to reopen this or create a new issue.

Cheers!

closed 2024-04-22 by Indastri