espressif/esp-idf #9819: simple_ota_example test

BATAxjh, 2022-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

V4.4.2  

Why this problem
I (5708) simple_ota_example: Starting OTA example
W (5708) esp_https_ota: Continuing with insecure option because CONFIG_OTA_ALLOW_HTTP is set.
E (7978) TRANSPORT_BASE: poll_read select error 104, errno = Connection reset by peer, fd = 54
E (7978) esp_https_ota: Failed to establish HTTP connection
E (7988) simple_ota_example: Firmware upgrade failed

[image]
[image]

hmalpani, 2022-09-21:
Hello @BATAxjh ,
Can you send the debug logs. Enable `LOG_DEFAULT_LEVEL_VERBOSE`, `LWIP_SOCKETS_DEBUG` and `LWIP_TCP_DEBUG` in menuconfig to get the debug logs.

hmalpani, 2022-10-10:
Hello @BATAxjh 
Any update on this?

BATAxjh, 2022-10-10:
> 

oh  ,This problem is caused by the inconsistent network segments I set,Although it can ping.    sorry

closed 2022-10-10 by hmalpani