espressif/esp-idf #9820: esp_wifi_set_max_tx_power() and esp_wifi_get_max_tx_power() don't work as expected

BeKindToBeings, 2022-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

For convenience, `get()` refers to `esp_wifi_get_max_tx_power()`
and `set()` refers to `esp_wifi_set_max_tx_power()`

1. After start-up, doing `get()` returns `78`.
I at least expect that doing `set(78)` and then get() again returns 78 again. 

2. I expect that doing `set()` with _any_ value of power in the range according to the `3. Param power [---] range` found in the [Programming Guide](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html?highlight=esp_wifi_set_max_tx_power#_CPPv425esp_wifi_set_max_tx_power6int8_t) would have `get()` return the value according to `5. Relationship between set value and actual value.` in the [Programming Guide](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html?highlight=esp_wifi_set_max_tx_power#_CPPv425esp_wifi_set_max_tx_power6int8_t).

### What is the actual behavior?

**1.** It returns a lower value, either 64, 65 or 65. A different value might be obtained, if resetting and doing set(78) and get() again.

**2a.** After doing `set()` with a value in the higher end of the range, `get()` returns the value exemplified in `1.`.

**2b.** Then doing `set()` with _any_ higher value than the value returned by `get()` as brought out in `1.` does not seem to have any effect.

**2c.** The `5. Relationship between set value and actual value.` in the [Programming Guide](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html?highlight=esp_wifi_set_max_tx_power#_CPPv425esp_wifi_set_max_tx_power6int8_t) does not seem to hold. The values returned by get() were sometimes different after reset. One example: 6, 14, 20, 30, 38, 42, 46, 52, 58, 64. Another example: ... 31, 39, 43, 47, 53, 59, 65.

### Steps to reproduce.

Start wifi. Wifi works. Everything works.
Call `esp_wifi_set_max_tx_power()` with a value of power in the range according to the `3. Param power [---] range` found in the [Programming Guide](https://docs.espressif.com/projects/esp-idf/en/v4.4.2/esp32/api-reference/network/esp_wifi.html?highlight=esp_wifi_set_max_tx_power#_CPPv425esp_wifi_set_max_tx_power6int8_t)
and then call `esp_wifi_get_max_tx_power()` as follows:

int8_t power = 0;
esp_err_t retval = esp_wifi_get_max_tx_power(&power);
if ( ESP_OK != retval )
{
  /* log failure with code */
}
else
{
  /* print the value to debug interface */
}

### More Information.

How I built:
* `. ~/esp/esp-idf/export.sh`
* `cd esp/build`
* `cmake ..`
* `make menuconfig`
* ` make`

AxelLin, 2022-12-17:
@mhdong Any update?

kriegste, 2023-04-28:
Still not fixed in 4.4.4.

On top of the mentioned problem, there is more:

wifi_country_t.max_tx_power (after esp_wifi_get_country) is 20.
esp_wifi_get_max_tx_power returns 78.
The manual says (about esp_wifi_get_max_tx_power) that the unit is 0.25dBm. So 78 is 19.5dBm.
How to set 20dBm?

After a brownout, this warning appears: "phy_init: saving new calibration data because of checksum failure, mode(0)".
wifi_country_t.max_tx_power is still 20.
But now, esp_wifi_get_max_tx_power returns 52, i.e. 13dBm. I know, this is a feature, but in this situation, after waiting for a few seconds, I'd like to manually set 20dBm (without restarting).

therealergo, 2026-10-30:
@mhdong Still not fixed in 5.5.1. This issue is present when ESP_PHY_REDUCE_TX_POWER is set to either "y" or "n".