espressif/esp-idf #9821: Allow core affinity setting for dpp taks

allard-potma, 2022-09-20:
### Describe the solution you'd like.

Most of the system tasks in esp-idf allow to be pinned to a certain core. The DPP task does not, see: https://github.com/espressif/esp-idf/blob/16a4ee7c36a848ca155791677ce011f3ca75c519/components/wpa_supplicant/esp_supplicant/src/esp_dpp.c#L612

It would be nice if this task can also be pinned. Im making an application that needs a whole core just for one high-performance task, I do not want DPP to intervene.

sarveshb14, 2022-09-21:
@allard-potma you can use `os_task_create_pinned_to_core()` api from the following patch to assign core affinity for a task.
[allow_core_affinity_dpp_task.txt](https://github.com/espressif/esp-idf/files/9616541/allow_core_affinity_dpp_task.txt)

allard-potma, 2022-09-29:
Thank you. It would be nice if the affinity would be settable trough menuconfig in the future.

closed 2022-10-21 by nachiketkukade