espressif/esp-idf #9822: Error while init SD(SDIO) card in SPI mode

zanzaradinord, 2022-09-20:
Hi all, im trying to init unnamed SD card in SPI mode. This card works normally with Win and Linux. I use ESP32-WROOM-32. This card was detected as SDIO card (see log on bottom). With an another good card the mcu works correct.

Environment:
> SPI: MISO - 14, MOSI - 23, SCK - 13, CS - 5,
> ESP32-WROOM-32,
>  ESP-IDF v4.4.2, 
> ubuntu

Anyone with any ideas?
Thanks in advance

`
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6660
load:0x40078000,len:14848
ho 0 tail 12 room 4
load:0x40080400,len:3792
entry 0x40080694
I (29) boot: ESP-IDF v4.4.2 2nd stage bootloader
I (29) boot: compile time 22:20:00
I (29) boot: chip revision: 1
I (32) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0da4ch ( 55884) map
I (131) esp_image: segment 1: paddr=0001da74 vaddr=3ffb0000 size=025a4h (  9636) load
I (135) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=214e4h (136420) map
I (186) esp_image: segment 3: paddr=0004150c vaddr=3ffb25a4 size=00014h (    20) load
I (187) esp_image: segment 4: paddr=00041528 vaddr=40080000 size=0d174h ( 53620) load
I (214) esp_image: segment 5: paddr=0004e6a4 vaddr=50000000 size=00010h (    16) load
I (221) boot: Loaded app from partition at offset 0x10000
I (221) boot: Disabling RNG early entropy source...
I (234) cpu_start: Pro cpu up.
D (235) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 15 bit
D (235) efuse: In EFUSE_BLK0__DATA5_REG is used 1 bits starting with 20 bit
D (241) efuse: In EFUSE_BLK0__DATA3_REG is used 3 bits starting with 9 bit
D (248) efuse: In EFUSE_BLK0__DATA3_REG is used 1 bits starting with 2 bit
I (255) cpu_start: Starting app cpu, entry point is 0x40081168
I (0) cpu_start: App cpu up.
D (269) clk: RTC_SLOW_CLK calibration value: 3289792
I (282) cpu_start: Pro cpu start user code
I (282) cpu_start: cpu freq: 160000000
I (282) cpu_start: Application information:
I (286) cpu_start: Project name:     sd_card
I (291) cpu_start: App version:      1
I (296) cpu_start: Compile time:     Sep 20 2022 22:20:11
I (302) cpu_start: ELF file SHA256:  4950e16078f00a1f...
I (308) cpu_start: ESP-IDF:          v4.4.2
V (313) memory_layout: reserved range is 0x3f40da34 - 0x3f40da6c
D (319) memory_layout: Checking 7 reserved memory ranges:
D (324) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (331) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb2f18
D (337) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (343) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (350) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (356) memory_layout: Reserved memory range 0x40078000 - 0x40080000
D (363) memory_layout: Reserved memory range 0x40080000 - 0x4008d174
D (369) memory_layout: Building list of available memory regions:
[... 246 lines cut ...]
V (1836) bus_lock: dev 5 released.
D (1846) sdspi_host: data CRC set=1
V (1846) sdspi_transaction: r1 = 0x01 hw_cmd.r[0]=0xffffffff
V (1846) sdmmc_cmd: cmd response 00000001 00000000 00000000 00000000 err=0x0 state=0
D (1856) sdmmc_init: sdmmc_card_init: card type is SDIO
D (1866) sdmmc_common: sdmmc_init_host_frequency: using 400 kHz bus frequency
D (1866) vfs_fat_sdmmc: using pdrv=0
V (1876) sdmmc_cmd: sending cmd slot=1 op=13 arg=0 flags=1c00 data=0x0 blklen=0 datalen=0 timeout=1000
V (1886) sdspi_host: sdspi_host_start_command: slot=1, CMD13, arg=0x00000000 flags=0x8, data=0x0, data_size=0 crc=0x06
V bus_lock: dev 5 acquired.
D (1896) spi_master: device5 locked the bus
V (1906) spi_master: polling trans
V (1906) spi_master: polling trans done
V (1906) spi_master: polling trans
V (1916) spi_master: polling trans done
D (1916) spi_master: device5 release bus
V (1916) bus_lock: dev 5 released.
V (1926) sdspi_transaction: r1 = 0x01 hw_cmd.r[0]=0xffffff00
V (1926) sdmmc_cmd: cmd response 000001ff 00000000 00000000 00000000 err=0x0 state=0
W (1936) vfs_fat_sdmmc: failed to mount card (1)
E (1946) vfs_fat_sdmmc: mount_to_vfs failed (0xffffffff).
I (1946) gpio: GPIO[5]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
E (1956) example: Failed to mount filesystem. If you want the card to be formatted, set the CONFIG_EXAMPLE_FORMAT_IF_MOUNT_FAILED menuconfig option.`

[1gb-sd-card.txt](https://github.com/espressif/esp-idf/files/9611120/1gb-sd-card.txt)

adokitkat, 2023-04-18:
Hello. It is possible a SD card from an unknown brand does not fully comply with the SD standard and does not implement SPI mode at all. Maybe you can try 1-line (1-bit) SDMMC mode?

adokitkat, 2023-04-24:
Thank you for pointing out the fix tip! I will look into it.

closed 2023-04-25 by espressif-bot