espressif/esp-idf #9823: SPI Slave HD Dual and Quad (DOUT, DIO, QOUT, QIO) Dummy Cycles Mismatch

zmswanson, 2022-09-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### Summary
SPI slave HD Segment Mode example works as expected in 1-bit mode between ESP32 master and ESP32-S3 slave. However, example does not work in dual- or quad-spi modes (DIO, DOUT, QIO, QOUT) due to a clock-cycle mismatch in dummy stage. The slave device always requires 8 dummy clock cycles regardless of mode and master ESP Serial Slave LINK (ESSL) driver is programmed to always provide 4 dummy clock cycles in dual and quad mode (8 otherwise). **Quick fix**: Modify esp_serial_slave_link component to always provide 8 dummy clock cycles (see essl_spi.c > get_hd_dummy_bits). **Todo**: verify dummy requirements of other non-S3 ESP32 slave devices. **Recommendation**: Update esp_serial_slave_link component in ESP-IDF to always provide 8 dummy cycles or add a transaction flag to specify which type of slave device is connected.

### Description
I am using SPI peripherals in QIO mode to provide communication between an ESP32 module (master) and an ESP32-S3 module (slave). I'm using the SPI2 host on both modules with the dedicated IO mux pins for the respective peripherals. I started with the spi_slave_hd example (examples/peripherals/spi_slave_hd/segment_mode) in 1-bit mode and it worked as expected. 

I switched to QIO mode by adding the flags "SPI_TRANS_MODE_QIO | SPI_TRANS_MODE_DIOQIO_ADDR" to each ESSL read/write operation in seg_master/main/app_main.c. The slave and master applications started but never proceeded past the "wait_for_slave_ready" state. I hooked up a logic analyzer and observed as shown below a) the correct command was sent over 8 cycles b) the correct address was sent over two cycles (I played around with other values than 0 to verify) c) data from the slave appeared to be transmitted after about 8 dummy cycles and d) the data transmitted from the slave included the correct 0xEE flag. The 32-bit value I should have received was 0x000000EE but I was instead receiving 0x00EE0000 or 0xEEEE0000. So, it seemed like something was up with the dummy stage of the transition. I also observed the similar issues half-duplex dual spi mode.

[image]

By analyzing the SPI USER and USER1 registers, I discovered that the dummy_bits variable of the spi_slave_hd_slot_config_t struct (configuring slave) and the spi_device_interface_config_t struct (configuring master) did not actually change the values of dummy-related bits in these registers.  The dummy cycles for the master was always getting set to 4 and I tracked that down to the following function in essl_spi.c which gets called by every ESSL read and write function to determine how many dummy bits:

```
static int get_hd_dummy_bits(uint32_t flags)
{
    //dummy is always 4 cycles when dual or quad mode is enabled. Otherwise 8 cycles in normal mode.
    if (flags & (SPI_TRANS_MODE_DIO | SPI_TRANS_MODE_QIO)) {
        return 4;
    } else {
        return 8;
    }
}
```
I was also able to modify spi_slave_hd.c to actually apply the dummy bits set in spi_slave_hd_slot_config_t to change the USER and USER1 registers. This was verified by monitoring the raw memory locations of the registers. However, modifying dummy bits in the registers did not impact how many clock cycles the slave actually waited: it always waited 8 bits. There's a small blurb in the ESP32-S3 technical reference manual (page 1092) about the DUMMY bits for the GP-SPI peripheral in working as a slave (section 30.5.9.1):

> 3. DUMMY:
> • Its value is meaningless. SPI slave prepares data in this state;
> • Bit mode of FSPI/SPI3 bus is also meaningless here;
> • Last for eight SPI_CLK cycles.

So the problem is that the ESP32-S3 (I have not test on other slave devices) always uses 8 dummy clock cycles to prepare data as a slave and the ESP32 master always provides 4 dummy cycles causing a dummy synchronization issue.

### Solution
Edit get_hd_dummy_bits in essl_spi.c to always return 8:
```
static int get_hd_dummy_bits(uint32_t flags)
{
    return 8;
}
```
If this only applies to ESP32-S3 add a transaction flag and edit get_hd_dummy_bits to check the slave type:
```
static int get_hd_dummy_bits(uint32_t flags)
{
    //dummy is always 4 cycles when dual or quad mode is enabled. Otherwise 8 cycles in normal mode.
    if (flags & (SPI_TRANS_MODE_DIO | SPI_TRANS_MODE_QIO)) {
        if (flags & SPI_TRANS_SLAVE_S3) {
            return 8;
        } else {
            return 4;
        }
    } else {
        return 8;
    }
}
```

Icarus113, 2022-09-21:
HI @zmswanson , which IDF version are you using?

zmswanson, 2022-09-21:
@Icarus113 Originally encountered this issue with v4.4.1; switched to v4.4.2 and still encountered issue.

Icarus113, 2022-09-21:
@zmswanson  Thanks for the reply. On master branch, this [commit](https://github.com/espressif/esp-idf/commit/3610b14aef055f51def67bddaf98a5609d99c9b5) should fix the issue. Could you please give it a try?

The fix doesn't catch up with the release v4.4.2, but next release should contain it.

zmswanson, 2022-09-22:
@Icarus113 I was able to test and the quad SPI communication worked as expected. Thank you!

closed 2022-09-22 by zmswanson

Alvin1Zhang, 2022-09-23:
Thanks for reporting, the fix on release/4.4 is available at https://github.com/espressif/esp-idf/commit/0f630acf43252c867891c8da27bd22725421455c and will be included in the next release of release/4.4. Thanks.