espressif/esp-idf #9824: system: allow using SRAM1 as IRAM

ESP-Marius, 2022-09-21:
### Is your feature request related to a problem?

Currently on the ESP32 the SRAM1 region is used as DRAM, but the chip also allows for this region to be used as IRAM. For users that are running low on IRAM it would be beneficial to be able to use parts of SRAM1 as IRAM.

### Describe the solution you'd like.

We should make this functionality available to users, either automatically at compile time or through a kconfig option if this turns out to be more convenient. 

### Additional context.

Related issue: https://github.com/espressif/esp-idf/issues/7876

someburner, 2022-11-08:
+1 for this. With release/v5 IRAM fills up (especially with bluetooth) unless we disable optimizations. We don't have a very big dram footprint. Trying the changes suggested in that issue compiled and ran, but blufi couldn't init. Guessing something with the heap layout.

someburner, 2022-11-08:
Actually, got it working with these edits:
[iram.patch.txt](https://github.com/espressif/esp-idf/files/9957504/iram.patch.txt)

Would be nice to have it be a config option and officially supported.

alxvasilev, 2022-11-27:
I could successfully build and start my firmware with the above patch to increase the IRAM. 
However, it looks like the whole area of SRAM1 is not used for DRAM at all! Regardless of how much of SRAM1 I add to the IRAM, the total amount of heap memory, as reported by `heap_caps_get_total_size()` is always 178944. Any idea why this is happening?

TeunVR, 2022-12-20:
+1 We really need more IRAM in our project

someburner, 2023-03-24:
In case this helps anyone, this seems to be the best sweet spot for nimble BLE + wifi + wifi iram optimizations, with a little to spare. The other one is too much iram in my case.

[iram-bt.patch](https://github.com/espressif/esp-idf/files/11059868/iram-bt.patch)

closed 2023-04-10 by espressif-bot

ESP-Marius, 2023-04-11:
Added a kconfig option for this in 5cbd311, please let me know if there is any issues with it.

ESP-Marius, 2023-04-12:
@darthcloud Yeah, you are right unfortunately. I've updated the docs to better reflect that this feature wont work with an older bootloader.

I don't see a good way of getting tolder bootloaders to accept using SRAM1 as IRAM. 

Lately we've also been adding more and more options for placing code that was previously in IRAM into flash. You can keep checking in on https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#optimizing-iram-usage which we will keep updating with all these options.

darthcloud, 2023-04-12:
@ESP-Marius the biggest problem is that the IDF iram usage increase by around 1Kb to 2Kb every six months using the same project & same sdkconfig. That's not really sustainable. I think any new IRAM usage by IDF should come with an opt-out option.

I feel my only option is to stop updating the idf and run my own fork were I cherry pick only the update I need. This might be fine for a while but I can see this becoming quite a burden as the branches divergent over time.

Can''t we somehow make the old bootloader load an app that is actually an updated bootloader that will in turn boot the real app?

KaeLL, 2023-04-12:
@ESP-Marius generalizing @darthcloud's question: what exactly is **not** possible/feasible to update once devices are already on the field (with secure boot/encrypted nvs enabled)? Let's say the need to update the 2nd stage bootloader or the partition table arose. Could it be done by the app running on it or something?

ESP-Marius, 2023-04-13:
> the biggest problem is that the IDF iram usage increase by around 1Kb to 2Kb every six months using the same project & same sdkconfig. That's not really sustainable. I think any new IRAM usage by IDF should come with an opt-out option.

We are aware of this issue and are trying to be stricter internally about placing stuff in IRAM (i.e. only code that must always be in IRAM should go there by default)

> Can''t we somehow make the old bootloader load an app that is actually an updated bootloader that will in turn boot the real app?

I'm afraid something like this isn't an easy change, as it would mean modifying the boot work-flow a lot.

> @ESP-Marius generalizing @darthcloud's question: what exactly is not possible/feasible to update once devices are already on the field (with secure boot/encrypted nvs enabled)? Let's say the need to [...]

In theory there is nothing stopping us from writing to flash and modifying the bootloader, but the main issue is how to do this safely (i.e. device should not brick if something goes wrong during this update).

Doing this safely would probably require some kind of third stage bootloader so you would have:

ROM-> 2. stage bootloader (very simple, but non-updatable)-> 3 stage bootloader (more complex, but can be updated, and roll-back on failure etc) -> app. Which is starting to get pretty complex.

So IDF tries keep the functionality in the bootloader as simple as possible, so that the chance of a bootloader update being required is kept to a minimum, in order to avoid this complexity.

ESP-Marius, 2023-04-21:
> Is it still possible to customize the IRAM/DRAM tradeoff with this?

Not sure exactly what you refer to here but yes, this basically trades heap for IRAM.

someburner, 2023-04-21:
I mean is it possible to tweak `SRAM1_IRAM_LEN` without affecting bootloader, and also curious about these regions still:

```
+//    { 0x3FFF0000, 0x8000, 1, 0x400A8000}, //pool 7 <- can be used for MAC dump
+//    { 0x3FFF8000, 0x4000, 1, 0x400A4000}, //pool 6 blk 1 <- can be used as trace memory
+//    { 0x3FFFC000, 0x4000, 1, 0x400A0000}, //pool 6 blk 0 <- can be used as trace memory
```

ESP-Marius, 2023-04-21:
> I mean is it possible to tweak SRAM1_IRAM_LEN without affecting bootloader,

No, as this area was previously reserved for the bootloader

> Or do these get used as heap automatically depending on sdkconfig?

Even with this option enabled, these areas are available as heap if no IRAM code is put there, how much will depend on your binary of couse.

someburner, 2023-04-21:
Should have been a little more clear- is it possible to -reduce- `SRAM1_IRAM_LEN` and have a bit more DRAM?

Abhay-2412, 2023-12-06:
> Actually, got it working with these edits: [iram.patch.txt](https://github.com/espressif/esp-idf/files/9957504/iram.patch.txt)
> 
> Would be nice to have it be a config option and officially supported.

hey i am complete newbie and i ran out of iram can you please tell me how to apply the patch

someburner, 2023-12-08:
latest IDF has an option for this now in menuconfig. `ESP_SYSTEM_ESP32_SRAM1_REGION_AS_IRAM`