espressif/esp-idf #9827: Production flash

netanbi, 2022-09-21:
Hi 
I am during development with the esp32-c3
So far i work with eclips and evk for the esp-c3 ,
I flash the evk with the built in usb or with the connecting to  gpio18/19 ,gnd, 5v and all works good.
Now i want to design the my board , so i want to test how to flash the module without the usb the in the evk
,I connect my usb to serial  to the uart ,gnd, 3.3 and use the pytool to program and it semss the he didnt find the module , but if i use the pytool for programin with the built in usb connector he succeced to flash .

Any ide what is the best and easy way for program the esp32-c3 for production?

suda-morris, 2022-09-22:
have you designed any "Auto Program Circuit" on your board? Please check [the reference schematic](https://dl.espressif.com/dl/schematics/SCH_ESP32-C3-DEVKITM-1_V1_20200915A.pdf) out

netanbi, 2022-09-22:
Hi, thank you for your replay, 
I still didn't finish to design my board, i just stitch it now on the evk ,  
I didn't implement that and this is my problem, it's because I didn't have CTC and RTS lines in the USB 2 serial, 

Now I try to flash the esp32-c3 with JTAG, 
Do you have a production tool for flashing with jtag ?  because the pytool is for serial connection ...

so what is your recommendation for production flashing?
do we can to do also with Two-wire Interface JTAG ?   or I need the full JTAG connection?

arosyc, 2022-09-26:
There is no mass production download tool for JTAG. The most recommended way is UART.

netanbi, 2022-09-28:
thanks, arosyc for your replay 
In my design I need the UART for communication with the external MCU, is that o.k to connect in parallel to the pin for the flashing device?

arosyc, 2022-09-29:
It is best to use the chip's UART1 to communicate with an external MCU, and UART0 (the default port for downloading firmware) is used to download firmware.

closed 2024-05-10 by igrr