espressif/esp-idf #9828: mcpwm_bldc_hall_control example won't build on ESP32.  Missing definitions.

linuxguy123, 2022-09-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-dev-4257-g508661b03e

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

It should build.

### What is the actual behavior?

1) If one uses the stock includes for the example, one gets this:

fatal error: driver/mcpwm_prelude.h: No such file or directory
  386 | #include "driver/mcpwm_prelude.h"

OR

2) If one uses #include "driver/mcpwm.h", one gets this:

error: unknown type name 'mcpwm_cap_timer_handle_t'; did you mean 'esp_timer_handle_t'?
  715 |     mcpwm_cap_timer_handle_t cap_timer = NULL;

### Steps to reproduce.

1. Install esp-idf ESP32 only.   Not ESP32-S3.

2. Download and build the example: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/mcpwm/mcpwm_bldc_hall_control

pytest_bldc_hall_control.py runs the test for esp32s3, but not plain esp32.   I suspect that the esp32 mcpwm include library is messed up for esp32 but is correct for esp32s3.

### Build or installation Logs.

```plain
fatal error: driver/mcpwm_prelude.h: No such file or directory
  386 | #include "driver/mcpwm_prelude.h"

error: unknown type name 'mcpwm_cap_timer_handle_t'; did you mean 'esp_timer_handle_t'?
  715 |     mcpwm_cap_timer_handle_t cap_timer = NULL;
```

### More Information.

find doesn't find mcpwn_prelude.h on my computer. 
grep doesn't find mcpwm_cap_timer_handle_t on my computer.

linuxguy123, 2022-09-21:
Actually, it will build in my source tree.   What won't build is the code from here:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/mcpwm/mcpwm_bldc_hall_control

main.c is different.

suda-morris, 2022-09-22:
Hi @linuxguy123 Since IDF-5.0, we've introduced a new driver for MCPWM peripheral. The examples are also adapted to the new driver APIs. See the [Migration Guide for more information](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/peripherals.html#mcpwm).

That means the example on the master branch or on the release/v5.0 branch can't be built on the earlier release branch. You can either upgrade your SDK version to the latest release/5.0 commit or use the master branch to build this example.

> pytest_bldc_hall_control.py runs the test for esp32s3, but not plain esp32. I suspect that the esp32 mcpwm include library is messed up for esp32 but is correct for esp32s3.

The example does run on esp32s3 in the CI, but the example will still be built for esp32 in the CI.

linuxguy123, 2022-09-27:
My code compiled once I upgraded to v5.1-dev.   It would not compile with v5.0-dev.

Thanks for your help.

Alvin1Zhang, 2022-09-28:
Thanks for the updates, feel free to reopen.

closed 2022-09-28 by Alvin1Zhang