espressif/esp-idf #9829: Increase use of SPIRAM

wenxiao11, 2022-09-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### Environment: 
IDF v4.4

### Question: 
I have already enabled `CONFIG_ESP32_SPIRAM_SUPPORT` and `CONFIG_SPIRAM_USE_MALLOC`, but the IRAM is still not enough. Is there any other ways to increase the amount of use for SPIRAM to reduce the pressure of IRAM?

### sdkconfig: 
[sdkconfig](https://github.com/espressif/esp-idf/files/9620864/sdkconfig.txt)

ESP-Marius, 2022-09-22:
Hi,

Generally speaking using SPIRAM doesn't reduce the pressure of IRAM, but rather DRAM.

Have you tried taking a look at our [guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#optimizing-iram-usage) for minimizing IRAM usage? There are some options mentioned there that might help with your current issue.

wenxiao11, 2022-09-22:
Hi,

Thanks for responding. I have already tried these options, but the IRAM is still not enough. However, I tried to change `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=0`, the available IRAM increased a lot and is enough currently. Is there any side effect about doing this action?

ESP-Marius, 2022-09-23:
> However, I tried to change CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=0, the available IRAM increased a lot and is enough currently

Are you sure you are not confusing IRAM with DRAM or heap? Because `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL` shouldn't affect IRAM usage.

>  Is there any side effect about doing this action?

Apart from that there might be some potential slowdown (accessing variables through cache in SPIRAM is slower than DRAM), everything should work OK. We have seen issues before when setting it to 0 due to some variables we expected to be in internal memory ended up in external memory, but all known issues regarding this has been fixed on latest release branches. 

If you do experience an issue then please report it.

ESP-Marius, 2022-10-18:
Closing this one as the original question has been answered, feel free to reopen if you still think there is any outstanding issue.

closed 2022-10-18 by ESP-Marius