espressif/esp-idf #9830: CONFIG_BT_ACL_CONNECTIONS cannot be set to 1 when using SPP (Classic)

travis012, 2022-09-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I need to prevent our companies interface from allowing more than one SPP connection at a time. When I set CONFIG_BT_ACL_CONNECTIONS to 1, It allows us to connect one time only. IE: If we connect then disconnect, the interface has to be reset before we can connect again.  A strange message appears in the log on this second connect attempt:
`BT_BTM: btm_sec_l2cap_access_req() PSM: 3 no application registerd`

We are using legacy paring if that makes any difference.

ESP32-MINI-01
4.4.2 framework

Regards,
Travis
TTS Inc.

boblane1, 2022-09-22:
Hi @travis012,

Thanks for your report! `CONFIG_BT_ACL_CONNECTIONS ` in host means the number of L2CAP connection. Before the start of SPP connection, the host will initiate an SDP connection to check if the peer device has SPP service.

If you want to implement the functionality of `I need to prevent our companies interface from allowing more than one SPP connection at a time. `. There are two ways:

1. Set the `BTDM_CTRL_BR_EDR_MAX_ACL_CONN` to 1 which is located in the configuration of Bluetooth controller.
2. Call the function `esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);` once the SPP connection is established.

travis012, 2022-09-22:
Excellent!  Thank you very much for your quick response and clarification.  This solved my problem.

For clarification to others that might read this:
I was setting `Bluedroid Option->BT/BLE MAX ACL` Connections to 1
The correct config item is `Bluetooth Controller->BR/EDR Max ACL Connections`

closed 2022-09-22 by travis012