espressif/esp-idf #9831: add udev rules inclusion among troubleshooting tips in docs

acnologia000, 2022-09-22:
module : esp32c3
board: Ai Thinker NodeMCU ESP-C3-12F Development Board 
host os: fedora 36 

i started by following the official tutorial which went good until the flashing step which after research i added my user to dial group and what not but flashing kept failing so i decided to try the vscode extension way which also did not work because due to some reason it was detecting my esp32 at ttyusb1 but in reality it was ttyusb0, during  installation extension told me to execute certain command that copies udev rules from sdk folder to appropriate location, i would like that command to be added to troubleshooting tips

brianignacio5, 2022-09-22:
Hi @acnologia000 Do you mean in the [vscode extension troubleshooting](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting) ? or in the ESP-IDF documentation ?

acnologia000, 2022-09-22:
esp idf documentation @brianignacio5

igrr, 2023-03-27:
Fixed in https://github.com/espressif/esp-idf/commit/b08d08d766a38f66f97ed591d9945cae59b1e9ef.