espressif/esp-idf #9835: Help on eFuse

ifoxbr, 2022-09-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

I need help to be able to read and write the efuse UART_PRINT_CONTROL.
This eFuse is responsible for sending booting messages using UART0 at startup, but I have a printer conected on Uart0.
I used the following code to read this eFuse, but ESP-IDF crashes on PANIC.
I would be very grateful if anyone can help

Regards

```
int turns_off_boot_messages(void)
{
    // Read UART_PRINT_CONTROL eFuse
    uint32_t FUSE = esp_efuse_read_reg(EFUSE_BLK0, ESP_EFUSE_UART_PRINT_CONTROL);

    // If Zero, change do 3
    if(FUSE == 0)
    {
        esp_err_t ret = esp_efuse_write_reg(EFUSE_BLK0, ESP_EFUSE_UART_PRINT_CONTROL, 0x03);
        if(ret == 0)
            ret = 1;		// done
        else
            ret = 0;		// error
    }
}
```

ESP-Marius, 2022-09-23:
`esp_efuse_read_reg` expects a reg number but `ESP_EFUSE_UART_PRINT_CONTROL` is a `esp_efuse_desc_t*`. The compiler will give you a type warning here if you havent turned it off.

ESP-IDF already provides an API for controlling UART_PRINT_CONTROL, `esp_efuse_set_rom_log_scheme`. There is even a menuconfig option for enabling this: `Boot ROM Behavior ` -> ` Permanently change Boot ROM output`. Is this sufficient for what you need?

ifoxbr, 2022-09-23:
Hi Marius
Thank you very much, i will check these 2 suggestions

ifoxbr, 2022-09-23:
Unfortunately it didn't work using menuconfig....
[image]
Which include file contains the register number to disable these messages on boot ?

negativekelvin, 2022-09-23:
Your first picture logs are not from rom bootloader they are from second stage bootloader and app

ESP-Marius, 2022-09-24:
As @negativekelvin points out, the efuse only controls the ROM output. 

For secondary bootloader/app you can take a look at:

- `CONFIG_ESP_CONSOLE_NONE`
- `CONFIG_BOOTLOADER_LOG_LEVEL_NONE`
- `CONFIG_LOG_DEFAULT_LEVEL_NONE`

ifoxbr, 2022-09-26:
I thank you both for the great help, thank you very much.

closed 2022-09-26 by ifoxbr