espressif/esp-idf #9836: Visual Code return an erro code when flasing ESP32-S2-DevKitC1

spetike, 2022-09-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The terminal process "C:\Users\nanos\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe 'c:\Users\nanos\esp32\esp-idf\components\esptool_py\esptool\esptool.py', '-p', 'COM4', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32s2', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', '2MB', '0x1000', 'bootloader/bootloader.bin', '0x10000', 'oneshot_read.bin', '0x8000', 'partition_table/partition-table.bin'" terminated with exit code: 1.

brianignacio5, 2022-09-23:
Hi @spetike  are you using the [esp-idf vscode extension](https://github.com/espressif/vscode-esp-idf-extension) ?

In this case you should open an issue in [here]() with your [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension#troubleshooting).

spetike, 2022-09-23:
Thanks for the suggestions, Brian. I did what a (stupid) beginner would do. I just cleaned and re-installed everything, included VSCode. That solved all the problems, everything is peachy now.

closed 2022-09-23 by spetike