espressif/esp-idf #9837: KeyError: 'idfSelectedId'

jdjingdian, 2022-09-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

install all the dependency

### What is the actual behavior?

KeyError: 'idfSelectedId'

### Steps to reproduce.

1. macOS Ventura with Python3 download from python.org
2. git clone -b v4.3.4 --recursive https://github.com/espressif/esp-idf.git esp-idf-v4.3.4
3. cd esp-idf-v4.3.4
4. ./install.sh

### Build or installation Logs.

```plain
`magicdian@jingdiandeMBP esp-idf-v4.3.4 % ./install.sh 
Detecting the Python interpreter
Checking "python" ...
/Users/magicdian/Downloads/esp-idf-v4.3.4/tools/detect_python.sh: line 16: python: command not found
Checking "python3" ...
Python 3.10.6
"python3" has been detected
Installing ESP-IDF tools
Traceback (most recent call last):
  File "/Users/magicdian/Downloads/esp-idf-v4.3.4/tools/idf_tools.py", line 1852, in <module>
    main(sys.argv[1:])
  File "/Users/magicdian/Downloads/esp-idf-v4.3.4/tools/idf_tools.py", line 1848, in main
    action_func(args)
  File "/Users/magicdian/Downloads/esp-idf-v4.3.4/tools/idf_tools.py", line 1415, in action_install
    targets = clean_targets(args.targets)
  File "/Users/magicdian/Downloads/esp-idf-v4.3.4/tools/idf_tools.py", line 1078, in clean_targets
    export_targets_to_idf_env_json(targets_from_tools_json)
  File "/Users/magicdian/Downloads/esp-idf-v4.3.4/tools/idf_tools.py", line 1046, in export_targets_to_idf_env_json
    targets = list(set(targets + get_user_defined_targets()))
  File "/Users/magicdian/Downloads/esp-idf-v4.3.4/tools/idf_tools.py", line 1092, in get_user_defined_targets
    if env == idf_env_json['idfSelectedId']:
KeyError: 'idfSelectedId'`
```

dobairoland, 2022-09-23:
Hi @jdjingdian. Thanks for the report. May I ask you that have you run recently the install script for the master branch before switching to v4.3.4?

Could you please share your  `~/.espressif/idf-env.json` file?

jdjingdian, 2022-09-23:
> Hi @jdjingdian. Thanks for the report. May I ask you that have you run recently the install script for the master branch before switching to v4.3.4?
> 
> Could you please share your `~/.espressif/idf-env.json` file?

I guess I clone the master branch and run the install.sh, then clone v4.3.4 in other folders(~/esp-idf-v4.3.4).

`magicdian@jingdiandeMBP esp-idf-v4.3.4 % cat ~/.espressif/idf-env.json
{
    "idfInstalled": {
        "/Users/magicdian/Documents/VSCODE/esp-idf-v5.1": {
            "version": "5.1",
            "path": "/Users/magicdian/Documents/VSCODE/esp-idf",
            "features": [
                "core"
            ],
            "targets": [
                "esp32c2",
                "esp32c6",
                "esp32c3",
                "esp32s2",
                "esp32",
                "esp32s3",
                "esp32h2"
            ]
        }
    }
}`

jdjingdian, 2022-09-23:
just deleted the ~/.espressif folder, install.sh works! Thanks!

dobairoland, 2022-09-23:
Thanks. We will fix this soon. This will be closed by commit with a proper and permanent fix.

AxelLin, 2022-09-23:
I also hit this issue with v4.3.4-50-ge40b55e854 .

AxelLin, 2022-09-28:
@dobairoland  Hope to get the fix for v4.3 soon, I cannot build with v4.3 for several days.

dobairoland, 2022-09-28:
@AxelLin  You don't have to wait for any fix. Just remove `~/.espressif/idf-env.json` and re-run the install script. 

> I also hit this issue with v4.3.4-50-ge40b55e854 .
> And the same issue in v4.4.2-153-g2bce0a19f6 .

The issue is in the `master` branch. Switching from there to any of these release branches causes the issue. The proper fix will arrive soon to all branches. Until then, the above workaround can be used.

dobairoland, 2022-09-29:
Fixes are now in the release branches.

closed 2022-09-29 by dobairoland

zzzh, 2022-10-14:
I think it's a python virtual environment issue.

How I got this error:
1. clone the master branch, and run install.py
2. checkout a stable branch and do submodule update
3. run idf.sh --version  --> this error come out with complains of many python packages mismatch info...
4. then, even get_idf have same error and can't continue.

So, is there a quick way to switch to a new idf version of python virtual environment?
For example, how to switch from idf5.1_py3.8_env to idf4.2_py3.8_env?

gojimmypi, 2022-12-22:
Heads up I encountered this problem today with Espressif ESP-IDF v5 running from a VisualGDB install.  Broke my ability to run the VisualGDB directory of ESP-IDF for all versions. See [sysprogs forum](https://sysprogs.com/w/forums/topic/problems-using-visualgdb-espressif-toolchain-from-commandline/).

The [solution](https://github.com/espressif/esp-idf/issues/9837#issuecomment-1255964249) from @jdjingdian worked for me:

> just deleted the ~/.espressif folder

dobairoland, 2022-12-22:
I believe so that the issue has been fixed if you are using an ESP-IDF with the linked patch included.

> just deleted the ~/.espressif folder

In case some of you still looking for a workaround, please be aware that removing `~/.espressif/idf-env.json` is enough. If you remove the whole directory then all toolchains will have to be downloaded again. You don't want that if you have slow Internet connection.

gojimmypi, 2022-12-22:
Thanks for the tip re delete file vs directory. 

I noticed that there was a new ESP-IDF offline _installer_ released just an hour ago:

https://github.com/espressif/idf-installer/releases/tag/offline-5.0

Does this contain the fix regarding `idf-env.json`?

Prathamesh08glovatrix, 2023-01-19:
~/.espressif
can anyone please tell me where exactly this folder is located in Apple mackboom M1?

gojimmypi, 2023-01-19:
@Prathamesh08glovatrix does the Mac have an equivalent `ESP-IDF` environment variable that contains your folder location?

I don't have one to test, but a Mac is a bit Linux-like, no? Perhaps `echo $ESP-IDF` or first `cd $ESP-IDF` and then `pwd`?

Prathamesh08glovatrix, 2023-01-19:
@radimkarnis thanks bro its gets install perfectly. love u