espressif/esp-idf #9838: PROBLEMS with new lines in monitor

ecobdoor, 2022-09-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v4.4.2-dirty

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

ESP32 dev module

### Power Supply used.

USB

### What is the expected behavior?

I hope than the console gives the same display in esp-idf monitor, arduino_ide & putty

### What is the actual behavior?

esp-idf monitor doesn't display correctly the lines jumps. But arduino_ide terminal & putty display them correctly

### Steps to reproduce.

1. idf.py menuconfig for arduino components (setup - loop) with single \n end of line, 1000 Hz, etc...
2. Compiling with esp-idf or Arduino_ide or eclipse sloeber are right
3. Flashing any build by any bootloader but the behavior is the same. Look at the screenshots.
4. In all cases esp-idf monitor, after a Ctrl-q Ctrl-s the buffered lines are displayed correctly. So the problem seems to be in monitor...

### Debug Logs.

```plain
No error in log
```

### More Information.

Not really more  information. I'm a retired engineer running Ubuntu 22.04 16 GB ram with Eclipse in a VM machine. 
- ESP-IDF v4.4.2-dirty
- Arduino 1.8.15 vs Esp32 2.0.4

The behavior doesn't depend of USB speed  115200 or 2000000 baudrate. I run menuconfig for:

- CONFIG_ESPTOOLPY_MONITOR_BAUD
- Choice of CRLF, CR, LF
- And others parameters to obtain runtime tasks statistics

I note than printf doesn't like lines beginning with \n, maybe a strange bug in esp-idf monitor or in the UART core output. In the application, it seems to me preferable to start the new terminal lines with \n to possibly complete them with more details when it is useful.

Here 3 terminal screen copy. The esp-idf one has been shot just after a Ctrl-Q (good lines before) & stopped after a Ctrl-S (bad lines again).

[image]

[image]

[image]

Using melt I don't see significant problems in my sdkconfig

An idea??? Not cool to swap from esp-idf & others terminals

dobairoland, 2022-09-23:
Hi @ecobdoor. Please help me a little because I struggle to understand what is the issue. 

What is that Ctrl-Q Ctrl-S key combination? Is it for an IDE? It is certainly not for IDF Monitor.

If you are using ESP-IDF with an IDE then could you try the same without an IDE in a terminal where ESP-IDF was set up with `install.sh` and `. export.sh`?

If you are still experiencing this issue, then can you please give a minimal example about what you are trying to print (C code) and what is appearing in IDF Monitor? The example in the screenshots are too complex to understand.

ecobdoor, 2022-09-25:
Entering the Ctrl-Q Ctrl-S keys in a terminal (the name of this quite standard protocol is XON XOFF see https://en.wikipedia.org/wiki/Software_flow_control). It is used to block or continue the flow of lines (in the monitor window, the printer or other peripherals...) when they arrive too quickly (either for human reading, printer output or others...).
I'm embarrassed to write a minimal code because I'm committed to a very large project (I'm going to do it this because I noticed the problem on simpler examples like [runstats](https://github.com/espressif/esp-idf/tree/master/examples/system/freertos/real_time_stats)).
It seems that monitor doesn't like printf("\n\n") (only outputs one) or lines starting with "\n".
It may be a monitor bug since it correctly displays the lines once buffered after an XOFF and then displayed when the XON is received.
It could also be a problem with sdkconfig. I'm not sure I understand all the print job configuration parameters in compatibility with the Arduino components environment.

ecobdoor, 2022-09-25:
I write the singlest example I can:
=============================================================

#if !defined(__cplusplus)
#error >>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>C++ compiler required.
#endif
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wunused-variable"
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wunused-function"  
#include "Arduino.h"
///////////////////////////////////////////////////////////////////////
char* chn="beware";
int cnt=0;
///////////////////////////////////////////////////////////////////////
void setup(){
//	Serial.begin(115200);
	Serial.begin(2000000);
	delay(1000);
}
///////////////////////////////////////////////////////////////////////
void loop(){
	printf("\n\n %i: '",cnt); // HERE only ONE \n !!!
	for(size_t i=0;i<rand() % 10;i++){
		int len=strlen(chn)+rand() % 10;
		printf("[%*s(%i)]",len,chn,len);
	}
	printf("'");
	cnt++;
}
///////////////////////////////////////////////////////////////////////
=============================================================
Why only one \n appears ?
What happens for line 18391? Why WDT watchdog?
[image]

Good luck, sincerely yours, Eric.

dobairoland, 2022-09-27:
Hi @ecobdoor. Thank you for the additional information. Please open an issue in https://github.com/espressif/arduino-esp32. I'm not able to help you. Sorry.

closed 2022-09-27 by dobairoland