espressif/esp-idf #9841: [IDF 5.0] esp_partition_mmap failed on esp32s3 with big partition(12MB) when psram is enabled.

inapp123, 2022-09-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0-beta1-641-gc321739074

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom board with ESP32S3-WROOM-1-N16R8 module

### Power Supply used.

USB

### What is the expected behavior?

Partition being mapped successfully

### What is the actual behavior?

It crash with ESP_ERR_NO_MEM

### Steps to reproduce.

Code can be found [here](https://github.com/inapp123/esp32s3-mmap-test)

1. use `idf.py menuconfig` to enable psram (Octal psram) and change serial flasher - flash size to 16M to match module specification.
2. config partition table to add a new custom partition with size 12MB, labeled "asset".
3. build and flash the code.

the main part of code is:
```c
    const esp_partition_t* part = esp_partition_find_first(0x40, 0x00, "asset");
    if (part == NULL) {
        ESP_LOGE(TAG, "Failed to find asset partition");
        abort();
    }    
    spi_flash_mmap_dump();
    ESP_LOGI(TAG, "Found asset partition at offset 0x%" PRIx32 " with size %" PRIx32, part->address,part->size);

    uint8_t* mmaped_ptr;
    spi_flash_mmap_handle_t handle;
    ESP_ERROR_CHECK(esp_partition_mmap(part, 0, part->size, SPI_FLASH_MMAP_DATA, &(mmaped_ptr),&handle));
    ESP_LOGI(TAG, "Mounted at offset 0x%" PRIx32 "", (uint32_t)mmaped_ptr);
```

### Debug Logs.

```plain
I (29) boot: ESP-IDF v5.0-beta1-641-gc321739074 2nd stage bootloader
I (29) boot: compile time 23:59:38
I (29) boot: chip revision: V001
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
[... 409 lines cut ...]
page 494: refcnt=1 paddr=32878
page 495: refcnt=1 paddr=32879
page 496: refcnt=1 paddr=32880
page 497: refcnt=1 paddr=32881
page 498: refcnt=1 paddr=32882
page 499: refcnt=1 paddr=32883
page 500: refcnt=1 paddr=32884
page 501: refcnt=1 paddr=32885
page 502: refcnt=1 paddr=32886
page 503: refcnt=1 paddr=32887
page 504: refcnt=1 paddr=32888
page 505: refcnt=1 paddr=32889
page 506: refcnt=1 paddr=32890
page 507: refcnt=1 paddr=32891
page 508: refcnt=1 paddr=32892
page 509: refcnt=1 paddr=32893
page 510: refcnt=1 paddr=32894
page 511: refcnt=1 paddr=32895
I (1494) main: Found asset partition at offset 0x3f1000 with size c00000
I (1504) main: Mounted at offset 0x3c031000
hello world!
hello world!
hello world!
hello world!
```

igrr, 2022-09-25:
It looks like a limitation of the available address space. ESP32-S3 has 32MB of address space available for external Flash and Psram. 8MB is occupied by PSRAM, some space is occupied by the app itself, which leaves less than 12 MB available, so the attempt to mount fails.

AxelLin, 2023-12-16:
> Edit: after looking at the the dump output (sorry that i didn't do this in the first place) there should be enough pages available (192). So this might indeed be a bug!

How is the status of this issue now?

inapp123, 2023-12-25:
> > Edit: after looking at the the dump output (sorry that i didn't do this in the first place) there should be enough pages available (192). So this might indeed be a bug!
> 
> How is the status of this issue now?

With the latest ESP-IDF stable release(5.1.2), it just runs flawlessly...
It might be fixed around 5.0.x, according to toitlang/toit#1555

closed 2023-12-25 by inapp123