espressif/esp-idf #9844: Some files are missing

pkx890, 2022-09-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, UVC based on ESP-IDF driver USB camera, found missing #include "libuVC/libuvC. h", "libuvc_helper. H", "libuVc_adapter.h", Is this a document that you wrote yourself but didn't put into this project? Or is it something you can find elsewhere? Thank you for your reply!

nopnop2002, 2022-09-24:
You need to add idf_component.yml to your project's main directory.

```
$ cat idf_component.yml
## IDF Component Manager Manifest File
dependencies:
  idf: ">=4.4"
  usb_host_uvc: "1.0.0"
  mdns:
    rules:
      - if: "idf_version >= 5.0"

```

tore-espressif, 2022-09-30:
@pkx890 I have nothing more to add to @nopnop2002 's answer. Does it solve your problem? can we close this?

closed 2023-02-16 by Alvin1Zhang