espressif/esp-idf #9845: as a ble servce device，i program base on the  "gatt_server_service_table" example. i change the char vale.but app client side the char value can not change.

giu-wxyao, 2022-09-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-IDF V4.4,TARGET:ESP32

as a ble servce device，i program base on the  "gatt_server_service_table" example. i use a task to  change the char_value.but app client side the i read  char_value ,i found the read value is aways 0x12, 0x13, 0x14, 0x15. 

THE PART CODE AS BELOW:

static uint8_t char_value[4] = {0x12, 0x13, 0x14, 0x15};

static esp_gatts_attr_db_t gatt_db[HRS_IDX_NB] =
    {
        // Service Declaration
        [IDX_SVC] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&primary_service_uuid, ESP_GATT_PERM_READ, sizeof(uint16_t), sizeof(service_uuid), (uint8_t *)&service_uuid}},

        /* Characteristic Declaration */
        [IDX_CHAR_A] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ, CHAR_DECLARATION_SIZE, CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_read_write_notify}},

        /* Characteristic Value */
        [IDX_CHAR_VAL_A] =
            {{ESP_GATT_RSP_BY_APP}, {ESP_UUID_LEN_128, (uint8_t *)&char_uuid_A, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, GATTS_DEMO_CHAR_VAL_LEN_MAX, sizeof(char_value), (uint8_t *)char_value}},

        /* Client Characteristic Configuration Descriptor */
        [IDX_CHAR_CFG_A] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_client_config_uuid, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, sizeof(uint16_t), sizeof(heart_measurement_ccc), (uint8_t *)heart_measurement_ccc}},

        /* Characteristic Declaration */
        [IDX_CHAR_B] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ, CHAR_DECLARATION_SIZE, CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_read}},

        /* Characteristic Value */
        [IDX_CHAR_VAL_B] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_128, (uint8_t *)&char_uuid_B, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, GATTS_DEMO_CHAR_VAL_LEN_MAX, sizeof(char_value), (uint8_t *)char_value}},

        /* Characteristic Declaration */
        [IDX_CHAR_C] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_16, (uint8_t *)&character_declaration_uuid, ESP_GATT_PERM_READ, CHAR_DECLARATION_SIZE, CHAR_DECLARATION_SIZE, (uint8_t *)&char_prop_write}},

        /* Characteristic Value */
        [IDX_CHAR_VAL_C] =
            {{ESP_GATT_AUTO_RSP}, {ESP_UUID_LEN_128, (uint8_t *)&char_uuid_C, ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE, GATTS_DEMO_CHAR_VAL_LEN_MAX, sizeof(char_value), (uint8_t *)char_value}},

};

void sendTask(void *para)
{
    ESP_LOGI("@david", "TASK begin.....");
    vTaskDelay(pdMS_TO_TICKS(1000));
    while (1)
    {
        char_value[0] = rand() % 100;
        char_value[1] = rand() % 100;
        char_value[2] = rand() % 100;
        char_value[3] = rand() % 100;
         vTaskDelay(pdMS_TO_TICKS(500));
    }
}

giu-wxyao, 2022-09-24:
now，if i change ESP_GATT_AUTO_RSP to ESP_GATT_RSP_BY_APP,and send response though the read event by  esp_ble_gatts_send_response(gatts_if, param->read.conn_id, param->read.trans_id,ESP_GATT_OK, &rsp); i can send the data.

esp-cjh, 2022-09-26:
Hi @giu-wxyao，

Bluedroid will malloc a characteristic and copy your characteristic value When you create service table. So that the characteristic value in bluedroid database will not change when you change the characteristic value in your app.

There are two ways:
1 If use ESP_GATT_AUTO_RSP, you should invoke `esp_ble_gatts_set_attr_value()` to change the char value in bluedroid.
2 If use ESP_GATT_RSP_BY_APP, do as you say. But you should send responese according to the handle of the param in read event.

giu-wxyao, 2022-09-27:
thanks @esp-cjh。after you explainning i understand it better。

closed 2022-09-27 by giu-wxyao