espressif/esp-idf #9846: Bootloader size is documented as 0xC00, but is set to 0x1000

axos88, 2022-09-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

With the default partition table offset at 0x8000, and the partition table length of 0xC00, the offset of the first partition should be able to start at 0x8C00, as documented here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html

### What is the actual behavior?

https://github.com/espressif/esp-idf/blob/af28c1fa21fc0abf9cb3ac8568db9cbd99a7f4b5/components/partition_table/gen_esp32part.py#L26

### Steps to reproduce.

Using any example project, modify the partition table so that the first partition starts at offset 0x8C00, and attempt to build

### Build or installation Logs.

```plain
CSV Error: First partition offset 0x8c00 overlaps end of partition table 0x9000
```

### More Information.

Either the documentation needs to be updated, or the https://github.com/espressif/esp-idf/blob/af28c1fa21fc0abf9cb3ac8568db9cbd99a7f4b5/components/partition_table/gen_esp32part.py#L26 needs to be updated according to the actual size. 

1K space  is lost this way.

igrr, 2022-09-25:
@axos88 the error message is misleading here, the actual reason why the partition can't start at 0x8c00 is that we don't support partitions which aren't aligned to SPI Flash erase sector size (0x1000).
We'll take a look at the tool and make changes so that the correct error is reported.

axos88, 2022-09-26:
Ah thanks for updating the title. Yeah I can see why that would be a problem, but it indeed needs to throw a better error message, and it probably needs mention in the partition table documentation

closed 2022-10-24 by espressif-bot

harsha123-123, 2024-12-13:
hey there,
I am getting partition table error
what I mean is :- I am able to dump the code for the device but device is not booting up and it's showing up 
that you have used :>
below I have attached the image 
and please reply me asap
thanks 
harsha
[image]

harsha123-123, 2024-12-13:
here's the screen shot of the above problem
[image]

KonstantinKondrashov, 2024-12-13:
Hi @harsha123-123!
Your issue is unrelated to this topic. Could you consider creating a new one?
To solve your issue you need to increase PARTITION_TABLE_OFFSET in the `idf.py menuconfig`.
```
    config PARTITION_TABLE_OFFSET
        hex "Offset of partition table"
```

Your partition table should not have any partition offsets (leave them empty). The tool will calculate based on the CONFIG_PARTITION_TABLE_OFFSET  and the partition size. See the partition table example https://github.com/espressif/esp-idf/blob/master/components/partition_table/partitions_two_ota.csv