espressif/esp-idf #9847: esp-tls mbedtls - leakage of open sockets when connection to host failed

klaudiusz223, 2022-09-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-WROOM + ETH W5500 SPI

### Power Supply used.

External 3.3V

### What is the expected behavior?

correct closure of a socket after calling  `esp_tls_conn_destroy` in case of a failed connection to the server using `esp_tls_conn_new_sync`   esp-tls and mbedtls.

### What is the actual behavior?

despite calling 
esp_tls_conn_destroy
socket is not closed so that the number of available sockets is exhausted and no other connection can be established
in log can be observed a message
`E (72500) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)`

### Steps to reproduce.

my configuration 
eth module to connect to AWS MQTT using esp-aws-iot,  esp-tls,  mbedtls
tls configuration set to non blocking mode ` non_block = true` in `esp_tls_cfg_t`

the connection to the MQTT server must be blocked to simulate connection problems e.g. with a firewall.
other network traffic such as DNS should work

step 1 - attempt to connect to `esp_tls_conn_new_sync `server (failed -host unreachable)
step 2 - `esp_tls_conn_destroy`

repeat the above until all available sockets are taken and you see the message in the log 
`esp-tls : Failed to create socket (family 2 socktype 1 protocol 0)`
at this point no network connection can be established anymore

### Debug Logs.

```plain
D (47002) esp-tls: [sock=62] Resolved IPv4 address: 18.198.145.228
D (47008) esp-tls: [sock=62] Connecting to server. HOST: qwertyuiop1234-ats.iot.eu-central-1.amazonaws.com, Port: 8883
D (47019) esp-tls: connecting...
D (47191) spi_master: Allocate RX buffer for DMA
D (50022) esp-tls: select() timed out
W (50022) esp-tls: Failed to open new connection in specified timeout
D (52829) esp-tls: host:qwertyuiop1234-ats.iot.eu-central-1.amazonaws.com: strlen 49
D (52898) esp-tls: [sock=63] Resolved IPv4 address: 3.120.71.141
D (52899) esp-tls: [sock=63] Connecting to server. HOST: qwertyuiop1234-ats.iot.eu-central-1.amazonaws.com, Port: 8883
D (52905) esp-tls: connecting...
D (55908) esp-tls: select() timed out
W (55908) esp-tls: Failed to open new connection in specified timeout
D (58432) spi_master: Allocate RX buffer for DMA
D (60156) esp-tls: host:qwertyuiop1234-ats.iot.eu-central-1.amazonaws.com: strlen 49
E (60164) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (60171) esp-tls: Failed to open new connection
D (67124) esp-tls: host:qwertyuiop1234-ats.iot.eu-central-1.amazonaws.com: strlen 49
E (67132) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (67139) esp-tls: Failed to open new connection
D (70771) spi_master: Allocate RX buffer for DMA
D (72113) esp-tls: host:qwertyuiop1234-ats.iot.eu-central-1.amazonaws.com: strlen 49
E (72121) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (72128) esp-tls: Failed to open new connection
D (72492) esp-tls: host:qwertyuiop1234-ats.iot.eu-central-1.amazonaws.com: strlen 49
E (72500) esp-tls: Failed to create socket (family 2 socktype 1 protocol 0)
E (72507) esp-tls: Failed to open new connection
```

### More Information.

Proposed solution
This line 
https://github.com/espressif/esp-idf/blob/1b16ef6cfc2479a08136782f9dc57effefa86f66/components/esp-tls/esp_tls_mbedtls.c#L256

should be replaced with
`if (tls->is_tls && tls->server_fd.fd != -1) {`

This has been successfully tested by me on the above example

The problem was introduced during the resolution of this bug #6163

An incorrect assumption was made, that the socket would be closed by calling `mbedtls_net_free`
Unfortunately, if the connection to the server fails (e.g. host is unreachable), `mbedtls_net_context `  struct is not initialised - the previously opened socket is not assigned to `fd`  thus can't be closed by mbedtls.

socket number is assigned to fd in this place 
https://github.com/espressif/esp-idf/blob/1b16ef6cfc2479a08136782f9dc57effefa86f66/components/esp-tls/esp_tls_mbedtls.c#L67-L73

but it's only invoked after connection is established 
https://github.com/espressif/esp-idf/blob/1b16ef6cfc2479a08136782f9dc57effefa86f66/components/esp-tls/esp_tls.c#L413-L414

Unfortunately the connection will not always be established

Someone is experiencing similar problems https://esp32.com/viewtopic.php?t=29821

ESP-YJM, 2022-09-26:
@klaudiusz223 Thanks for reporting and suggesting the fix. We will raise an internal MR.

AxelLin, 2022-09-28:
v4.3 fix: 3a92e82e9749110ae03ba95e0c5bd16332393fbc

ESP-YJM, 2022-09-28:
@AxelLin Yes, when ctx->fd is -1, mbedtls_net_free() will return directly. But it will make `tls->sockfd` to be -1, and make original socket fd not be closed.

closed 2022-10-08 by espressif-bot