espressif/esp-idf #9848: HTTPD stops working -- error in recv : 11

sergioaiello, 2022-09-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The error is very easy to reproduce and apologies if it has been already detected but I have not found it in the community's journal.
--- test setup ---
ESP32 (4MB, version 1) acting as AP, REST HTTP server and file server (SPIFFS, no memory card)
and 
STA (btw an ESP-camera) acting as HTTP client
--- procedure ---
1) STA starts a file transfer (btw a jpg picture of about 10KB)
2) I force a power down of the STA just unplugging the power cord
The file transfer cannot be completed (I see from LwIP debugging)
3) I power again the STA and the AP's DHCP supplies the IP address (ping works) but STA's HTTP client (also Windows HTTP clients) cannot connect anymore to AP's HTTP server

Vice versa in normal conditions the file transfers are hundreds without problems/errors.
Hope to receive hints. Thanks!
Here attached AP's log as a STA attempts to transmit:
lwip_recv_tcp: netconn_recv err=-3, pbuf=0x0
lwip_recv_tcp: p == NULL, error is "Timeout."!
lwip_recvfrom(57):  addr=192.168.200.12 port=59262 len=-1
W (227909) httpd_txrx: httpd_sock_err: error in recv : 11
lwip_recvfrom(57, 0x3ffc7800, 2048, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x0

ESP-YJM, 2022-10-11:
@sergioaiello I think TCP keepalive could help you to detect the TCP session was broken. Could you please tell me which IDF version you used?

sergioaiello, 2022-10-11:
Hi, the IDF version is 4.4 (which I checked at the time I open the thread) is the latest IDF version. Am I correct?
The keepalive will increase robustness, I agree.
Thank you.

sergioaiello, 2022-10-12:
@ESP-YJM are you providing a solution? Is the TCP keepalive considered a solution or a workaround?

sergioaiello, 2022-10-13:
@ESP-YJM / "Do your AP server only allow one http client to connect at one time?"
No, the AP is used to work with several STAs at time. The test I'm carrying now relates to three APs and randomly one of them is powered OFF and ON. The DHCP performs as expected but the HTTPD stops working for all STAs (TCP root cause expected).
Did you succeed in replicating the case in your lab?

ESP-YJM, 2022-10-13:
@sergioaiello Not yet. Could you please change log level to Debug and offer the whole log.

sergioaiello, 2022-10-14:
@ESP-YJM / "Could you please change log level to Debug and offer the whole log"
[image]
Please, refer to attached picture, as the tracing level I've set for LWIP.
I'm carrying out that way now ...

ESP-YJM, 2022-10-14:
@sergioaiello I think you could only enable socket debug messages and need enable ESP log level to Debug(Component config->Log output->Default log verbosity(Debug)).

sergioaiello, 2022-10-14:
@ESP-YJM 
just read your msg. I'll do.
By now, pls, see the trace I got. The file transfer was interrupted and the HTTPD stopped. I was able to trap the bug at the first try (even the plant setup forced awkly me to unplug a connector instead to go via a switch to power off the station).
I think the event that triggers the file transfer starts at row 288.
Pls, let me know if I found info to your investigation.
[IDFGH-8379_01.txt](https://github.com/espressif/esp-idf/files/9785255/IDFGH-8379_01.txt)

sergioaiello, 2022-10-14:
@ESP-YJM / "enable ESP log level to Debug"
Unfortunately, I got an error at compile time: "binary size 0x70b0 bytes is too large for partition table"
Is this trace important to you?
If so, I'll proceed in changing partition table to reserve more space.
Please, let me know

ESP-YJM, 2022-10-17:
@sergioaiello Enable log level to Debug will make image larger. So you could en-large your ota partition in partition table.

sergioaiello, 2022-10-17:
@ESP-YJM / "So you could en-large your ota partition in partition table"
yes, your statement is true. Thank you.
Can you clarify if you need me to carry out the partition table change in order to compile?
In other words, could you answer to the message sent 3 days ago:
"Is this trace important to you?
If so, I'll proceed in changing partition table to reserve more space.
Please, let me know"
?
Thank you

ESP-YJM, 2022-10-17:
@sergioaiello Yes,  i need debug log trace to check the reason why http client can't connect to server after powered OFF and ON. So in your code, enable ESP log level to Debug(Component config->Log output->Default log verbosity(Debug)) may increase your image size, and you need modify the partition table.

sergioaiello, 2022-10-17:
OK, I'll do asap

Meantime I send you some info.
I checked the size with different verbosity in order to estimate the space increase I should have made.
- size bin file with def verbosity "warning": 786 KB (805,136 bytes)
- size bin file with def verbosity "info": 786 KB (805,136 bytes)
I expected different values. What's wrong here?

The present partition table:
storage,  data, spiffs,  0x10000, 0x1EF000,
ota_0,    app,  ota_0,   0x200000,0x1M,
ota_1,    app,  ota_1,   0x300000,0x1M,
Do you have a suggestion on how to modify it?
Thank you.

sergioaiello, 2022-10-17:
@ESP-YJM / "the reason why http client can't connect to server after powered OFF and ON"
The STA connected to the AP-"file server" goes from ON to OFF.
The attached file might help to figure out how to httpd reaches the fault state. 
[debugLevel-01.txt](https://github.com/espressif/esp-idf/files/9798714/debugLevel-01.txt)

W (227909) httpd_txrx: httpd_sock_err: error in recv : 11
After that httpd stops to communicate.

ESP-YJM, 2022-10-17:
@sergioaiello It seems the http server not receiving complete data, and receive timeout. You need to check whether the client data is sent over. It may be that the length of data sent by the client is different from the total data length `content length`, or it may be that the client has sent data but the server has not received it, resulting in a timeout.  You need to capture packets to find out if the client is not sending data, or the server is not receiving data。

sergioaiello, 2022-10-17:
@ESP-YJM
option1 / "whether the client data is sent over"
my comment: I do not think the problem is on STA side, since no other STA can communicate
option2 / "the length of data sent by the client is different from the total data length"
my comment: I do not think the problem is on STA side, since no other STA can communicate
option3 / "the client has sent data but the server has not received it"
my comment: I do not know how to capture packets on the air. In other words, I should be able to monitor the WLAN traffic since both the STA and the AP are ESP32 based boards. Is that possible? Do you know how?
I can trace the packets from a PC to the AP with wireshark.
Is that acceptable to you?

sergioaiello, 2022-10-17:
The STA-ESP32 received the IP address from DHCP but no HTTP communication.
1)
From the STA's monitor
I got this trace every time the STA tries to send a GET:
*******************************************************************
W (57718) wifi:<ba-add>idx:0 (ifx:0, 34:86:5d:3b:3e:f5), tid:0, ssn:0, winSize:64
E (62714) esp-tls: [sock=54] select() timeout
E (62714) TRANSPORT_BASE: Failed to open a new connection: 32774
E (62714) HTTP_CLIENT: Connection failed, sock < 0
*********************************************************************
Note: from AP monitor there no reaction.
2)
From AP monitor this is the only sequence (cyclic, 20" period)
**************************************************************
I (503797) MainFilSvr4Camra: station 0c:dc:7e:63:51:2c leave, AID=1
lwip_recv_tcp: netconn_recv err=-3, pbuf=0x0
lwip_recv_tcp: p == NULL, error is "Timeout."!
lwip_recvfrom(57):  addr=192.168.200.10 port=49514 len=-1
W (504297) httpd_txrx: httpd_sock_err: error in recv : 11
D (504297) httpd_txrx: httpd_recv_with_opt: error in recv_fn
D (504307) httpd_txrx: httpd_req_recv: error in httpd_recv
D (504307) httpd_txrx: httpd_req_recv: remaining length = 2860
D (504317) httpd_txrx: httpd_recv_with_opt: requested length = 2048
lwip_recvfrom(57, 0x3ffc7888, 2048, 0x0, ..)
lwip_recv_tcp: top while sock->lastdata=0x0
***************************************************************
3)
capturing packets between AP and PC, the TCP connection is established but AP does not answer to the GET request
[GETfromWinBrowser_01.txt](https://github.com/espressif/esp-idf/files/9804765/GETfromWinBrowser_01.txt)
STA-PC sends the GET HTTP request only at the first attempt. Every following page refreshes no GET messages (still the TCP seems ok).
Note: no messages on the AP monitor (neither to the first GET attempt monitored in the trace attached here above)
4)
sending a PING from the STA-PC, the AP regularly replies. I can see ICMP packets on wireshark trace
Again, **no** messages on the AP monitor. (only the msg @ point 2, here above)

pls, let me know for any question
thank you.

ESP-YJM, 2022-10-18:
@sergioaiello I offer a keep-alive solution patch and you can test it with your  ESP STA and ESP AP.
[esp_http_server_keep_alive.zip](https://github.com/espressif/esp-idf/files/9806181/esp_http_server_keep_alive.zip)

ESP-YJM, 2022-10-21:
@sergioaiello Sorry that i use the master branch to generate the patch file. I now re-generate the pacth according to v4.4. Please retry the new patch file.
[esp_http_server_keep_alive_v4.4.zip](https://github.com/espressif/esp-idf/files/9836157/esp_http_server_keep_alive_v4.4.zip)

sergioaiello, 2022-10-21:
@ESP-YJM
thank you.
Looking to the problem I'm facing the master branch could give me a better result indeed. Doesn't it?
I read ("https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#") the master branch contains "the latest ESP-IDF code".

Since I downloaded the ESP-IDF 4.4 
1)
following this procedure
https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md
2)
and using the Espressif server (and not Github)

I'm concerned to follow the procedure reported in this chapter:
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/versions.html#updating-to-master-branch
I'm concerned also because of the issue described in the following POST:
https://www.esp32.com/viewtopic.php?f=13&t=30195

What are the proper operative instructions to migrate from ESP-IDF 4.4 to the present Master Branch?

Another question relates to version:
the original patch you sent was related to Branch Release of 
- Minor Release ESP-IDF 4.4 or
- Bugfix Release 4.4.2?

In case your patch relates to Branch Release of the Bugfix Release 4.4.2 what should be the update path?
Should I apply the following two-steps sequence:
1)
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#updating-to-stable-release
2)
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/versions.html#updating-to-master-branch
?

Step 1 is related to the sentence reported by this page https://github.com/espressif/esp-idf/releases:
"ESP-IDF v4.4.2 is the latest stable release at the time of writing"

thank you

ESP-YJM, 2022-10-21:
@sergioaiello You can manually modify IDF code according to the changes in the patch, and verify it on the version you are currently using. This function has nothing to do with the IDF version.

ESP-YJM, 2022-10-24:
@sergioaiello Thanks for your feeback. You could use release v4.4.2 tag version. This is the current stable version.

sergioaiello, 2022-10-24:
@ESP-YJM
thank you for the confirmation.
I tried already to follow that direction. Please see this note, where I describe a blocking problem:
https://www.esp32.com/viewtopic.php?f=13&t=30195
Could you kindly help me in understanding how to fix the problem I encountered?
Please, clarify. Thank you.

ESP-YJM, 2022-10-27:
@sergioaiello I have reply for https://www.esp32.com/viewtopic.php?f=13&t=30195. And keep this question issue in  ESP32 Forum. For `HTTPD stops working -- error in recv : 11` and the patch i offered, do you have any update?

sergioaiello, 2022-10-27:
@ESP-YJM
The server can setup stable connections to the three STAs.
The HTTP server was able to manage the power down reset of the three STAs meanwhile there was a file transfer on the fly.
Therefore, the STA were able to connect after bootstrap and operate via HTTP. 
The patch works for me

closed 2022-11-02 by espressif-bot