espressif/esp-idf #9852: bt spp acceptor

denruss, 2022-09-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, I found that ESP32 Bluetooth will receive the first bytes of data for a long time, about 500ms, esp32 is the classic Bluetooth mode, using bt_spp_acceptor demo.
Subsequent bytes arrive quickly.
If you pause for about 5 seconds, then the situation repeats itself, the first bytes after the pause arrive with a delay of 500ms, the next ones quickly. 
The situation is similar with Arduino -> SerialToSerialBT.ino

xiongweichao, 2022-09-26:
Hi @denruss ,

If no data packets are sent and received for 5 seconds, it will enter sniff mode. The interval of sniff mode is 800 slots (500 ms).  In addition, I don't think the delay is 500 ms every time, maybe it is within 500 ms many times.

Thanks

denruss, 2022-09-26:
Hi, yes you are right it is within 500ms
but, how I can increase 5 seconds to maybe 60 seconds or  
or disable switch to sniff mode?

xiongweichao, 2022-09-26:
Hi @denruss ,

I think you can add `*settings &= (~HCI_ENABLE_SNIFF_MODE);`  to this [line](https://github.com/espressif/esp-idf/blob/af28c1fa21fc0abf9cb3ac8568db9cbd99a7f4b5/components/bt/host/bluedroid/stack/btm/btm_acl.c#L874).

Thanks

xiongweichao, 2023-02-03:
Hi @denruss ,

May I ask whether the problem has been solved?

Thanks

closed 2023-02-16 by Alvin1Zhang