espressif/esp-idf #9854: 使用esp-idf编译hello_world比andrino中内存占用要大,为何呢?

eAtmApp, 2022-09-25:
### Is your feature request related to a problem?

esp32芯片
esp-idf v4.4.1编译hello_world,
esp_get_free_heap_size()
空闲内存为:297788
如果使用cpp.则为295372
.
.
但是使用arduino-esp32 2.0.5(看说明是idf的v4.4.2的版本)编译:
Serial.println(ESP.getFreeHeap());
空闲内存为:344164

.
.
.
丢失了40KB+...
sdkconfig中编译选项已选择Optimize for size (-OS)
需要怎样配置 才能找回那丢失的40KB+呢,?

ESP-Marius, 2022-09-26:
`ESP.getFreeHeap()` and `esp_get_free_heap_size()` do not actually measure the same number.

`ESP.getFreeHeap()` calls `heap_caps_get_free_size(MALLOC_CAP_INTERNAL)` 
while `esp_get_free_heap_size()` calls `heap_caps_get_free_size( MALLOC_CAP_DEFAULT )`.

Not all memory with `MALLOC_CAP_INTERNAL` has  `MALLOC_CAP_DEFAULT`, e.g. on ESP32 IRAM is internal, but not default. Thus the number of bytes returned by the Arduino API is larger.

For comparison: with the Ardunio sdkconfig applied to master, `heap_caps_get_free_size(MALLOC_CAP_INTERNAL)` returns 374956 bytes.

ESP-Marius, 2022-10-18:
Closing this one as the original question has been answered, feel free to reopen if you still think there is any outstanding issue.

closed 2022-10-18 by ESP-Marius