espressif/esp-idf #9857: 如何使用ESP32C3扫描蓝牙信标发射的125kbs的广播信息

qqqq29, 2022-09-26:
我目前使用的使ESP32C3开发板，请问技术专家有没有相关案例可以支持ESP32C3作为主机扫描其他设备发送的125kbs速率的蓝牙广播？

qqqq29, 2022-09-27:
> 可以使用 extended scan，将 scan PHY 设置为 LE Coded PHY，开启扫描，然后在 `ESP_GAP_BLE_EXT_ADV_REPORT_EVT` 事件中可以获取广播信息。
> 
> 具体代码可以参考：examples/bluetooth/bluedroid/ble_50/ble50_security_client

工程师你好，我用仔细阅读了examples/bluetooth/bluedroid/ble_50/ble50_security_client下的mian文件源代码，请问具体该修改哪一个函数的参数才能将PHY设置成LE Coded PHY。
我目前只能看到`esp_ble_gap_conn_params_t phy_coded_conn_params` 函数与phy有关，但是不会设置。请工程师帮助解答。谢谢，感谢您的耐心解答。

esp-cjh, 2022-09-27:
目前的代码已经支持 Coded PHY 扫描，PHY 的设置是这里：
[image]

如果想只在 Coded PHY 上扫描，可以设置为
```
static esp_ble_ext_scan_params_t ext_scan_params = {
    .own_addr_type = BLE_ADDR_TYPE_PUBLIC,
    .filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
    .scan_duplicate = BLE_SCAN_DUPLICATE_DISABLE,
    .cfg_mask = ESP_BLE_GAP_EXT_SCAN_CFG_CODE_MASK,
    .coded_cfg = {BLE_SCAN_TYPE_ACTIVE, 40, 40},
};
```

qqqq29, 2022-09-28:
> 目前的代码已经支持 Coded PHY 扫描，PHY 的设置是这里： [image]
>
> 如果想只在 Coded PHY 上扫描，可以设置为
>
> ```
> static esp_ble_ext_scan_params_t ext_scan_params = {
>     .own_addr_type = BLE_ADDR_TYPE_PUBLIC,
>     .filter_policy = [...]

工程师你好，按照你的操作教程，我已经将板子运行在Coded PHY上，并使用监视器监视模块的运行日志。日志如下：
[image]
遇到的问题是：模块并没有输出Coded PHY扫描的情况，我目前能够保证我在模块接收范围内有125k的设备正在发送广播包（使用频谱仪可以看到发射）。请问这种情况是需要修改`ESP_GAP_BLE_EXT_ADV_REPORT_EVT`中的代码吗？如果需要修改请工程师指明。 若无需修改代码的话请问工程师是哪里有问题，该如何尝试更改？

esp-cjh, 2022-09-28:
我用 ESP32C3 发 125K 广播包，ESP32C3 进行扫描，是可以扫到的：
[image]

你可以试一下用 nRF connect app 进行扫描，如果你的设备正确发出了广播包，在 app 上是可以看到的。

qqqq29, 2022-09-28:
> 我用 ESP32C3 发 125K 广播包，ESP32C3 进行扫描，是可以扫到的： [image]
> 
> 你可以试一下用 nRF connect app 进行扫描，如果你的设备正确发出了广播包，在 app 上是可以看到的。

谢谢你的解答，能否给一份你用的来发射125k广播包的程序案例。

esp-cjh, 2022-09-28:
这是我测试时做的改动，发广播用的是 ble50_security_server，扫描用的是 ble50_security_client。
[0001-debug-for-coded-phy-adv.zip](https://github.com/espressif/esp-idf/files/9663682/0001-debug-for-coded-phy-adv.zip)

closed 2024-02-27 by Alvin1Zhang