espressif/esp-idf #9859: esp32s2 host msc

horsa1, 2022-09-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf4.4.1

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### Development Kit.

idf4.4.1

### Power Supply used.

USB

### What is the expected behavior?

使用HOST MSC 例程，在没有插入U盘的时候，以5s的间隔不断的初始化 ---逆初始化USB host msc ,  hostmsc正常安装卸载 

### What is the actual behavior?

使用HOST MSC 例程，在没有插入U盘的时候，以5s的间隔不断的初始化 ---逆初始化USB host msc ,   一段时间后内存分配失败 宕机

### Steps to reproduce.

使用msc例子修改的文件如下
...
[msc_test.txt](https://github.com/espressif/esp-idf/files/9645401/msc_test.txt)

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d082c  PS      : 0x00060933  A0      : 0x800d1ae8  A1      : 0x3fffbb30
0x400d082c: hcd_port_init at D:/esp-idf-v4.4.1/components/usb/hcd.c:1342 (discriminator 2)

A2      : 0x3fff9b94  A3      : 0x3fffbb60  A4      : 0x3fffbb70  A5      : 0x00000000
A6      : 0x3ffdbf94  A7      : 0x3f08b80c  A8      : 0x800d0824  A9      : 0x3f08b818
A10     : 0x00000001  A11     : 0xffffffff  A12     : 0x00000108  A13     : 0x00000000
A14     : 0x3fe8c584  A15     : 0x00000003  SAR     : 0x00000016  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000034  LBEG    : 0x00000108  LEND    : 0x00000000  LCOUNT  : 0x400273ed
0x400273ed: _xt_user_exc at D:/esp-idf-v4.4.1/components/freertos/port/xtensa/xtensa_vectors.S:633

Backtrace:0x400d0829:0x3fffbb300x400d1ae5:0x3fffbb60 0x400ccec2:0x3fffbba0 0x400b2603:0x3fffbbf0
0x400d0829: hcd_port_init at D:/esp-idf-v4.4.1/components/usb/hcd.c:1342 (discriminator 2)

0x400d1ae5: hub_install at D:/esp-idf-v4.4.1/components/usb/hub.c:890

0x400ccec2: usb_host_install at D:/esp-idf-v4.4.1/components/usb/usb_host.c:416

ELF file SHA256: aa05362dc48bd008
```

horsa1, 2022-09-26:
static port_t *port_obj_alloc(void)
{
	printf("---------------heap info start----------------\n");
	heap_caps_print_heap_info(MALLOC_CAP_DMA);
    port_t *port = calloc(1, sizeof(port_t));
    usbh_hal_context_t *hal = malloc(sizeof(usbh_hal_context_t));
    void *frame_list =  heap_caps_aligned_calloc(USBH_HAL_FRAME_LIST_MEM_ALIGN, FRAME_LIST_LEN,sizeof(uint32_t), MALLOC_CAP_DMA);
    SemaphoreHandle_t port_mux = xSemaphoreCreateMutex();
    if (port == NULL || hal == NULL || frame_list == NULL || port_mux == NULL) {
    	printf("port:%p\nhal:%p\nframe_list:%p\nport_mux:%p\n",port,hal,frame_list,port_mux);
        free(port);
        free(hal);
        free(frame_list);
        if (port_mux != NULL) {
            vSemaphoreDelete(port_mux);
        }
        return NULL;
    }
    printf("---------------heap info----------------------\n");
    heap_caps_print_heap_info(MALLOC_CAP_DMA);
    port->hal = hal;
    port->frame_list = frame_list;
    port->port_mux = port_mux;
    return port;
}
    void *frame_list =  heap_caps_aligned_calloc(USBH_HAL_FRAME_LIST_MEM_ALIGN, FRAME_LIST_LEN,sizeof(uint32_t)；
最后是这个分配失败了，像是内存未释放掉

closed 2022-09-27 by horsa1

tore-espressif, 2022-09-30:
@horsa1 is the issue really resolved?

esp-saurabhbansal, 2023-04-14:
Hello,

Just for your information, We have recently released USB Device MSC driver. Please check if it is useful to you. We highly appreciate the feedback.

The driver makes ESP based device recognizable by USB-hosts as a USB Mass Storage Device.
It either allows the embedded application to access the partition or Host PC accesses the partition over USB MSC. They can't be allowed to access the partition at the same time.

The driver supports storage media of two types:
1. SPI Flash
2. SD MMC Card

Driver can be found [here](https://github.com/espressif/idf-extra-components/blob/master/usb/esp_tinyusb/tusb_msc_storage.c).
Sample application can be found [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc).

Thanks.