espressif/esp-idf #9861: ESP-IDF V5 flash device bug:AttributeError: module 'esptool' has no attribute '_main'

kurau1, 2022-09-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.0(release branch)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### What is the expected behavior?

build project successfully and flash device successfully 

### What is the actual behavior?

build project successfully but fail to flash device.

### Steps to reproduce.

1. reinstall windows
2. install vscode IDE(don't install GIT and python, just install vscode)
3. install ESP-IDF exploer extension
4. 'view' - 'command palette' - 'ESP-IDF : Configure ESP-IDF extension' - 'select download server:Espressif' - 'select ESP-IDF versions: release/v5.0(release branch) - 'Enter ESP-IDF container directory:F:\ESP32\idf' - 'Enter ESP-IDF Tools directory(IDF_TOOLS_PATH):F:\ESP32\tool'
5. after installing successful,  'view' - 'command palette' - 'ESP-IDF : Show example project'  , then choosing a project and creating a project using example project.
6. press button "ESP-IDF build project", then build successfully. terminal print:
```
Total sizes:
Used stat D/IRAM:   55243 bytes ( 133173 remain, 29.3% used)
      .data size:    7504 bytes
      .bss  size:    2152 bytes
      .text size:   44560 bytes
   .vectors size:    1027 bytes
Used Flash size :  106835 bytes
      .text     :   77555 bytes
      .rodata   :   29024 bytes
Total image size:  159926 bytes (.bin may be padded larger) 
```
7.press button "ESP-IDF flash device", but fail. terminal print:
```
Traceback (most recent call last):
  File "f:\ESP32\idf\esp-idf\components\esptool_py\esptool\esptool.py", line 22, in <module>
    esptool._main()  # type: ignore
AttributeError: module 'esptool' has no attribute '_main'

 *  The terminal process "f:\ESP32\tools\python_env\idf5.0_py3.8_env\Scripts\python.exe 'f:\ESP32\idf\esp-idf\components\esptool_py\esptool\esptool.py', '-p', 'COM8', '-b', '460800', '--before', 'default_reset', '--after', 'hard_reset', '--chip', 'esp32s2', 'write_flash', '--flash_mode', 'dio', '--flash_freq', '80m', '--flash_size', '2MB', '0x1000', 'bootloader/bootloader.bin', '0x10000', 'hello_world.bin', '0x8000', 'partition_table/partition-table.bin'" terminated with exit code: 1. 
```

I using the same step to download release/v4.4(release branch). it can work and flash device successfully.

...

### More Information.

I using the same step to download release/v4.4(release branch). it can work and flash device successfully.

i reinstall windows and VScode so many times, now i just think it is a bug. i can't solve. i try so many times, it's a torture!

i saw other issues about `AttributeError: module 'esptool' has no attribute '_main'` , they reinstall vscode and then work successfully. but i can't even reinstall windows. i have no idea. 

if you have any suggestion, please reply me. i am waiting forever!

kurau1, 2022-09-29:
emmm, powershell can flash device.... maybe i should think that there is something wrong in vscode.

Delzye, 2022-10-12:
I have the same issue while trying to upgrade my esp-idf version.
Running the same script with a relative path (cd'ing into the "esptool" directory and running python .\esptool.py) works, with an absolute path (python ...\esptool\esptool.py) it fails with the "no _main()" error.

HX2003, 2022-11-05:
It appears that changing ..\esp-idf\components\esptool_py\esptool\esptool.py from `    sys.path.remove(os.path.dirname(os.path.realpath(__file__)))  # do not import this script` to `    sys.path.remove(os.path.dirname(os.path.abspath(__file__)))  # do not import this script` fixes my problem

dobairoland, 2022-11-07:
Could you please share the output of  `python -c "import sys; print(sys.path)"`? 

I'd like to understand how the above workaround helps so I could submit a proper patch.

By any chance, does it also solve the issue if you remove that entire line starting with `sys.path.remove....`?

HX2003, 2022-11-08:
The relevant output is `c:\Users\hx2003\esp\esp-idf\components\esptool_py\esptool`, so it imports itself. 

Using `os.path.abspath(__file__)` gives `c:\Users\hx2003\esp\esp-idf\components\esptool_py\esptool\esptool.py`,
while using `os.path.realpath(__file__)` gives `C:\Users\hx2003\esp\esp-idf\components\esptool_py\esptool\esptool.py`. So using abspath 'solves' it by chance, but there maybe be other consequences, so its probably not the right solution.

From Python 3.8 onwards, realpath gives uppercase letters which breaks it. https://github.com/python/cpython/issues/84548

dobairoland, 2022-11-11:
Thank you @HX2003 for the help! That makes sense. It also makes a proper cross-platform solution to be complex. Instead, I'm proposing to avoid import.

dobairoland, 2022-11-11:
[0001-Tools-Fix-esptool-wrappers-by-avoiding-importing-the.patch.txt](https://github.com/espressif/esp-idf/files/9990151/0001-Tools-Fix-esptool-wrappers-by-avoiding-importing-the.patch.txt)

If anyone would be kind to help test the proposed fix then it is attached as patch. Thanks!

xiaolongba, 2022-11-12:
> [0001-Tools-Fix-esptool-wrappers-by-avoiding-importing-the.patch.txt](https://github.com/espressif/esp-idf/files/9990151/0001-Tools-Fix-esptool-wrappers-by-avoiding-importing-the.patch.txt)
> 
> If anyone would be kind to help test the proposed fix then it is attached as patch. Thanks!

could you please show me how to apply this patch?

bigcat26, 2022-11-15:
> abspath

I have encountered the same issue, you saved my day.

bigcat26, 2022-11-15:
> It appears that changing ..\esp-idf\components\esptool_py\esptool\esptool.py from ` sys.path.remove(os.path.dirname(os.path.realpath(__file__))) # do not import this script` to ` sys.path.remove(os.path.dirname(os.path.abspath(__file__))) # do not import this script` fixes my problem

I have encountered the same issue, you saved my day.

yaniot, 2022-11-21:
> I have encountered the same issue, you saved my day.

> 似乎将 ..\esp-idf\components\esptool_py\esptool\esptool.py 从` sys.path.remove(os.path.dirname(os.path.realpath(__file__))) # do not import this script`更改为` sys.path.remove(os.path.dirname(os.path.abspath(__file__))) # do not import this script`解决了我的问题

I have encountered the same issue, you saved my day.

closed 2022-11-22 by espressif-bot

EspritElf, 2022-11-25:
I am able to flash using the icon again, much thanks to HX2003 and dobairoland.