espressif/esp-idf #9862: [IDF 4.4.2] ESP32-S3 SPI Master Timing Problem While Reading SPI Data

mikucukyilmaz, 2022-09-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**Operating System used.**
Windows

**How did you build your project?**
Command line with idf.py

**If you are using Windows, please specify command line type.**
PowerShell

**Development Kit.**
Custom board with ESP32-S3FN8 IC.

**What is the expected behavior?**
Reading correct SPI data below 16 MHz SPI Clock Frequency.

**What is the actual behavior?**
When SPI Clock is set below 16 MHz, the data read is wrong.

In the pictures below I uploaded serial decoder output for different SPI frequencies, as you can see DW1000 IC outputs 0xDECA3001 after sending 0x00. My problem is even though scope verifies the data on the MOSI line is correct, below 16 MHz i received corrupted data which I assume is due to a timing problem.

4 Byte read at 2 MHz,  ESP32-S3 reads 0xBC950360 rather than 0xDECA3001 
> I (383) app_main: Initializing bus SPI2 [2] MHz...
> I (448) Spi TX:: 00
> I (448) Spi RX:: fe 60 03 95 bc
> I (1448) Spi TX:: 00
> I (1448) Spi RX:: fe 60 03 95 bc

4 Byte read at 8 MHz,  ESP32-S3 reads 0xBC950360 rather than 0xDECA3001 
> I (383) app_main: Initializing bus SPI2 [8] MHz...
> I (448) Spi TX:: 00
> I (448) Spi RX:: fe 60 03 95 bc
> I (1448) Spi TX:: 00
> I (1448) Spi RX:: fe 60 03 95 bc

> I (383) app_main: Initializing bus SPI2 [16] MHz...
> I (448) Spi TX:: 00
> I (448) Spi RX:: ff 30 01 ca de
> I (1448) Spi TX:: 00
> I (1448) Spi RX:: 7f 30 01 ca de

> I (383) app_main: Initializing bus SPI2 [20] MHz...
> I (448) Spi TX:: 00
> I (448) Spi RX:: ff 30 01 ca de
> I (1448) Spi TX:: 00
> I (1448) Spi RX:: 7f 30 01 ca de

After 16MHz I can correctly read the data actually on the bus. Input Delay seems doesn't have any effect on ESP32-S3. 

[image]
[image]
[image]
[image]

mikucukyilmaz, 2022-09-26:
Comparison of the data that i read and data expected is below as binary array. My whole data is bit shifted

0x7F 3001 CADE Lsh 1 = 0xFE 6003 95BC

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/MIK/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/MIK/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#0563C1" vlink="#954F72">

 
--
Data Readed
FE | 60 | 3 | 95 | BC
11111110 | 01100000 | 00000011 | 10010101 | 10111100
Data Expected
7F | 30 | 1 | CA | DE
01111111 | 00110000 | 00000001 | 11001010 | 11011110

</body>

</html>

mikucukyilmaz, 2022-09-28:
I find that issue is not related to ESP32-S3. Setting SPI_EDGE bit of DW1000 to 1 fixed that problem.
> **SPI_EDGE (reg:04:00 bit:10)**
SPI data launch edge. This bit allows the system integrator the ability to control the
launch edge used for SPI data from the DW1000 on the MISO SPI data output line. This
may be used to select the MISO output operation most suitable to the target system.
When SPI_EDGE is 0 the DW1000 uses the sampling edge to launch MISO data. This
setting should give the highest rate operation. When SPI_EDGE is 1 the DW1000 uses the
opposite edges to launch the data. This setting may give a more robust operation.

closed 2022-09-28 by mikucukyilmaz

oursland, 2023-04-27:
Anyone else encountering this issue, here's the process to get it to work:

1. Reset the DW1000.
2. Configure SPI for 16 MHz to 20 MHz for high-speed operation.
3. Set the `SPI_EDGE` bit of `SYS_CFG` register.
4. Configure SPI for 1 MHz to 3 MHz for low-speed operation.
5. Perform rest of the initialization and configuration process, including LDE loading which requires low-speed SPI operation.
6. Configure SPI for high-speed operation.
7. Device is now ready.

It's key to ensure that steps 1, 2, and 3 are performed in order to ensure that the low-speed operations necessary for device initialization will be successful.

forqzy, 2023-05-02:
> Anyone else encountering this issue, here's the process to get it to work:
>
> 1. Reset the DW1000.
> 2. Configure SPI for 16 MHz to 20 MHz for high-speed operation.
> 3. Set the `SPI_EDGE` bit of [...]

I meet the same problem, and found that if I set the SPI mode to 1, in some board it will return correct value but then Tx not ready.  And I do find after finish the low speed configuration, then looks it back to normal.

mmrein, 2024-09-26:
I just got bitten by this too. Found even safer solution on Qorvo forums: https://forum.qorvo.com/t/dwm1000-returns-wrong-vaules/5745/3 

0. HW reset 
1. "Write the 16bit value `0x1600` to the `SYS_CFG` (0x04) register. That will set the SPI_EDGE bit (and will not touch other bits in the SYS_CFG after the startup)."

Do not read-modify-write as your read value might be incorrect (especially on speeds lower than 16MHz in case of S3). Just execute `write16bit(SYS_CFG, 0x1600);`. 

With this approach you can even use lower SPI speed < 3MHz from the beginning.