espressif/esp-idf #9865: NVS initialisation error while running ble_compatibility_test on stable (v4.2.2)

ljden, 2022-09-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Test runs without error

### What is the actual behavior?

Error logged

### Steps to reproduce.

Install esp-idf v4.2.2
```bash
idf.py set-target esp32
idf.py -p /dev/ttyUSB0 flash monitor
```

### Debug Logs.

```plain
vbox@vbox:~/esp/esp-idf/examples/bluetooth/bluedroid/ble/ble_compatibility_test$ idf.py -p /dev/ttyUSB0 flash monitor
Executing action: flash
Running ninja in directory /home/vbox/esp/esp-idf/examples/bluetooth/bluedroid/ble/ble_compatibility_test/build
Executing "ninja flash"...
[1/5] cd /home/vbox/esp/esp-idf/examples/bluetooth/bluedroid/ble/ble_compatibility_test/build/esp-idf/esptool_py && /home/vbox/.espressi...able/partition-table.bin /home/vbox/esp/esp-idf/examples/bluetooth/bluedroid/ble/ble_compatibility_test/build/ble_compatibility_test.bin
ble_compatibility_test.bin binary size 0xa8c10 bytes. Smallest app partition is 0x100000 bytes. 0x573f0 bytes (34%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /home/vbox/esp/esp-idf/examples/bluetooth/bluedroid/ble/ble_compatibility_test/build/bootloader/esp-idf/esptool_py && /home/vbox/.espressif/python_env/idf4.4_py3.10_env/bin/python /home/vbox/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/vbox/esp/esp-idf/examples/bluetooth/bluedroid/ble/ble_compatibility_test/build/bootloader/bootloader.bin
Bootloader binary size 0x6330 bytes. 0xcd0 bytes (11%) free.
[2/3] cd /home/vbox/esp/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH="/home/vbox/esp/esp-idf" -D SERIAL_TOOL="/home/vbox/...idf/examples/bluetooth/bluedroid/ble/ble_compatibility_test/build" -P /home/vbox/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py esp32 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 ble_compatibility_test.bin 0x8000 partition_table/partition-table.bin
esptool.py v3.3.2-dev
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WDR2-V3-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 44:17:93:ba:b7:80
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00007fff...
[... 121 lines cut ...]

E (1739) BT_OSI: config_save: NVS not initialized. Call nvs_flash_init before initializing bluetooth.
E (1749) BT_OSI: config_save, err_code: 0x2

E (1829) BT_OSI: config_save: NVS not initialized. Call nvs_flash_init before initializing bluetooth.
E (1829) BT_OSI: config_save, err_code: 0x2

E (1839) BT_OSI: config_save: NVS not initialized. Call nvs_flash_init before initializing bluetooth.
E (1849) BT_OSI: config_save, err_code: 0x2

E (1849) BT_OSI: config_save: NVS not initialized. Call nvs_flash_init before initializing bluetooth.
E (1859) BT_OSI: config_save, err_code: 0x2

E (1869) BT_OSI: config_save: NVS not initialized. Call nvs_flash_init before initializing bluetooth.
E (1879) BT_OSI: config_save, err_code: 0x2

I (1899) BLE_COMP: create attribute table successfully, the number handle = 11

I (1909) BLE_COMP: (0) ***** advertising start successfully *****
```

### More Information.

First time using esp-idf so I'm not sure where to begin looking for further debugging. Pointers are welcome
Same error message in https://github.com/espressif/esp-idf/issues/5566 closed with no actionable fix

esp-cjh, 2022-09-27:
You can fixed this bug by commenting out line 630 of code in ble_compatibility_test.c.
`//ESP_ERROR_CHECK(nvs_flash_erase());`

We will fix it ASAP.

ljden, 2022-09-27:
Thanks, what's the underlying issue?

esp-cjh, 2022-09-27:
If the nvs is initialized, `nvs_flash_erase()` first de-initializes it. Afterwards, the nvs has to be initialized again to be used.

ljden, 2022-09-27:
Cheers, thought it might have been something like that

closed 2023-05-24 by Alvin1Zhang