espressif/esp-idf #9866: esp_bt_controller_disable hangs in eternal loop when called from core 0

ztefanjo, 2022-09-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

Battery

### What is the expected behavior?

It should be possible to call any espressif-function from any of the cores. The "PRO_CPU"/"APP_CPU" concept has as I understand it been left.

### What is the actual behavior?

When calling esp_bt_controller_disable from core 0, it sometimes hangs in `while (!btdm_power_state_active()){}`. The while loop is waiting for something to happen in one of the internal tasks, which obviously can not happen as these are pinned to core 0.
Solutions to this could be:
1. Do a small sleep in esp_bt_controller_disable.
2. Do not pin internal tasks to any core
3. Document that calls to all or certain interface functions are not allowed from core 0.

### Steps to reproduce.

Happens sometimes when calling esp_bt_controller_disable from core 0.

ztefanjo, 2022-10-06:
As a work-around, we have now pinned all our application threads to core 1. Is this a proper work-around? As we do not know if there are more similar constructions in the ESP-IDF, we played it safe and pinned all threads. But from a performance perspective, it would of course be good if the application could use both cores.

wmy-espressif, 2022-10-09:
Hi @ztefanjo I  need more information on reproducing this issue. I am not sure what is the relationship between the dead loop and the core used to invoke `esp_bt_controller_disable()` yet.  So currently I cannot answer your question or whether your workaround is a all right or risky.

1. To which core do you pin Bluetooth Controller task and Bluedroid tasks? Please take a look at your sdkconfig file and check the value of `CONFIG_BT_CTRL_PINNED_TO_CORE` and `CONFIG_BT_BLUEDROID_PINNED_TO_CORE`.
2. Can this issue be stably reproduced? How often does this issue occur?
3. Do you have a minimum project to reproduce this issue?

ztefanjo, 2022-10-10:
Hi @wmy-espressif ,
1. All Espressif threads are using their default core, we have not altered CONFIG_BT_CTRL_PINNED_TO_CORE or CONFIG_BT_BLUEDROID_PINNED_TO_CORE.
2. It was possible to stably reproduce this error (100%) by calling esp_bt_controller_disable from core 0 directly after starting BLE. Pinning the application thread to core 1 instead solved the crash.
3. No, not currently. You should be able to see in your code what `btdm_power_state_active` is waiting for, to be able to confirm that it requires execution in core 0.

ztefanjo, 2022-10-10:
1. It was seen by halting the debugger before the watchdog resets the MCU.
2. `btdm_power_state_active()` waits for a global variable to be set by BTC task? Yes, I agree that the calling thread if running on core 0 should be pre-empted by BTC, but of some reason, this does not seem to happen then.
3. Yes, we are using light sleep mode. [sdkconfig.defaults.txt](https://github.com/espressif/esp-idf/files/9744648/sdkconfig.defaults.txt) is attached. Would it be possible for you to share a patch with the bugfix?
4. In this case, we are starting up BLE, but then immediately are finding out that BLE should be disabled, so we are calling `esp_bt_controller_disable`, but without call to `esp_bluedroid_disable` and `esp_bt_bluedroid_deinit`. Are these two required to make `esp_bt_controller_disable` to work? It seems to work fine after switching to core 1.

wmy-espressif, 2022-10-11:
Hi @ztefanjo , here are my comments on your response:

> 2. btdm_power_state_active() waits for a global variable to be set by BTC task? Yes, I agree that the calling thread if running on core 0 should be pre-empted by BTC, but of some reason, this does not seem to happen then.

What is the calling thread's task priority? If it has higher priority than controller task then this issue is very likely to happen.

> 3. Yes, we are using light sleep mode. [sdkconfig.defaults.txt](https://github.com/espressif/esp-idf/files/9744648/sdkconfig.defaults.txt) is attached. Would it be possible for you to share a patch with the bugfix?

Thanks for the provided file. The bugfix I mentioned is merged into release/v4.4 in commit SHA(836f9875b10f7242b5c53598a50a4b483de47f75). I can provide a separate patch based on your commit.

> 4. In this case, we are starting up BLE, but then immediately are finding out that BLE should be disabled, so we are calling esp_bt_controller_disable, but without call to esp_bluedroid_disable and [...]

Yes these two functions are required before invoking esp_bt_controller_disable. This is because the host(Bluedroid) can still communicating with controller through HCI messages, the HCI messages may interfere the controller disable procedure. Also, host stack may not be able to work again after controller is disabled and then enabled in this condition.

ztefanjo, 2022-10-11:
2: It is called from the "main_task" which should have prio 1 if it has not inherited a higher priority e.g. from the async_wakeup_request? I can't see the priority in the exclipse debugger unfortunately, but obviously the main_task is getting all the execution time. See attached screenshot from debugger showing relevant tasks.
[image]

3. OK. We are using the released 4.4.2.

4. OK, we will try to rework to deinit bluedroid and bt_bluedroid before _disable. I guess we will need to disable advertisments and tear down existing connections in before this as well, so we will need to do some refactoring.

wmy-espressif, 2022-10-12:
Hi @ztefanjo ,

3. I am sorry that I provided the wrong information. The issue can cause RTC watchdog reset is resolved in v4.4.2. The issue releated to BLE in light sleep mode that is fixed after v4.4.2 is an issue that may cause BLE link to disconnect. As for the patch, we need to update the Bluetooth Controller library `libbtdm_app.a` for ESP32-S3. I know that I have released a patch for you that can optimize the BLE multi-connection performance in coexistence scenario. Do I need to provide a library file based on previous patch? If not, the patch for v4.4.2 is here:  /home/wmy/work/proj/esp-idf-ws/01/esp-idf/components/bt/controller/lib_esp32c3_family/esp32s3/libbtdm_app.a

4. All right, please make the change and stop Bluetooth activities and then disable + deinit Bluedroid, then do disable controller

ztefanjo, 2022-10-12:
@wmy-espressif :
3. Yes, we would need a new "SDK" with the coexistence patch applied as well.
4. Yes.

wmy-espressif, 2022-10-12:
Hi @ztefanjo  here is the updated Bluetooth library based on the coexistence patch 
[libbtdm_app.zip](https://github.com/espressif/esp-idf/files/9762754/libbtdm_app.zip)

I will ask my collegue to synchronize the patch through esp-idf-patch repo for you.

ztefanjo, 2022-10-13:
@wmy-espressif : Thanks for the patch. We have reported an issue in our gitlab space regarding that `btdm_sleep_exit_phase3_wrapper` calls `btdm_sleep_clock_sync` in a never-ending loop, see https://glab.espressif.cn/technical_support/husqvarnagroup/esp-idf-patch/issues/9. 
Do you think that this issue is be fixed by your libbtdm_app patch?

wmy-espressif, 2022-10-14:
Hi @ztefanjo , I don't think the patch can resolve the issue you mentioned. The issue resulted from missing of a hardware state change. I will take a look into the issue.

wmy-espressif, 2023-02-01:
HI @ztefanjo , I see your updates on https://glab.espressif.cn/technical_support/husqvarnagroup/esp-idf-patch/issues/9, and thanks for the information. Can I close this issue?

ztefanjo, 2023-02-01:
Hi @wmy-espressif ,
We have worked around this issue by pinning our tasks to core 1 and also improved our handling of bluedroid/bt according to 4 above, but I guess that the original issue still could happen if calling esp_bt_controller_disable from core 0?
For my side, I consider the problem solved (work-around with only using core 1 is sufficient), so up to you if you want to close the issue or do any changes.

wmy-espressif, 2023-02-01:
Hi @ztefanjo thanks for your update. I've tried to reproduce this issue locally but failed. I think we can close this issue for now, and if you find this issue blocks the project, please feel free to re-open it.

closed 2023-02-02 by ztefanjo