espressif/esp-idf #9868: Configuration descriptor larger than control transfer max length

nopnop2002, 2022-09-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1-dev-862-g09f7589ef2-dirty

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### Development Kit.

A board using Ai-Thinker's ESP-S3-12K

### Power Supply used.

USB

### What is the expected behavior?

I tried this example.
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/uvc

This USB camera(SANWA) work fine.

```
$ sudo v4l2-ctl --all
Driver Info:
        Driver name      : uvcvideo
        Card type        : USB 2.0 Camera: USB Camera
        Bus info         : usb-0000:00:1d.0-1.2
        Driver version   : 5.15.53
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : uvcvideo
        Model            : USB 2.0 Camera: USB Camera
        Serial           :
        Bus info         : usb-0000:00:1d.0-1.2
        Media version    : 5.15.53
        Hardware revision: 0x00000000 (0)
        Driver version   : 5.15.53
Interface Info:
        ID               : 0x03000002
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : USB 2.0 Camera: USB Camera
        Function         : V4L2 I/O
        Flags         : default
        Pad 0x01000007   : 0: Sink
          Link 0x02000010: from remote pad 0x100000a of entity 'Extension 4': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height      : 640/480
        Pixel Format      : 'YUYV' (YUYV 4:2:2)
        Field             : None
        Bytes per Line    : 1280
        Size Image        : 614400
        Colorspace        : Default
        Transfer Function : Default (maps to Rec. 709)
        YCbCr/HSV Encoding: Default (maps to ITU-R 601)
        Quantization      : Default (maps to Limited Range)
        Flags             :
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 640, Height 480
        Default     : Left 0, Top 0, Width 640, Height 480
        Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 640, Height 480, Flags:
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 480, Flags:
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 25.000 (25/1)
        Read buffers     : 0
                     brightness 0x00980900 (int)    : min=-64 max=128 step=1 default=112 value=112
                       contrast 0x00980901 (int)    : min=0 max=64 step=1 default=34 value=34
                     saturation 0x00980902 (int)    : min=1 max=128 step=1 default=57 value=57
                            hue 0x00980903 (int)    : min=-8 max=8 step=1 default=0 value=0
                          gamma 0x00980910 (int)    : min=0 max=8 step=1 default=4 value=4
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
                      sharpness 0x0098091b (int)    : min=0 max=6 step=1 default=4 value=4
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
                                1: Manual Mode
                                3: Aperture Priority Mode
              exposure_absolute 0x009a0902 (int)    : min=1 max=5000 step=1 default=625 value=625 flags=inactive
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1

```

However, this USB camera (Logitech C615) displays the following error:
```
E (2502) HUB: Configuration descriptor larger than control transfer max length
E (2502) HUB: Stage failed: CHECK_SHORT_CONFIG_DESC
```

```
$ sudo v4l2-ctl --all
Driver Info:
        Driver name      : uvcvideo
        Card type        : HD Webcam C615
        Bus info         : usb-0000:00:1d.0-1.2
        Driver version   : 5.15.53
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : uvcvideo
        Model            : HD Webcam C615
        Serial           : 205F2260
        Bus info         : usb-0000:00:1d.0-1.2
        Media version    : 5.15.53
        Hardware revision: 0x00000011 (17)
        Driver version   : 5.15.53
Interface Info:
        ID               : 0x03000002
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : HD Webcam C615
        Function         : V4L2 I/O
        Flags         : default
        Pad 0x01000007   : 0: Sink
          Link 0x02000019: from remote pad 0x100000a of entity 'Extension 5': Data, Enabled, Immutable
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height      : 640/480
        Pixel Format      : 'YUYV' (YUYV 4:2:2)
        Field             : None
        Bytes per Line    : 1280
        Size Image        : 614400
        Colorspace        : sRGB
        Transfer Function : Rec. 709
        YCbCr/HSV Encoding: ITU-R 601
        Quantization      : Default (maps to Limited Range)
        Flags             :
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 640, Height 480
        Default     : Left 0, Top 0, Width 640, Height 480
        Pixel Aspect: 1/1
Selection Video Capture: crop_default, Left 0, Top 0, Width 640, Height 480, Flags:
Selection Video Capture: crop_bounds, Left 0, Top 0, Width 640, Height 480, Flags:
Streaming Parameters Video Capture:
        Capabilities     : timeperframe
        Frames per second: 30.000 (30/1)
        Read buffers     : 0
                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=32 value=32
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=32 value=32
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=64 value=64
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4000 value=4000 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=22 value=22
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=1 value=1
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
                                1: Manual Mode
                                3: Aperture Priority Mode
              exposure_absolute 0x009a0902 (int)    : min=3 max=2047 step=1 default=166 value=166 flags=inactive
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1
                   pan_absolute 0x009a0908 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                 focus_absolute 0x009a090a (int)    : min=0 max=255 step=17 default=51 value=51 flags=inactive
                     focus_auto 0x009a090c (bool)   : default=1 value=1
                  zoom_absolute 0x009a090d (int)    : min=1 max=5 step=1 default=1 value=1
```

### What is the actual behavior?

Logitech C615 displays the following error:
```
E (2502) HUB: Configuration descriptor larger than control transfer max length
E (2502) HUB: Stage failed: CHECK_SHORT_CONFIG_DESC
```

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
![CHECK_SHORT_CONFIG_DESC](https://user-images.githubusercontent.com/6020549/192467495-b650b8b9-0f41-4781-a4c1-0f7a85e3d935.jpg)
```

tore-espressif, 2022-09-27:
`idf.py menuconfig` Component config -> USB -> Largest size (in bytes) of transfers to/from default endpoints

Set the value you need

nopnop2002, 2022-09-27:
@tore-espressif 

Thank you for the reply.

>Component config -> USB-OTG -> Largest size (in bytes) of transfers to/from default endpoints

I changed this value from 1024 to 4096.

uvc_find_device() is pass.
But i get this fail.

```
Device found
DEVICE CONFIGURATION (046d:082c/2) ---
Status: idle
VideoControl:
        bcdUVC: 0x0100
VideoStreaming(1):
        bEndpointAddress: 129
        Formats:
        UncompressedFormat(1)
                  bits per pixel: 16
                  GUID: 5955593200001000800000aa00389b71 (YUY2)
                  default frame: 1
                  aspect ratio: 0x0
                  interlace flags: 00
                  copy protect: 00
                        FrameDescriptor(1)
                          capabilities: 01
                          size: 160x120
                          bit rate: 1536000-4608000
                          max frame size: 38400
                          default interval: 1/15
                          interval[0]: 1/15
                          interval[1]: 1/10
                          interval[2]: 1/7
                          interval[3]: 1/5
                        FrameDescriptor(2)
                          capabilities: 01
                          size: 176x144
                          bit rate: 2027520-6082560
                          max frame size: 50688
                          default interval: 1/15
                          interval[0]: 1/15
                          interval[1]: 1/10
                          interval[2]: 1/7
                          interval[3]: 1/5
        MJPEGFormat(2)
                  bits per pixel: 0
                  GUID: 4d4a5047000000000000000000000000 (MJPG)
                  default frame: 1
                  aspect ratio: 0x0
                  interlace flags: 00
                  copy protect: 00
                        FrameDescriptor(1)
                          capabilities: 01
                          size: 640x480
                          bit rate: 24576000-147456000
                          max frame size: 614400
                          default interval: 1/30
                          interval[0]: 1/30
                          interval[1]: 1/24
                          interval[2]: 1/20
                          interval[3]: 1/15
                          interval[4]: 1/10
                          interval[5]: 1/7
                          interval[6]: 1/5
                        FrameDescriptor(2)
                          capabilities: 01
                          size: 160x120
[... 70 lines cut ...]
                          interval[2]: 1/20
                          interval[3]: 1/15
                          interval[4]: 1/10
                          interval[5]: 1/7
                          interval[6]: 1/5
END DEVICE CONFIGURATION
E (32172) USBH: Dev 1 EP 0 Error
E (32172) Transfer failed: Status 1
E (32172) USBH: Device 1 gone
bmHint: 0001
bFormatIndex: 2
bFrameIndex: 4
dwFrameInterval: 333333
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 61
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 38400
dwMaxPayloadTransferSize: 512
bInterfaceNumber: 3
transfer_status: 1E (32192) example: UVC error: Unknown error

assert failed: app_main main.c:231 (0)
```

nopnop2002, 2022-09-29:
I compared it with:
https://github.com/libuvc/libuvc

"uvc_get_stream_ctrl_format_size" in esp-idf does not work properly.

# Linux
```
uvc_get_stream_ctrl_format_size format=3 640x480 30fps
bmHint: 0001
bFormatIndex: 1
bFrameIndex: 1
dwFrameInterval: 333333
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 61
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 614400
dwMaxPayloadTransferSize: 3060
bInterfaceNumber: 3
Streaming...
Enabling auto exposure ...
 ... full AE not supported, trying aperture priority mode
 ... enabled aperture priority auto exposure mode
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
```

# ESP-IDF
```
I (23712) example: uvc_get_stream_ctrl_format_size FORMAT=3 640x480 FPS=15
Negotiating streaming format failed, trying again...
Negotiating streaming format failed, trying again...
Negotiating streaming format failed, trying again...
Negotiating streaming format failed, trying again...
```

nopnop2002, 2022-09-30:
Mayby libusb_open() issues.

# uvc_print_diag() using Linux

```
DEVICE CONFIGURATION (046d:082c/205F2260) ---
Status: idle
VideoControl:
        bcdUVC: 0x0100
VideoStreaming(1):
        bEndpointAddress: 129
        Formats:
        UncompressedFormat(1)
                  bits per pixel: 16
                  GUID: 5955593200001000800000aa00389b71 (YUY2)
                  default frame: 1
                  aspect ratio: 0x0
                  interlace flags: 00
                  copy protect: 00
                        FrameDescriptor(1)
                          capabilities: 01
                          size: 640x480
                          bit rate: 24576000-147456000
                          max frame size: 614400
                          default interval: 1/30
                          interval[0]: 1/30
                          interval[1]: 1/24
                          interval[2]: 1/20
                          interval[3]: 1/15
                          interval[4]: 1/10
                          interval[5]: 1/7
                          interval[6]: 1/5
                        FrameDescriptor(2)
                          capabilities: 01
                          size: 160x120
                          bit rate: 1536000-9216000
                          max frame size: 38400
                          default interval: 1/30
                          interval[0]: 1/30
                          interval[1]: 1/24
                          interval[2]: 1/20
                          interval[3]: 1/15
                          interval[4]: 1/10
                          interval[5]: 1/7
                          interval[6]: 1/5
                        FrameDescriptor(3)
                          capabilities: 01
                          size: 176x144
                          bit rate: 2027520-12165120
                          max frame size: 50688
                          default interval: 1/30
                          interval[0]: 1/30
                          interval[1]: 1/24
                          interval[2]: 1/20
                          interval[3]: 1/15
                          interval[4]: 1/10
                          interval[5]: 1/7
                          interval[6]: 1/5
                        FrameDescriptor(4)
                          capabilities: 01
                          size: 320x240
                          bit rate: 6144000-36864000
                          max frame size: 153600
                          default interval: 1/30
                          interval[0]: 1/30
                          interval[1]: 1/24
                          interval[2]: 1/20
                          interval[3]: 1/15
                          interval[4]: 1/10
                          interval[5]: 1/7
[... 438 lines cut ...]
                          bit rate: 8110080-48660480
                          max frame size: 202752
                          default interval: 1/30
                          interval[0]: 1/30
                          interval[1]: 1/24
                          interval[2]: 1/20
                          interval[3]: 1/15
                          interval[4]: 1/10
                          interval[5]: 1/7
                          interval[6]: 1/5
                        FrameDescriptor(7)
                          capabilities: 01
                          size: 640x360
                          bit rate: 18432000-110592000
                          max frame size: 460800
                          default interval: 1/30
                          interval[0]: 1/30
                          interval[1]: 1/24
                          interval[2]: 1/20
                          interval[3]: 1/15
                          interval[4]: 1/10
                          interval[5]: 1/7
                          interval[6]: 1/5
END DEVICE CONFIGURATION
```

tore-espressif, 2024-03-05:
@nopnop2002 I'm very sorry for the extra ordinary slow response. As you can see from your logs, the USB camera descriptors differs for HighSpeed (linux) and FullSpeed (esp32-sx).

In latest master branch we merged bunch fixes for USB cameras. Could you try again?

nopnop2002, 2024-03-06:
@tore-espressif 

Thank you for your reply.
I tried it on the latest master branch.

```
$ idf.py --version
ESP-IDF v5.3-dev-2381-g9f4e8eb0cd
```

However, Using the Logitech C615 I get the following error:

```
E (10943) HUB: Configuration descriptor larger than control transfer max length
E (10943) HUB: Stage failed: CHECK_SHORT_CONFIG_DESC
```

nopnop2002, 2024-03-12:
>We'd need a USB trace to see what exactly caused the problem.

Please let me know if there is anything I can do.

tore-espressif, 2024-09-03:
In the past 6 months we have tested with a dozen of different USB cameras and we did not see similar error. We can keep this open if someone gets the same error

nopnop2002, 2024-09-03:
@tore-espressif 
>In the past 6 months we have tested with a dozen of different USB cameras and we did not see similar error. 
I would like to know the device IDs of the 12 types of cameras.
We can check the device ID with lsusb.

sbyang369, 2024-11-02:
Hi! I also have been struggling with the same issue. I was wondering if you guys were able to find a fix.

[image]

Thank you!

nopnop2002, 2024-11-03:
@sbyang369 

It's hard to find a working USB camera.

Older Microsoft models work reliably.

Microsoft LifeCam Cinema -> Success
Microsoft LifeCam Studio -> Success
Microsoft LifeCam HD3000 -> Success
Microsoft LifeCam HD5000 -> Success

sbyang369, 2024-11-03:
Thank you!

If I want to run this project, do you think I should buy a new camera, or try different code?

nopnop2002, 2024-11-03:
You should buy a new camera.

tore-espressif, 2024-12-02:
Hello everyone who follows this very old issue.

We had many requests for USB cameras support and with High Speed capabilities of ESP32-P4 the pressure for this feature increased even more.
We are implementing UVC driver v2 which does not use libuvc anymore, but is natively built on top of ESP USB Host library. 

If you are interested in pre-release testing you can have a look here https://github.com/espressif/esp-usb/pull/84

nopnop2002, 2025-01-22:
I tried UVC driver v2.
https://github.com/espressif/esp-usb/tree/master/host/class/uvc/usb_host_uvc/examples/basic_uvc_stream

The device used is Logitech C615.

It does not work with esp-idf's usb_host_uvc.

# linux libuvc
It can capture in YUY2 and 640x480.
```
First format: (YUY2) 640x480 30fps
bmHint: 0001
bFormatIndex: 1
bFrameIndex: 1
dwFrameInterval: 333333
wKeyFrameRate: 0
wPFrameRate: 0
wCompQuality: 61
wCompWindowSize: 0
wDelay: 0
dwMaxVideoFrameSize: 614400
dwMaxPayloadTransferSize: 3060
bInterfaceNumber: 3
Streaming...
Enabling auto exposure ...
 ... full AE not supported, trying aperture priority mode
 ... enabled aperture priority auto exposure mode
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
callback! frame_format = 3, width = 640, height = 480, length = 614400, ptr = 12345
```

# esp-idf libuvc
It can't capture in YUY2 and 640x480.
```
I (2255) UVC example: Opening UVC device 0x0000:0x0000-0
        640x480@15.0FPS...
I (2255) UVC example: UVC_VS_FORMAT_YUY2
I (2255) main_task: Returned from app_main()
E (2985) USBH: Dev 1 EP 0 Error
E (2985) uvc: uvc_host_usb_ctrl(827): Control transfer error
E (2985) uvc-control: uvc_host_stream_control(66): Control request failed
E (2995) USBH: HCD Pipe not in active state
E (2995) USB HOST: Submit CTRL URB error: ESP_ERR_INVALID_STATE
E (3005) uvc: uvc_host_usb_ctrl(821): CTRL transfer failed
E (3015) uvc-control: uvc_host_stream_control(66): Control request failed
E (3015) uvc: uvc_host_stream_open(545): Failed to negotiate requested Video Stream format
I (3025) UVC example: USB: All devices freed
I (3035) UVC example: Failed to open device
```

tore-espressif, 2025-01-22:
@nopnop2002 I ordered the camera and will test ASAP

tore-espressif, 2025-01-30:
@nopnop2002 thank you for your patience. 

Could you please test if https://github.com/espressif/esp-usb/pull/129 fixes your problem?

nopnop2002, 2025-01-30:
@tore-espressif 

```
$ git branch
* fix/uvc_set_interface
  master

$ head host/class/uvc/usb_host_uvc/CHANGELOG.md
## 2.0.1

- Fixed negotiation for some non-conforming UVC devices (https://github.com/espressif/esp-idf/issues/9868)

## 2.0.0

- New version of the driver, native to Espressif's USB Host Library
- Removed libuvc dependency

## 1.0.4
```

```
I (2256) UVC example: Opening UVC device 0x0000:0x0000-0
        640x480@15.0FPS...
I (2256) UVC example: UVC_VS_FORMAT_YUY2
I (2256) main_task: Returned from app_main()
E (2566) USBH: Dev 0 EP 0 Error
E (2566) ENUM: Bad transfer status 1: CHECK_SHORT_DEV_DESC
E (2566) ENUM: [0:0] CHECK_SHORT_DEV_DESC FAILED
I (7256) UVC example: Failed to open device
```

tore-espressif, 2025-01-31:
@nopnop2002 Sorry about that, I force-pushed to https://github.com/espressif/esp-usb/pull/129 and tested with your frame format, all works now. Thank you again!

nopnop2002, 2025-02-02:
@tore-espressif 

I used Microsoft LifeCam Studio.
This warning will appear.
What should I change in the source?

```
I (2256) frame_handling: Opening UVC device 0x0000:0x0000-0
        640x480@15.0FPS...
I (2256) frame_handling: UVC_VS_FORMAT_YUY2
I (2256) main_task: Returned from app_main()
I (2676) frame_handling: Device 0x0000:0x0000-0 OPENED!
I (2776) frame_handling: Stream 0 start. Iteration 0
W (3156) UVC example: Frame buffer overflow
W (4016) frame_handling: Stream 0: Frame not received on time
I (4016) frame_handling: Stream 0 stop
```

tore-espressif, 2025-02-03:
@nopnop2002 You must set [frame_size](https://github.com/espressif/esp-usb/blob/43cc64e7b03be73f83d8ecd8e7460e156f0e5073/host/class/uvc/usb_host_uvc/include/usb/uvc_host.h#L132) to size of your frame, based on your logs, this is 614400 bytes .

Transferring frames so big at 15 FPS will not work on USB Full Speed though, this settings will need at lease 614kB x 15 = 9.2MB/s which is 10 times more of what we can do on USB Full Speed.

So we encourage users to use JPEG encoded frames to save bandwidth

nopnop2002, 2025-02-03:
@tore-espressif 

>So we encourage users to use JPEG encoded frames to save bandwidth

I tried UVC_VS_FORMAT_MJPEG.
It's work fine.
Thank you.

```
I (2225) UVC example: Installing USB Host
I (2255) UVC example: Installing UVC driver
I (2255) UVC example: Opening UVC device 0x0000:0x0000-0
        640x480@15.0FPS...
I (2255) UVC example: UVC_VS_FORMAT_MJPEG
I (2255) main_task: Returned from app_main()
I (2665) UVC example: Device 0x0000:0x0000-0 OPENED!
I (2765) UVC example: Stream 0 start. Iteration 0
I (3105) UVC example: Stream 0: New frame! Len: 20313
I (3185) UVC example: Stream 0: New frame! Len: 21795
I (3245) UVC example: Stream 0: New frame! Len: 21731
I (3305) UVC example: Stream 0: New frame! Len: 21703
I (3385) UVC example: Stream 0: New frame! Len: 21747
I (3445) UVC example: Stream 0: New frame! Len: 21702
```