espressif/esp-idf #9870: crosstool-ng version esp-2022r1 doesn't match supported version   esp-2022r1-RC1

ddomnik, 2022-09-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

012fafb82781f2720f2c636ed3b0ebd0db02cd9d (HEAD, tag: v5.1-dev)

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Complete installation ready to build.

### What is the actual behavior?

Switching from master to the tag v5.1-dev results in the following error when building the project using Eclipse.

```
Building in: C:\Espressif\frameworks\esp-idf-master\workspace\hello_world\build
Configuring in: C:\Espressif\frameworks\esp-idf-master\workspace\hello_world\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\Espressif\frameworks\esp-idf\tools\cmake\toolchain-esp32s3.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32s3 C:\Espressif\frameworks\esp-idf-master\workspace\hello_world
-- Found Git: C:/cygwin64/bin/git.exe (found version "2.36.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
-- Checking Python dependencies...
Skipping the download of C:\Espressif\espidf.constraints.v5.1.txt because it was downloaded recently.
Constraint file: C:\Espressif\espidf.constraints.v5.1.txt
Requirement files:
 - C:\Espressif\frameworks\esp-idf\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.1_py3.10_env\Scripts\python.exe
Python requirements are satisfied.
-- Project sdkconfig file C:/Espressif/frameworks/esp-idf-master/workspace/hello_world/sdkconfig
CMake Error at C:/Espressif/frameworks/esp-idf/tools/cmake/crosstool_version_check.cmake:37 (message):
-- Configuring incomplete, errors occurred!
See also "C:/Espressif/frameworks/esp-idf-master/workspace/hello_world/build/CMakeFiles/CMakeOutput.log".
  

  Toolchain:
  C:/Espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe,
  crosstool-ng version esp-2022r1 doesn't match supported version
  esp-2022r1-RC1

  Please try to run 'idf.py fullclean' to solve it quickly.

  Check Getting Started documentation if the error continues.

  You can override this error and proceed with build by defining the
  IDF_MAINTAINER environment variable.

Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf/components/esp_common/project_include.cmake:7 (crosstool_version_check)
  C:/Espressif/frameworks/esp-idf/tools/cmake/build.cmake:380 (include)
  C:/Espressif/frameworks/esp-idf/tools/cmake/build.cmake:601 (__build_process_project_includes)
  C:/Espressif/frameworks/esp-idf/tools/cmake/project.cmake:434 (idf_build_process)
  CMakeLists.txt:8 (project)

cmake --build . -- -v
ninja: error: loading 'build.ninja': Das System kann die angegebene Datei nicht finden.

Build complete (0 errors, 0 warnings): C:\Espressif\frameworks\esp-idf-master\workspace\hello_world\build
Total time taken to build the project: 12.736 ms
```

### Steps to reproduce.

1. Install Eclipse + IDF via esp-idf-tools-setup-online-2.16.exe installer (selecting download IDF master branch)
2. Run hellow_world example as test (successfull)
3. Clone idf via Git in frameworks folder (next to esp-idf-master) (git clone https://github.com/espressif/esp-idf.git)
5.  cd esp-idf/
6.  git checkout tags/v5.1-dev
7. git submodule update --init --recursive
8. In Eclipse choose Espressif > Download or use ESP-IDF -> Use an existing ESP-IDF directory ...
9.  Select the esp-idf folder
10. Press YES on popup to install tools
11. Let defaults as is:
[image]

igrr, 2022-09-27:
It looks like some installation step was missed, and the environment variables in Eclipse are still pointing to the newer toolchain used in the `master` branch. @kolipakakondal Please let me know if this issue should be moved to idf-eclipse-plugin or not.

@ddomnik just a question, why do you need to switch to `v5.1-dev` tag when you have already checked out `master` (which is newer than v5.1-dev)?

ddomnik, 2022-09-27:
@igrr thanks for the fast reply. We want to develop under the v5.1-dev tag. Using master was just a result of a new clean install via the Windows installer.

kolipakakondal, 2022-09-27:
Hello @ddomnik 

When you switch idf versions it's recommended to run the project full clean command - 

1. Right-click on the project in eclipse
2. Click on `ESP-IDF: Project Full Clean`

[image]

kolipakakondal, 2022-09-27:
If you still have a problem, please share the Install tools console log and product information log
https://github.com/espressif/idf-eclipse-plugin#espressif-idf-tools-console
https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-view-product-information

ddomnik, 2022-09-27:
hello @kolipakakondal the full clean command did not solve the problem.

Install tools log after fullclean:
```
[C:\Espressif\python_env\idf5.1_py3.10_env\Scripts\python.exe, C:\Espressif\frameworks\esp-idf\tools\idf.py, fullclean]
Executing action: fullclean
Executing action: remove_managed_components
Done
```

Product information:
```
Executing C:\Users\ddufner\AppData\Local\Programs\Python\Python310\python.exe C:\Espressif\frameworks\esp-idf\tools\idf_tools.py list
* xtensa-esp-elf-gdb: GDB for Xtensa
  - 11.2_20220823 (recommended, installed)
* riscv32-esp-elf-gdb: GDB for RISC-V
  - 11.2_20220823 (recommended, installed)
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2022r1-11.2.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2022r1-11.2.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2022r1-11.2.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 14.0.0-38679f0333 (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2022r1-11.2.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.35_20220830 (recommended, installed)
* cmake: CMake build system
  - 3.24.0 (recommended, installed)
  - 3.16.3 (supported)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20220706 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.3 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 4.6.2 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

CDT Build environment variables
PATH: C:\Espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;C:\Espressif\tools\xtensa-clang\14.0.0-38679f0333\xtensa-esp32-elf-clang\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2\;C:\Espressif\tools\idf-exe\1.0.3\;C:\Espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;C:\Espressif\python_env\idf5.1_py3.10_env\Scripts;C:\Espressif\frameworks\esp-idf-master\tools;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Bitvise SSH Client;C:\Program Files\PuTTY\;C:\cygwin64\bin;;C:\Program Files\Docker\Docker\resources\bin;C:\ProgramData\DockerDesktop\version-bin;C:\Users\ddufner\AppData\Local\Programs\Python\Python310\Scripts\;C:\Users\ddufner\AppData\Local\Programs\Python\Python310\;C:\Users\ddufner\AppData\Local\Microsoft\WindowsApps;C:\Users\ddufner\AppData\Local\Programs\Git\cmd;C:\Users\ddufner\AppData\Local\Programs\Microsoft VS Code\bin
USERDOMAIN_ROAMINGPROFILE: AED
PROCESSOR_LEVEL: 6
SESSIONNAME: Console
ALLUSERSPROFILE: C:\ProgramData
PROCESSOR_ARCHITECTURE: AMD64
ESP_IDF_VERSION: 5.1
PSModulePath: C:\Program Files\WindowsPowerShell\Modules;C:\WINDOWS\system32\WindowsPowerShell\v1.0\Modules
SystemDrive: C:
OPENOCD_SCRIPTS: C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\share\openocd\scripts
IDF_DEACTIVATE_FILE_PATH: C:\Users\ddufner\AppData\Local\Temp\tmp4ke2s7qwidf_10304
IDF_CCACHE_ENABLE: 1
USERNAME: ddufner
USERDNSDOMAIN: AED.DIR
ProgramFiles(x86): C:\Program Files (x86)
PATHEXT: .COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
DriverData: C:\Windows\System32\Drivers\DriverData
ProgramData: C:\ProgramData
IDF_COMPONENT_MANAGER: 1
ProgramW6432: C:\Program Files
__PSLockDownPolicy: 0
HOMEPATH: \Users\ddufner
PROCESSOR_IDENTIFIER: Intel64 Family 6 Model 158 Stepping 9, GenuineIntel
ProgramFiles: C:\Program Files
PUBLIC: C:\Users\Public
windir: C:\WINDOWS
[... 90 lines cut ...]
Requirement already satisfied: charset-normalizer<3,>=2 in c:\espressif\python_env\idf5.1_py3.10_env\lib\site-packages (from requests<3->idf-component-manager->-r C:\Espressif\frameworks\esp-idf-master\tools\requirements\requirements.core.txt (line 9)) (2.1.1)
Requirement already satisfied: idna<4,>=2.5 in c:\espressif\python_env\idf5.1_py3.10_env\lib\site-packages (from requests<3->idf-component-manager->-r C:\Espressif\frameworks\esp-idf-master\tools\requirements\requirements.core.txt (line 9)) (3.4)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in c:\espressif\python_env\idf5.1_py3.10_env\lib\site-packages (from requests<3->idf-component-manager->-r C:\Espressif\frameworks\esp-idf-master\tools\requirements\requirements.core.txt (line 9)) (1.26.12)
Skipping the download of C:\Espressif\espidf.constraints.v5.1.txt because it was downloaded recently.
Installing Python packages
 Constraint file: C:\Espressif\espidf.constraints.v5.1.txt
 Requirement files:
  - C:\Espressif\frameworks\esp-idf-master\tools\requirements\requirements.core.txt

[... cut ...] idf5.1_py3.10_env\Scripts;C:\Espressif\frameworks\esp-idf-master\tools;%PATH%
Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
Executing C:\Espressif\python_env\idf5.1_py3.10_env\Scripts\python.exe -m pip install websocket-client
Requirement already satisfied: websocket-client in c:\espressif\python_env\idf5.1_py3.10_env\lib\site-packages (1.4.1)
Configured CMake toolchain. You can check Preferences > C/C++ > CMake
Install tools completed.

````

igrr, 2022-09-27:
> We want to develop under the v5.1-dev tag.

Please make sure you understand the purpose of this tag. It is not a release tag, it just indicates the point where development of v5.1 is started. It is not recommended to use such tags for product development. Generally, we won't provide support for issues you may encounter for such a tag. It's better to use a stable release or a release branch for development. More information about the releases is available at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#which-version-should-i-start-with.

ddomnik, 2022-09-27:
> Please make sure you understand the purpose of this tag. It is not a release tag, it just indicates the point where development of v5.1 is started. It is not recommended to use such tags for product [...]

Thank you for the detailed information regarding this. In that case we will stay with our current version and will not update to the 5.1-dev tag. Regarding that, is there any date planned for the stable v5.0 release tag? Thanks

kolipakakondal, 2022-09-28:
Thanks @ddomnik for sharing logs.

Looks like I don't see any issue in installing tools at least looking at the logs, then the problem might be in the esp-idf selection while installing tools.

  > Toolchain:
  C:/Espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe,
  crosstool-ng version esp-2022r1 doesn't match supported version
  esp-2022r1-RC1

esp-idf `v5.1-dev` tag is having `esp-2022r1-RC1-11.2.0` recommended version for `esp32s3` as per [this](https://github.com/espressif/esp-idf/blob/012fafb82781f2720f2c636ed3b0ebd0db02cd9d/tools/tools.json#L358) but what's getting installed above is `esp-2022r1-11.2.0` so please make sure your esp-idf `C:\Espressif\frameworks\esp-idf\` is actually pointing to `v5.1-dev`

However, as recommended by Ivan, please stick to the stable releases.

igrr, 2022-09-29:
I can't give you a specific date, unfortunately. Our QA test has found a number of issues in v5.0-beta1 release and currently we are working on fixing them. This might take a few weeks to do the fixes and verify them.
In the meantime, the tip of release/v5.0 branch will give you a version which is not likely to change too much until the final release, so you can do most of the development work there.

Maksons, 2022-10-05:
Small note, about switching IDF version in eclipse which could be related.
Most places get updated automatic, but one place I found do not get updated is "Run configuration" - "Argument".
I always have to change it by hand.
Menu "Run" -> "Run Configurations..." -> "Arguments"
But I use predownloaded ESP-IDF and "Use an existing ESP-IDF directory from file system".
And I have modified "Arguments" with higher bitrate and only app-flash.
Like:
[image]

AxelLin, 2022-12-18:
> Regarding that, is there any date planned for the stable v5.0 release tag? Thanks

https://github.com/espressif/esp-idf/releases/tag/v5.0

kolipakakondal, 2022-12-20:
> Small note, about switching IDF version in eclipse which could be related. Most places get updated automatic, but one place I found do not get updated is "Run configuration" - "Argument". I always have to change it by hand. Menu "Run" -> "Run Configurations..." -> "Arguments" But I use predownloaded ESP-IDF and "Use an existing ESP-IDF directory from file system". And I have modified "Arguments" with higher bitrate and only app-flash. It would be nice to have Eclipse internal variables with selected ESP-IDF path, environment path, etc, so project could be easy portable between PCs. Like: [image]

Agree with you. We will work on this. We could maintain the relative path (relative to IDF_PATH and .espressif folder) for the Location, working directory, and arguments section in the launch configuration wizard so that when the user switches between IDF versions and systems, the launch configuration doesn't need to make any changes.

For our internal reference https://jira.espressif.com:8443/browse/IEP-837

kolipakakondal, 2025-03-12:
As we have implemented dynamic variables for the esp-idf verison in the [espressif-ide](https://docs.espressif.com/projects/espressif-ide/en/latest/index.html), we can close this issue.

[image]

closed 2025-03-12 by kolipakakondal