espressif/esp-idf #9872: 如何將esp_prov的python腳本轉換為能夠在瀏覽器中執行的javascript程式

nelson40514, 2022-09-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### 環境
ESP-IDF v4.4.2

### 過程
使用https://github.com/espressif/esp-idf/tree/master/tools/esp_prov的python腳本進行開發測試

### 問題

想請問關於 esp-idf/tools/esp_prov 當中的 esp_prov.py 腳本，
進行 wifi provision 的這些操作步驟是否有 javascript 的版本能夠做使用？
希望能夠將操作流程移植到瀏覽器上。

lawrenceko-intecular, 2022-10-03:
The Wi-Fi provisioning can be migrated to webpage via JS? The python example includes a lot of library function call and is hard to migrate to JS.

mahavirj, 2022-10-06:
Current resources for provisioning are highlighted [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/provisioning/provisioning.html#provisioning-tools). We do not have plans for JS implementation here, however there was some discussion on this topic at https://github.com/espressif/esp-idf/issues/4399 that you may find useful.

closed 2022-10-08 by Alvin1Zhang