espressif/esp-idf #9874: 使用ubuntu22.04 install.sh esp-idfv4.4.2报错

yuandonglp, 2022-09-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### What is the expected behavior?

使用ubuntu22.04 install.sh esp-idfv4.4.2

### What is the actual behavior?

无法install成功

### Steps to reproduce.

1. Step
2. Step
3. Step
...
1、git v4.4.2分支
2、install

### Build or installation Logs.

```plain
root@yd-virtual-machine:~/esp32s3/esp-idf# ../esp-gitee-tools/install.sh 
Detecting Python interpreter
Checking "python" ...
Python 3.10.6
"python" has been detected
Installing ESP-IDF tools
Current system platform: linux-amd64
Traceback (most recent call last):
  File "/root/esp32s3/esp-idf/tools/idf_tools.py", line 1860, in <module>
    main(sys.argv[1:])
  File "/root/esp32s3/esp-idf/tools/idf_tools.py", line 1856, in main
    action_func(args)
  File "/root/esp32s3/esp-idf/tools/idf_tools.py", line 1423, in action_install
    targets = clean_targets(args.targets)
  File "/root/esp32s3/esp-idf/tools/idf_tools.py", line 1085, in clean_targets
    export_targets_to_idf_env_json(targets_from_tools_json)
  File "/root/esp32s3/esp-idf/tools/idf_tools.py", line 1053, in export_targets_to_idf_env_json
    targets = list(set(targets + get_user_defined_targets()))
  File "/root/esp32s3/esp-idf/tools/idf_tools.py", line 1099, in get_user_defined_targets
    if env == idf_env_json['idfSelectedId']:
KeyError: 'idfSelectedId'
root@yd-virtual-machine:~/esp32s3/esp-idf#
```

AxelLin, 2022-09-28:
Duplicate of https://github.com/espressif/esp-idf/issues/9837
Fixed in v4.4 branch, 640f1c844b5cd935e809b0195df06f1639eb9979 .

dobairoland, 2022-09-28:
Please follow the linked issue. You can find temporary workaround there.

closed 2022-09-28 by dobairoland

guoxiaolong5669, 2024-05-28:
我也遇到了这个问题 不知道怎么解决