espressif/esp-idf #9883: UART Rx stalls after approx. 40 min of transfer

g3gg0, 2022-09-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF 4.4.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### Development Kit.

ESP32-WROOM

### Power Supply used.

External 5V

### What is the expected behavior?

I have built a WiFi <-> UART gateway using the ESP32.
It is expected to read data from UART via uart_read_bytes()  and pass to a TCP socket and vice versa.
(see https://www.g3gg0.de/wordpress/esp32/esp32-a-tiny-wifi-bt-to-true-rs232-adapter/ )

### What is the actual behavior?

Unfortunately after a 40 minute transfer period, the Rx path of the ESP32 stalls and no activity on UART is "seen" anymore.
If you then wait 40 minutes again, the data is suddenly seen again and the ESP32 works as before.

[40 min working] [40 min no Rx] [40 min working] ....

Tx is working as intended, the connected external peripheral is receiving the data as before.

### Details
First I used the arduino Serial implementation for Rx/Tx, but switched to the underlying IDF as a measure to overcome this problem.
But the problem persisted.
This problem is known to me for over a year now, not seeing any fix so far.
[image]
https://github.com/g3gg0/ESP232/blob/loopback_test/src/Serial.ino#L200
There is the code which reads data from the UART buffers, both Serial.available() and uart_read_bytes() **return zero** during that "no Rx" period although the data is definitely seen by the ESP32's GPIOs.
(confirmed with a scope, also being sane and decodeable)

### Deep dive
Now I digged into debugging, first by investigating the UART peripheral registers.

Status "Rx/Tx are working properly"
[image]

Status: "Tx working, Rx stalled"
[image]

There I could see, that the UART_INT_ENA_REG shows disabled Rx interrupts, but even when forcefully enabling again using a telnet based debug interface, they were disabled again instantly.
In the "no Rx" period, the UART_STATUS_REG however shows the RXFIFO_CNT field is counting continuously as opposed to the "working" period where it stays virtually zero.
[image]

### This leads me to my personal conclusion:
### The underlying ESP-IDF has a bug, causing the Rx path to maybe overflow some counters, queues or whatever.

### Steps to reproduce.

I modified my project (platformio based) to "self-serve" the communication using UART1 in the branch loopback_test. It connects UART0 and UART1 Rx/Tx together and is simulating some SCPI communication.
And guess what, it also stalls after approx. that period.
Sources: https://github.com/g3gg0/ESP232/tree/loopback_test

Just build and flash that project. The first time the SPIFFS will format a few seconds, but after that you should see some communication like that:

SCPI communication oder UART (simplified):
```
> *OPC?
< 1
> *CLS
> DISP:WIND1:DATA?
< " -0.0000 A"
> *STB?
< 1
```
Repeating 4 times a second.

Screenshot in action:
[image]

after the 40 minutes, the communication would look like this:
[image]

And this whole process will repeat over and over.

If you want to break into it, the lines 92, 129 and 165 will give you a good spot 
https://github.com/g3gg0/ESP232/blob/loopback_test/src/SCPI.ino#L129

g3gg0, 2022-09-29:
A dev member provided me a fork of platform-espressif32, which uses 
https://github.com/tasmota/esp-idf/tree/v4.4.3.5
which again seems to be at f4d33d27395e1ad9cc4a1979573cad3d9395c9a6 so it includes 1e117f2f83e200f4fb6fd9452a0d8485a29d2a62 ("UART: RX interrupts are now properly restored after a flush")

So my hope that this could also have addressed my issue is gone now :)

chipweinberger, 2022-10-01:
What tool are you using to inspect registers? looks useful

g3gg0, 2022-10-01:
this is just a quick-hack telnet interface for viewing and reading registers.
didn't want to build a JTAG interface with a day or more of effort must to realize "must be something in the driver"
see https://github.com/g3gg0/ESP232/blob/loopback_test/src/Debug.ino

g3gg0, 2022-10-01:
@igrr you are absolutely right.
Holy moly.
Maybe it *was* fixed long before, but I introduced my own bug.
Back then the error "no Rx" was much more sporadic, not bound to 35 minutes (and did not recover).

Thats why the UDP output in line 210 was even there - intially to send me *all* return values via UDP.
This flooded my whole network, but I could see every return value, also all the zeros that suddenly appeared.
So I concluded an issue in the driver and even digged into the IDF, looking for signed integers being able to overflow...
how ironic...

I am currently feeling ashamed and am running the testcase, but this looks like a quite obvious fail on my side :)
Thanks a lot for pointing me there!

g3gg0, 2022-10-01:
test confirmed: user error.

thanks @igrr!

closed 2022-10-01 by g3gg0