espressif/esp-idf #9887: V4.2.2 esp_restart() hangs infinitive in wifi:hmac tx: ifx0 stop, discard and does not reset

ralphde, 2022-09-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v4.4.2-201-g2b706909e1-dirty

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### Development Kit.

Custom S2 board

### Power Supply used.

External 5V

### What is the expected behavior?

I expect a reset, no matter if any module denies a stop. I do not expect  a clean stop of any service I expect a RESET

### What is the actual behavior?

program loops infinitive at  wifi:hmac tx: ifx0 stop, discard

### Steps to reproduce.

on an HTTPS server application issue an esp_restart()

AxelLin, 2022-09-30:
> ### IDF version.
> ESP-IDF v4.4.2-201-g2b706909e1-dirty

Please update your subject line if you mean V4.4.2 rather than V4.2.2.

ralphde, 2022-09-30:
ups updated

ralphde, 2022-10-09:
When will this be fixed?

KonssnoK, 2022-10-13:
having the same issue

KonssnoK, 2022-10-13:
**NOTE:**
the issue is related to using the external oscillator instead of an internal one
@ralphde please try to set internal oscillator in menuconfig, i guess your S2 custom board has an external one too.
If i put internal oscillator the board reboots correctly.

ralphde, 2022-10-14:
We are using ESP32-S2-WROVER-N4R2 and I can not locate any settings using sdkconfig in Eclipse regarding internal and external oscillator.

KonssnoK, 2022-10-14:
ah dammit, then i guess our issues are different..
i saw you put "custom s2 board" so i thought it was with an external osc as ours...
maybe the S2 has no osc pin
[image]

I'll open another issue then!

ralphde, 2022-10-22:
@MaxwellAlan 

> Could you please upload the full logs for this and sdkconfig.

what logs you are revering to? the program runs fine and does what it is supposed to and that is at this point to wait for a semaphore to come up, but it never comes. the uart log throughs this message "wifi:hmac tx: ifx0 stop, discard" and loops with this infinitive.

>Could you provide a demo code for we to reproduce this issue?

I can provide my app but in private message. The issue occurs in my app when calling esp_restart() in a HTTPS server task. 
Nevertheless, if I request a restart I give shit about any clean shutdown of ANY task. If such a behavior is programmed into this function it should be called esp32_shutdown_and_restart() and a hard esp32_restart() should do what any expects.

JustQY, 2022-12-06:
我也遇到这个问题了，这是log，偶然做功能时发现的
功能，本来是设计按键多长时间重启，结果代码错误，造成开机就重启，重启几次后就出现问题了

`rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:5700
load:0x40078000,len:14656
ho 0 tail 12 room 4
load:0x40080400,len:3728
0x40080400: _init at ??:?

entry 0x4008067c
I (29) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (29) boot: compile time 13:49:41
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (77) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (84) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (92) boot:  3 ota_0            OTA app          00 10 00010000 001e0000
I (99) boot:  4 ota_1            OTA app          00 11 001f0000 001e0000
I (107) boot:  5 coredump         Unknown data     01 03 003d0000 00010000
I (114) boot:  6 reserved         Unknown data     01 fe 003e0000 00020000
I (122) boot: End of partition table
I (126) boot_comm: chip revision: 3, min. application chip revision: 0
I (133) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=36158h (221528) map
I (222) esp_image: segment 1: paddr=00046180 vaddr=3ffbdb60 size=04ebch ( 20156) load
I (230) esp_image: segment 2: paddr=0004b044 vaddr=40080000 size=04fd4h ( 20436) load
I (239) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=123714h (1193748) map
I (671) esp_image: segment 4: paddr=0017373c vaddr=40084fd4 size=18b74h (101236) load
I (713) esp_image: segment 5: paddr=0018c2b8 vaddr=50000000 size=00010h (    16) load
I (727) boot: Loaded app from partition at offset 0x10000
I (728) boot: Disabling RNG early entropy source...
I (739) cpu_start: Pro cpu up.
I (740) cpu_start: Starting app cpu, entry point is 0x40080d70
0x40080d70: call_start_cpu1 at G:/espressif/frameworks/esp-idf-v4.4.2/components/esp_system/port/cpu_start.c:160

I (725) cpu_start: App cpu up.
I (756) cpu_start: Pro cpu start user code
I (756) cpu_start: cpu freq: 160000000
I (756) cpu_start: Application information:
I (760) cpu_start: Project name:     86Sensor
I (765) cpu_start: App version:      1
I (770) cpu_start: Compile time:     Dec  6 2022 14:14:02
I (776) cpu_start: ELF file SHA256:  138ccf49670a4004...
I (782) cpu_start: ESP-IDF:          v4.4.2-dirty
I (788) heap_init: Initializing. RAM available for dynamic allocation:
I (795) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (801) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (807) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (813) heap_init: At 3FFCC0C8 len 00013F38 (79 KiB): DRAM
I (819) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (825) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (832) heap_init: At 4009DB48 len 000024B8 (9 KiB): IRAM
I (839) spi_flash: detected chip: generic
I (843) spi_flash: flash io: dio
I (848) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
E (947) system: config 48595a4e
W (948) main: SystemUUID 58BF25E5FA98 ver 1.1.0
I (948) WifiApp: isneedap 1
I (954) wifi:wifi driver task: 3ffd2018, prio:23, stack:6656, core=0
[... 580 lines cut ...]
0x400998b9: ppTask at ??:?

0x40094581: vPortTaskWrapper at G:/espressif/frameworks/esp-idf-v4.4.2/components/freertos/port/xtensa/port.c:131

E (22954) task_wdt: Print CPU 1 backtrace

Backtrace:0x4008C6F9:0x3FFBEF100x400829ED:0x3FFBEF30 0x4000BFED:0x3FFBCE80 0x400947CF:0x3FFBCE90 0x4013FC5F:0x3FFBCEB0 0x4013FC6B:0x3FFBCEE0 0x400D369D:0x3FFBCF00 0x40092D95:0x3FFBCF20 0x40094581:0x3FFBCF40
0x4008c6f9: esp_crosscore_isr at G:/espressif/frameworks/esp-idf-v4.4.2/components/esp_system/crosscore_int.c:92

0x400829ed: _xt_lowint1 at G:/espressif/frameworks/esp-idf-v4.4.2/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x400947cf: vPortClearInterruptMaskFromISR at G:/espressif/frameworks/esp-idf-v4.4.2/components/freertos/port/xtensa/include/freertos/portmacro.h:571
 (inlined by) vPortExitCritical at G:/espressif/frameworks/esp-idf-v4.4.2/components/freertos/port/xtensa/port.c:319

0x4013fc5f: esp_task_wdt_reset at G:/espressif/frameworks/esp-idf-v4.4.2/components/esp_system/task_wdt.c:330

0x4013fc6b: idle_hook_cb at G:/espressif/frameworks/esp-idf-v4.4.2/components/esp_system/task_wdt.c:80

0x400d369d: esp_vApplicationIdleHook at G:/espressif/frameworks/esp-idf-v4.4.2/components/esp_system/freertos_hooks.c:51 (discriminator 1)

0x40092d95: prvIdleTask at G:/espressif/frameworks/esp-idf-v4.4.2/components/freertos/tasks.c:3973 (discriminator 1)

0x40094581: vPortTaskWrapper at G:/espressif/frameworks/esp-idf-v4.4.2/components/freertos/port/xtensa/port.c:131

`

JustQY, 2022-12-06:
use function “esp_system_abort” has no this issue

anatol-grabowski, 2023-03-26:
In my case the problem was that I was calling esp_restart() within an http server handler. I suspect there is a deadlock of some kind - trying to stop the server before restarting, but waiting for the handler that called for restart to finish before stopping the server.
Here is the solution (not tested, mine is a bit more complex, but should work this way):
```
xTaskCreate(esp_restart, "esp_restart task", 1024, NULL, 1, NULL);
```

zhangyanjiaoesp, 2023-05-22:
@ralphde Do you still have this problem? When there is a packet to be send ,and the wifi is not started, the  log `wifi:hmac tx: ifx0 stop, discard` will be shown.

closed 2024-02-29 by Alvin1Zhang